wled/WLED #1: Error Compiling

jdwolin, 2018-01-03:
This code looks awesome but I'm having a hard time compiling it on my end.   I’m using Arduino 1.8.4 and the latest ESP core.   I've tried multiple boards including the one you recommended in the readme.md file.  Everything seems to compile fine until almost the very end and get this error message:  

Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\JASONW~1\AppData\Local\Temp\arduino_cache_215143\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_7b72b833e5a4650743b2f437f2ae88d2.a
sketch\wled00.ino.cpp.o: In function `~StringSumHelper':

C:\Users\Jason Wolin\Dropbox\My Hobbies\ClockProduction\Code\Belgrade\wled00/wled12_alexa.ino:51: undefined reference to `NeoEsp8266Uart::NeoEsp8266Uart(unsigned short, unsigned int)'

sketch\wled00.ino.cpp.o: In function `__static_initialization_and_destruction_0':

C:\Users\Jason Wolin\Dropbox\My Hobbies\ClockProduction\Code\Belgrade\wled00/wled12_alexa.ino:51: undefined reference to `NeoEsp8266Uart::NeoEsp8266Uart(unsigned short, unsigned int)'

sketch\wled00.ino.cpp.o: In function `WS2812FX':

C:\Users\Jason Wolin\Dropbox\My Hobbies\ClockProduction\Code\Belgrade\wled00/wled12_alexa.ino:51: undefined reference to `NeoEsp8266Uart::~NeoEsp8266Uart()'

C:\Users\Jason Wolin\Dropbox\My Hobbies\ClockProduction\Code\Belgrade\wled00/wled12_alexa.ino:51: undefined reference to `NeoEsp8266Uart::~NeoEsp8266Uart()'

sketch\WS2812FX.cpp.o: In function `WS2812FX::setRange(int, int, unsigned int)':

sketch/WS2812FX.cpp:1904: undefined reference to `NeoEsp8266Uart::UpdateUart()'

sketch\WS2812FX.cpp.o: In function `NeoEsp8266UartMethodBase<NeoEsp8266UartSpeed800Kbps, NeoEsp8266Uart>::Update()':

sketch/WS2812FX.cpp:1904: undefined reference to `NeoEsp8266Uart::UpdateUart()'

sketch\WS2812FX.cpp.o:(.text._ZN11NeoPixelBusI13NeoGrbFeature24NeoEsp8266UartMethodBaseI26NeoEsp8266UartSpeed800Kbps14NeoEsp8266UartEE5BeginEv[_ZN11NeoPixelBusI13NeoGrbFeature24NeoEsp8266UartMethodBaseI26NeoEsp8266UartSpeed800Kbps14NeoEsp8266UartEE5BeginEv]+0x4): undefined reference to `NeoEsp8266Uart::InitializeUart(unsigned int)'

sketch\WS2812FX.cpp.o: In function `NeoEsp8266UartMethodBase<NeoEsp8266UartSpeed800Kbps, NeoEsp8266Uart>::Initialize()':

sketch/WS2812FX.cpp:1904: undefined reference to `NeoEsp8266Uart::InitializeUart(unsigned int)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jdwolin, 2018-01-04:
Fixed the issue:  I had to uninstall and reinstall the neopixelbus library.  Wroks perfect now.

closed 2018-01-04 by jdwolin

Aircoookie, 2018-01-08:
Just saw this now! Awesome that it works for you now! If you have any suggestions for future features, feel free to ask :)