wled/WLED #2: Not really an issue, but...

DaveDischord, 2018-01-05:
Would you happen to know how to port your project to ESP32?  All my kids want me to make them a night light/party light for their rooms, and I'm running out of esp8266s, but have several esp32s I could use.  Its an awesome application!

Aircoookie, 2018-01-08:
Thanks for the feedback :) Great that you like it!
I'll order an ESP32 and try porting the project. In fact, I already did this half a year ago, but failed since an important library (EEPROM) was not yet availabe at the time, but is now!
Supporting ESP32 was already on my long-term todo list for the project, but I'll make sure to add it as soon as possible 👍

Aircoookie, 2018-01-09:
Glad to report that even though I don't yet have any ESP32 hardware I got it to compile with the latest ESP32 arduino core!
I've added a binary and the code in the last commit. The IDE detects which board you compile for automatically, no need to change anything in the code. In the file "WS2812FX.h" you can change the LED pin in the 2nd line.
Keep in mind though that I was obviously unable to test it before releasing, so it might not work at all, or it might work perfectly fine for you.
One feature that doesn't work for sure is the binary OTA update via HTTP, since the library that mechanism requires doesn't exist for ESP32 (yet).

It would mean a lot to me if you could give me feedback on how well the ESP32 version works for you!

DaveDischord, 2018-01-10:
I've finally gotten around to trying it on an ESP32.  The code uploads, and the AP/Web interface works well, as with the ESP8266, however, my neopixels are glitchy/flickering.  I tried a 60-pixel and 12-pixel ring with and without a 3.3V level converter, adequate power, and a few different pins.  The first pixel works and even adjusts with the interface.

Aircoookie, 2018-01-10:
First off all, great that the web interface works!
I've looked into the flickering issue.
It is caused by the NeoPixelBus library which is not yet ironed out for the ESP32. They use software bitbanging instead of a hardware method like UART on the ESP8266. (https://github.com/Makuna/NeoPixelBus/issues/152)

I've added a delay(1); before the LED show() update. Also, the method was changed from "Neo800KbpsMethod" to "NeoWs2813Method". These might help solve the flickering. If it doesn't, change line 6 of WS2812FX.h back.
By the way, have you lowered the LED count in the settings ("led setup") to 12 or 60? Apparently the ESP32 performs better if it tries to use less LEDs. After that you could also try and reduce the max LED count in wled00.ino, line 53 (must still be at least the same LED count you have in the settings!).

On a different note, my ESP32 will arrive on Saturday so I can then test the code, too!
Thank you so much for helping me improve WLED!

DaveDischord, 2018-01-12:
I've uploaded your new changes and am testing/dabbling in settings, etc.  I apologize for not being Arduino-savvy, but I certainly don't mind blackbox testing.  Its a load of fun and this project is great, but my feedback might not be worded the best.  With that, currently I have kept your changes, aside from ledcount(line 91) and LEDCOUNT(line 53) both which have been set to 60.  On startup, there is no flicker, but also no color, and RGB sliders won't bring any up. As before however, the chase effect seems to work pretty well, and this time I couldn't see noticeable flickering on the unlit pixels.  
Something else I found with messing with Effects.
-   I pressed the Rocket icon and it started up (Effect set to 53), then I arrowed over to 54 and all pixels went to solid. 
-  From there I could go to the RGB sliders and change the color and brightness, affecting all pixels, with a lightly noticeable, transitional chase effect on change that lasted about a second.
-  Then, if I go to settings and Select the 'Ignore and use current...) option, and reboot, the neopixels start up with that color, rather than all off.  
I don't know what Effect value 54 does, but its a workaround as of now.  Colors seem to be on point.

Aircoookie, 2018-01-12:
Hi again, it makes me very happy that you like tinkering with the project so much and found a workaround!
From your description, it seems like the program is behaving like it's meant to, there are just two very strange exceptions.
- The changes seem to have successfully mitigated the flickering, which is great!
- The transitional chase effect is intended. In fact, you can toggle it and change the duration in the settings (under "Transitions")
- What is very weird though is that the default effect 0 doesn't light up the Neopixels. Also, if this worked at the beginning for you (when the flickering was there), I can't think of any way my changes could have caused it to stop working.
- Even stranger is that effect 54 is a solid color for you. It is in fact supposed to be a chase effect with a color-changing background.

I will make sure to try and reproduce your findings as soon as I get my hands on an ESP32 (hopefully tomorrow). In the meantime, you could try to paste **http://10.10.1.11/win&A=100&FX=0&&R=0&G=255&B=255** in your address bar (replace 10.10.1.11 with the IP of your ESP). It should put the pixels to a medium brightness light blue.

Thank you once again for taking the time to help me with the application!
By the way, the fact that you got it uploaded and working means that you are more arduino-savvy than 99% of people :) Be proud!

Aircoookie, 2018-01-16:
Good news! Today I finally got around connecting my ESP32 to some LEDs and was perfectly able to reproduce what you found out.
Based on further experiments I was able to find the culprit. An array that performs an anchient pixel lock function was not being initialized. In fact, it's stranger that this worked on ESP8266 than it is that it didn't on ESP32!
I've adapted the code and everything seems to be working as intended now! All the effects are functional. It's not quite as stable as the 8266 version yet, I could observe some rare reboots while changing values and there IS flickering if you set the effect speeds to insane values (the lower you set the led count in the settings, the better it gets, but no need to lower the max LEDCOUNT in wled00.ino anymore). However, for everyday use it seems fine for now! Hope that it'll work properly for you also! :)

DaveDischord, 2018-01-16:
Awesome!!!  I will play around with it as soon as I can.  Hopefully I can find a way to incorporate bluetooth, though I believe there's limitation on running BLE and Wifi simultaneously.

Aircoookie, 2018-01-16:
Bluetooth would be really cool! What would you plan to do with it? The only reason I'd think it would be useful is with a mobile app because then you would not have to enter the IP address in the browser, or in fact, don't even need the ESP connected to your network or opening an AP. However, it would be quite a complex task since all the bluetooth networking and a new UI would need to be implemented from scratch.

I've slightly changed the code again. Now the flickering is virtually not present anymore. It is still not an ideal approach, but I'm satisfied with it for now. The issue with my previous workaround was that it only worked with up to 33 pixels. 34 and beyond would flicker. By disabling interrupts, this should now be fixed. If you are interested, the issue is described here: https://github.com/adafruit/Adafruit_NeoPixel/issues/139 . At the moment, there is nothing more I could do to improve the flickering (except switching the library, which would remove support for RGBW or any strip type besides WS2812B), but like I said its definiately useable now! By the way, the reboots I've observed previously are not a software issue. My power cable was broken, so I only had 4V voltage, which lead to those instabilities.

Aircoookie, 2018-03-15:
I've finally found a solution for Hardware RMT ESP32 support. This means no flickering on ESP32 anymore (as of v0.6.0)! However, you'll need to use this fork of NeoPixelBus: https://github.com/svenihoney/NeoPixelBus/tree/b0485e09b374b368d5445d319b12501886da9788 since the changes aren't implemented in the master library (yet).  Once you have that library installed and the original NeoPixelBus removed from your libraries, you can comment out the `#define WORKAROUND_ESP32_BITBANG` in WS2812FX.h and enjoy a fast and flicker free ESP32 implementation of WLED!

closed 2019-10-25 by Aircoookie

tonyn0, 2020-01-01:
Is that fork of NeoPixelBus still relevant?

Aircoookie, 2020-01-06:
@tonyn0 no, Rmt support (even all 8 channels!) was added to NeoPixelBus master about a year ago, so that fork is obsolete now :)