wled/WLED #3: Alexa control

Master-P96, 2018-02-03:
Hi, I dont know if it is an issue or if I do a mistake.

I just set up my ESP8266 with your basic HTTP OTA updater installing the wled05dev_1801102_RGB_ESP8266.bin, wich was quite easy. Thumbs up!

After this I typed in my wifi data and everything seems to work. (I haven't connected my LED's yet).

But when I search for new devices with my echo dot, it cant find the esp8266.
The option "Emulate Alexa device" is on.
Do I have to activate a skill or so?

Aircoookie, 2018-02-03:
Hi @Master-P96 ,

thank you for trying out my software! 😄 Awesome you had no troubles with the installation!

The Alexa integration works without a skill. Please check first if the settings field "Alexa Invocation Name" is filled with the device name you want it to have. Please reset the ESP once, because Alexa Emulation will only work after a reboot. Also make sure your router has UPnP enabled or try searching via the Alexa app or multiple times. This may help :)

If that doesn't help, may I ask what ESP8266 board you are using (ESP-01, wemos d1, Nodemcu, ...)? The binaries should work for all ESPs but we could make sure that is not the problem by compiling a binary just for your board.

Hope I'm able to help!
-Aircoookie

Master-P96, 2018-02-04:
Thank you for your fast reply :)
I just connected my LED's and it works fantastic! Very nice work.

But back to my issue:
Alexa Invocation Name is filled (I also changed the name several times).
Resetting and multiple times of searching for a device did not solve the problem.
I have a FritzBox 7580 with UPnP enabled.
Does an update with the integrated updatefunction (upload the bin) count as a "fresh" installation?

I use an NodeMCU Lolin V3.

Thanks for your help!

Aircoookie, 2018-02-04:
I'm so happy you like my work :)

Here is a recent binary compiled for your board: https://drive.google.com/file/d/1Hcw5_LpsTbWiLaXj9gFqo1kOPUwi2ZOS/view
Since everything but Alexa control seems to work fine for you though I don't know if this will solve the issue. I have also turned on serial debug for this build. If you open the Arduino IDE serial monitor and ask Alexa to discover devices, it should normally flood it with debug data like "AlexaApiCall".

When I tried it myself again today, Alexa said she found no new devices, but voice control worked reliably anyways. To be honest, Alexa is a bit weird sometimes. It also occasionally happens that she says: "Device doesn't react. Please check power and network", but the light turns on anyways. One time though, Alexa was unable to control any device (even the ones already discovered) until I power cycled the Echo itself, then it was flawless. Strange stuff.

If that doesn't work, you could also try and use the binary "wled04_1712122_RGB.bin". It is an older version, but it uses a different way to communicate with Alexa (doesn't support dimming, just on and off, but it just might work for you)

A "fresh install" is unlikely to solve anything. Just in case you'd like to try it, the software preserves your settings (Wifi, led, Alexa...) in file system memory, which means it will "survive" software updates. If you use the "ip/update" page to do a .bin update, to reset the settings, you'll also need to scroll way down in the settings page and tick the "Factory Reset" checkbox, then reboot. It will delete all settings, including your WiFi password and everything you had changed.

I really hope for you that we can fix the issue! (since I myself have ~10 ESPs running the software and pretty much only control them via Alexa, since it's so much quicker than opening the browser) Please tell me if any of my suggestions worked :)

By the way, in 1-2 weeks I'll  upload a large update that will improve the looks of the web page (themes) and add a couple of new effects and features! Stay tuned for that!

Master-P96, 2018-02-05:
Okay here is the output:

`
.
WiFi Connected to Air Force 1
IP address: 192.168.178.54
Ask Alexa to discover devices

Connecting UDP
UDP success
HTTP Server started...
Received UDP: 192.168.178.1:60369
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1

Received UDP: 192.168.178.30:48667
:M-SEARCH * HTTP/1.1
Host: 239.255.255.250:1900
Man: "ssdp:discover"
ST: ssdp:all
MX: 5

Received UDP: 192.168.178.1:60369
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 5
ST: urn:schemas-upnp-org:device:avm-aha:1

Received UDP: 192.168.178.44:34746
:M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 2
ST: ssdp:all

Received UDP: 192.168.178.30:37992
:NOTIFY * HTTP/1.1
HOST:239.255.255.250:1900
CACHE-CONTROL:max-age=1810
LOCATION:http://192.168.178.30:8200/rootDesc.xml
SERVER: 3.4.6-generic Microsoft-Windows/6.1 Windows-Media-Player-DMS/12.0.7601.17514 DLNADOC/1.50 UPnP/1.0 QNAPDLNA/1.0
NT:urn:sch
Received UDP: 192.168.178.30:37992
:NOTIFY * HTTP/1.1
HOST:239.255.255.250:1900
CACHE-CONTROL:max-age=1810
LOCATION:http://192.168.178.30:8200/rootDesc.xml
SERVER: 3.4.6-generic Microsoft-Windows/6.1 Windows-Media-Player-DMS/12.0.7601.17514 DLNADOC/1.50 UPnP/1.0 QNAPDLNA/1.0
NT:urn:mic
`
While the output was created I made an attempt to find new devices with the echo.
The language is german.

Hope it helps you!

Aircoookie, 2018-02-05:
Ok, an der Sprache liegt es schonmal nicht :) Danke für die Hilfe!

From what you posted it seems like your Echo doesn't send the required UDP device discovery requests. There are none in the log at least...

I did some more testings and it seems like Echo dot searches for `urn:schemas-upnp-org:device:basic:1` devices while the Echo 2/Plus searches for `upnp:rootdevice` instead. This is very strange, but here is a new binary (which accepts both requests) that may or may not work for you:
https://drive.google.com/open?id=1aN8bO10qiLvRbGyXYBaWixOAP2nyQ46-
Please unplug and restart your Echo once again before trying it out, and don't forget to try it via the app also.

This binary should theoretically work for you if WLED 0.4 does, but at this point I can't guarantee for anything... If it doesn't, I have no idea how to fix it since the echo does not seem to send the discovery requests it is supposed to. I really wish for you that it will work though! :)

Master-P96, 2018-02-05:
Well I think you fixed it!
With the new version the echo found the ESP without problems :)

Nice work mate, and thanks for the great support!
I really looking forward to the upcoming development of this project :thumbsup:

Aircoookie, 2018-02-05:
That's perfect news! I was afraid it was not fixable, but it worked out and I'll definiately include the change in the next update :)

closed 2018-02-06 by Master-P96

wcbuerste, 2018-02-09:
Hi, this is an awesome project idea and I hope I´ll be able to get it working.
I basically have the same problem like Master-P96 while using a Adafruit huzzah and Echo Dot. Sadly i´m to stupid to get you custom binary uploaded so far... Any possibility to provide the customizes ino file?

Thank you so much for all the work you put into this!

Aircoookie, 2018-02-10:
Hi @wcbuerste ,

I'm very happy that you like my project, really appreciate it :)

Sadly I can't offer the updated ino files yet, because I'm currently working on a big update/restructuring, which means that the sketch does not compile/work at the moment.

However, you can easily fix the problem with Alexa yourself 👍 ! Just go to the sketch tab "wled12_alexa.ino" in the top of the Arduino IDE.
Then, in that file, just replace line 37:
This:
`if(request.indexOf("upnp:rootdevice") > 0) {`
needs to be replaced by:
`if(request.indexOf("upnp:rootdevice") > 0 || request.indexOf("device:basic:1") > 0) {`

That's all there is to it. Hope this will fix the problem for you like it did for Master-P96!

I will work hard to get the new version including the fix out as soon as possible :)

Have a nice weekend! :)

wcbuerste, 2018-02-10:
Hi, thank you for the fast reply. This worked like a charm, could instantly connect in the Alexa app! =)
Having some effects with a brightness that gets reseted, but will investigate when I have more time.

Ae3NerdGod, 2019-02-28:
idk if i should use this old issue or make a new one, but i think im having this same issue. nodemcu 1.0, serial monitor provides nothing useful tried latest release.bin and master, but either way the app will not find the device. tired on my phone and multiple fire hd tablets. btw i do not have an echo dot or echo, i just use the tablets which seem to control everything else perfectly ( fire TV, cheap magichome wifi bulbs) 

PS everything else ive tried works great, amazing job with this!

Aircoookie, 2019-02-28:
Hi @Ae3NerdGod , I'm afraid to tell you that WLED's Alexa support (based on Espalexa) doesn't work without a physical Echo device unfortunately. The other devices you can control probably use cloud APIs (as in you have to activate a skill and register with the light manufacturer). WLED uses the local Philips hue API, which is only supported by Echo Speakers at the moment, not by the apps themselves.

I hope that you'll have fun with WLED despite this issue and maybe we can find a solution at some point :)

Ae3NerdGod, 2019-02-28:
I understand, Do you know if using the app/tablets will work if I buy an echo? Because thats definitely worth it. Also, do you have any plans on adding Google home compatibility?

Aircoookie, 2019-02-28:
Yes, if you have an Echo, you can also control the devices via the app (even when not home)!
However, I can't promise that Amazon will keep support for the local API forever...

I'd love to add Google home/assistant support, but since that is strictly cloud-only, I don't have the resources for it at the moment (would need cloud user accounts, dedicated servers, secure connections between ESP and said server...).

Aircoookie, 2019-03-01:
As of the latest master, color support is now also functional for previous-gen Dots!