wled/WLED #5: Help with "Enhancement"

DaveDischord, 2018-02-23:
I would like to add a Status Indicator LED that works with certain states.  Something simple, such as:
LED Blinking when either not connected to the defined WiFi or Unconnected in AP-mode.
LED Solid when connected to defined Wifi or there's a connection to the AP AND RGB(W) lights are on.
LED OFF when RGB(W) lights are off.

I'm sifting through the code, and have gotten some additions in place, but I'm sure there's either an easier way, or I'm missing a few things, such as, where can I find a flag that something is connected to the AP, and also, a flag that the RGB(W) lights are on/off, so that this 'Status Indicator' can toggle with both hardware button and the web interface power button?

Aircoookie, 2018-02-23:
That's a nice addition! Might add something similar officially, although it's not a top priority now.

Getting the state of the RGB lights is simple. There is a global byte variable called `bri`. It contains the master brightness (which you set with the yellow slider). If the RGB leds are off, its value is 0.

Finding out the connection to the defined wifi is not too hard either. If  this: `(WiFi.status() != WL_CONNECTED)` is true, you are connected, otherwise not.

Finally, to see if someone is connected to the AP, you could try and use the method `wifi_softap_get_station_num()`, it should return the number of clients connected to the ESPs AP, however I never used or tested whether it actually works.

Hope it's easy to implement the features you like and let me know if I can do anything else to help :)

Aircoookie, 2018-04-10:
Closing since not an issue with WLED and no further discussion required for now. Reopen if necessary :)

commit 78098e61 (2020-11-19): Fix bug where messages sent to the group topic were incorrectly ignored.  Fixes #5
closed 2018-04-10 by Aircoookie