wled/WLED #6: Single yellow led on startup

StormPie, 2018-03-13:
Hey,

First thing - thank you, I already really enjoyed ws2182fx and this implementation is awesome, so many cool features! I am working on creating a mobile friendly UI for it.

My question: when I power up the device, a single led will turn yellow for a second or so and then it fades into the boot preset, is this intended? If so, can it be changed?

I'm using a NodeMCU ESP8266 and 30 SK6812 Leds.

Thanks!

Aircoookie, 2018-03-13:
Hi,

thank you so much for enjoying the project and even actively working on it!

Let me explain the LED:
It is in fact not intended behavior. The ESP8266 uses the UART bus to control the LEDs (thats why they need to be connected to gpio2). This is faster and more reliable than controlling them in software. However, on boot, the ESP's underlying OS (FreeRTOS, I believe) sends some random (debug?) data to that UART bus, which can cause some LEDs to turn on to random colors. On one of my lights running WLED, all lights turn white briefly, on another, just the first one turns green and on another, everything stays off. I don't believe that behavior can be changed.

One thing that's possible (which I consider) is turning the LEDs on earlier in the setup process, which currently works something like:
- Load all user settings from memory (takes a very short time)
- Connect Wifi and initialize interfaces (takes several seconds)
- Turn on LEDs
If I'd start the LEDs before WiFi initialization however, effects and fading transitions could severely stutter while the ESP is busy starting stuff up. Nevertheless, I will try it out and if it works out, add it to the 0.6.0 release that will be available in a week or so.

Thank you for working on a mobile friendly UI! It would be really cool if you could share it once you've got stuff working nicely. Let me know if you have a question or if I could help with something else!

StormPie, 2018-03-13:
Okay yeah awesome, hopefully moving the LED setup earlier in the script could do the trick! If worst comes to worse, I could just add in a relay that blocks all signal to the lights until it's ready.

Once the mobile UI is all looking nice and working well I will be glad to share it 😄at the moment I am mainly focusing on making an easy to use controller for actively playing with the lights, there'd be a stack more to do in /settings, I may have to wait until I have more free time to look into making them mobile friendly!

Aircoookie, 2018-03-15:
Hey, version 0.6.0 with the improved boot LED behavior is released now. Hope it works nicely for you! Sounds great what you are working on 😄 Thank you, I'll stay tuned!

StormPie, 2018-03-18:
Just wanted to reopen this as technically there is still this yellow LED on startup, I've been looking into things like https://github.com/esp8266/Arduino/issues/3047 which seem to have a solution for the debug data.

StormPie, 2018-03-18:
I've looked through for a while and can't find a way to change the output pin, could you help me out with that? 😄

Aircoookie, 2018-03-18:
On ESP8266, there are two (proper) ways to control the LEDs, UART and DMA. UART only works on GPIO2 and DMA only works on GPIO3 (at least with the NeoPixelBus library I use). Both however seem to cause the LED at startup/flash. See this: https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%2310
You could try DMA however. I never did since UART worked fine for me (except the statup LED😄)
For that you just need to connect the LEDs to GPIO3 and in `WS2812FX.h` replace the `NeoEsp8266Uart800KbpsMethod` with `NeoEsp8266Dma800KbpsMethod`. I'd suspect that doing so won't help but it's maybe something worth trying. Sorry that I can't change anything in the code which would prevent the startup behavior😅

StormPie, 2018-04-16:
I have managed to solve this problem by using a level shifter that has the ability to be turned on after the esp has booted which stops the junk data reaching the lights.

closed 2018-04-16 by StormPie