wled/WLED #7: Speed changing during setup

StormPie, 2018-04-05:
Hey!

Still loving this project and still working on an API based HTML user interface 😄 

Recently I was working on adding the presets, I've noticed that when the device now boots up with a fancy effect instead of a static colour, the effect seems to lag a bit and freezes for a moment during the first few seconds then goes to working normally. Is this just the hardware lagging when the WiFi connects or something?

Also it seems you can not update Preset 0, which is fine because you can change the boot preset to another number and update that. Just thought I'd note it as I thought I was doing something wrong for a while 😄

Thanks again! Do you have a contact email so I can send you the HTML UI I have been working on? I'm not very experienced with contributing via GitHub itself.

Aircoookie, 2018-04-05:
Hey!

Awesome that you love the project so much! I'm really looking forward to your upcoming UI project 😄

It is normal that the effects stutter on boot (or even for a short time when sending a web request). This is because of the ESP being busy with other stuff immediately after the WiFi is connected (initialising Webserver, starting interfaces). I can attempt to mitigate it by refreshing the strip a few times between initialization steps, but I can't tell you yet whether that will help in any way. Either way, I can't think 
of an use case where it would be a huge dealbreaker since you normally don't reboot the ESP often.

Preset 0 does currently not support saving since it is the default preset on boot. To be honest, I implemented it this way because it was easier at the time.  It would be no problem to change it so that saving preset 0 would be the same as saving the LED settings with the "Ignore and use current color..." option enabled, which would improve the experience, I believe.

You're welcome! I'm so glad you like the program!
Thanks for offering to send me your work! 😄 You can contact me via `dev.aircoookie@gmail.com`
Have a great weekend!

aenertia, 2019-10-20:
Yeah - i'm looking for some boot speed optimizations. I'm using the project for permanently inplace light strips around the house/kitchen tied to wall switches initial boot up is a noticable delay to last pre-set. 

I wonder if maybe something for hooking a burst over the pixel bus of the last preset using the ULP or something might mitigate this?

I guess this is a general - better boot speed to last preset from cold state feature request.

closed 2020-04-05 by stale[bot]

jenshenk, 2021-08-10:
First of all I want to get in line for further praises of this great project, thanks so much for all the effort that was put into it! I originally bought the SP108 before I became aware of this great project, yet it created a lot of headaches. I had a few spare ESP01Ses lying around and for the 5v vs 3v problem I found this neat little board for less than a euro on aliexpress where you just click your esp into and that takes care of the 5v. 

Since this setup now is already so great and striving for perfection is always fun I was wondering whether there is still work put into getting rid of the freeze during initialization issue... I want to put the whole setup behind an existing relay setup so it can easily be integrated in my home-automation and does not have always to be turned on. Since SP108 uses pretty much the same chip yet does not show a freeze during bootup there must be a way to do it. Looking at photos of it, it also seems the sp108 also does everything with one esp without any daughter board/chip lightening the load.