wled/WLED #8: Does not work with my WS2812B...

schroepfer, 2018-04-07:
I installed the binary, compiled your source myself and wled doesn‘t work with my 60 led WS2812B.
Only randomly flickering some leds.
I have a new NodeMCU, use a levelshifter on GPIO2.
The NodeMCU works with other programs, i messured the GPIO and the levelshifter.
Also my leds works with other programs....
The WS2812B has a separate 5V/4A Supply!

Regards
Jürgen

Aircoookie, 2018-04-07:
Hi Jürgen,

thank you for trying out my project! I'm sorry that it didn't work for you right away.
Your issue seems strange considering that your setup seems to work with other programs.

Just some general suggestions (these may sound stupid, sorry, but they may be able to help you):
- Please double check that the Ground pin of your NodeMCU is properly connected to the Ground of the WS2812B!! This is the most common cause of such random flicker when using external power supply.
- Please make sure you are in the default effect setting 0 (static) which should light all LEDs in a solid color (you can change this with the star icon in the UI)
- If you have a multimeter you can measure if your LEDs actually recieve close to 5V power. Because you have a good power supply, this shouldn't be the issue though...
- You can also try and omit the levelshifter (some levelshifters are too slow for the WS2812B protocol). From my experience connecting NodeMCU to LEDs directly without levelshifter works 99% of the time.

I hope I'll be able to help you use my software!

Best regards

schroepfer, 2018-04-08:
Thank you for your fast reply!
All points are ok, but my China-Levelshifter was the problem! It seems, that it is too slow.
Without Levelshifter it works, but after some changes in your programm some leds in my stripe beginns to flicker ore don't reacts anymore.
Do you have a tip for building my own, fast and working levelshifter? (I'm an electronican)

Regards
Jürgen

wcbuerste, 2018-04-08:
Hi Jürgen,

A while ago, I was running the Adafruit Levelshifter with around 200 LEDs without any problems on an Arduino and Raspberry Pi:

https://www.adafruit.com/product/757

Cheers, David

Aircoookie, 2018-04-08:
@schroepfer  I doubt that it would be feasible to build/design your own levelshifter. If you'd want to do that, I'm sorry that I have no idea how to help with that. However, @wcbuerste 's levelshifter seems very suitable for the task, thanks!
You've also mentioned that you experienced flickering again after changing the code. Does this mean that the stock code without the changes doesn't produce the flicker? If so, I'd be happy to help you find out which of your changes to the source led to the unwanted behavior!
Regards

schroepfer, 2018-04-08:
I think, there is a misunderstanding!
I didn’t change the source, i only changed the colors of the leds in the gui!

Aircoookie, 2018-04-08:
Ah ok, so I think you experience the flickering during the gradient transition from one color to another. Interestingly, i've got 10+ ESPs running WLED and 3 or so have stutter and flicker during the transition, but the others are silk smooth. Thank you for telling me, I'll look into trying to fix it!

Aircoookie, 2019-10-25:
Closing. If you still have related issues, please discuss in #230!

closed 2019-10-25 by Aircoookie