wled/WLED #9: Analog clock?

schroepfer, 2018-04-08:
You wrote:
_Features: (V0.6.2)
.......
NTP and configurable analog clock function_

NTP settings work, but where i can find the analog clock function?

Aircoookie, 2018-04-08:
You can find the Analog Clock function in the "Time & Macros" settings menu. Under the NTP time, there is a "Clock Overlay" dropdown menu. Select Analog Clock. Then you can also specify the LED range to display the clock and the LED where 12 o' clock should be. If you run into the issue that your clock goes anticlockwise, go to LED settings, scroll all the way down and on the bottom, select "Reverse LED order". Hope the function works well for you!

schroepfer, 2018-04-10:
I found it!
But: It works, but it blocks the web-interface. It seems, that this function blocks all other.
To stop this setup, i must erase the flash with esptool.

Aircoookie, 2018-04-10:
Uh oh this is terrible! This problem seems to cause the web-interface to never work with a certain settings configuration! Which means you can't fix it by a simple reflash because the settings are retained in flash.

Unfortunately I can't reproduce it. Enabling AP also doesn't cause it for me. Can you please send me your settings before enabling the clock? Then I can see if I can reproduce it. Also I took a quick look at the clock code and there is absolutely nothing in there that should cause such a severe and permanent issue. Thank you so much for discovering this problem and I'm sorry it's such a hassle.

Aircoookie, 2018-04-10:
By the way, I really need to add a settings dump feature (excluding WiFi creds) to make it easier for you and others to send a config that causes problems. Sorry!

schroepfer, 2018-04-10:
i only changed the number of leds to 60 and my wifi setting.