wled/WLED #11: WLED doesn't connect to my router

schroepfer, 2018-04-10:
I compiled your source WLED 0.5dev (build 1801102) for my NodeMCU, connect to the WLED-AP access-point and typed in my wifi-setup.
The NodeMCU doesn't connect to my Fritz!Box 7490.
The Wifi-Setup of WLED shows: "Client IP: Not connected ".
I als tested static IP. No connection.
Now i erased the flash with esptool and flashed WLED_0.6.2_ESP8266_RGB.bin.
Connect to WLED-AP, clicked on "Connect the module to your local WiFi here!", typed in my SSID and password, saved, reboot. 
No connection to router.
I have 6 other NodeMCU and Wemos D1 mini pro in my homenet and all work fine, but not with your WLED-Software.

Aircoookie, 2018-04-10:
This is very weird. Maybe the software has set incorrect default values after the flash erase. Can you confirm for me that the "Try connecting before opening AP for" option in the WiFi settings just above the Client IP is set to at least 30 seconds (32 should be default)?

Aircoookie, 2018-04-10:
Ok, I need to say I'm quite stumped on this one, sorry. A Fritzbox (6490) is also used by me without problems for >10 ESPs running WLED. I'll think about it some more. However, I have no idea what causes this... Judging from the previous issues you posted, you had it working before. Did you use a different version of WLED for that? Only thing I can suggest is to reboot your router and make sure there is no typo in the SSID or password (i bet you already double-checked that though), but I think it's really unlikely that is the problem... Again, sorry that you run into all these issues attempting to use my program :(

schroepfer, 2018-04-10:
I tried it with V 0.5 and 0.6.2. I will try it again later.... Thank you!

Aircoookie, 2018-04-10:
Thank you for the info!
I just remembered a peculiarity with the ESP8266 that might cause or amplify the effect this issue has. After you uploaded new firmware via USB (OTA is unaffected) using e.g. the Arduino IDE or esptool, the ESP can't reboot. Once you enter the WiFi data and save, it attempts to reset, but can't do that. You'd need to disconnect it from power once and plug it back in after uploading new firmware via USB for those self-resets to work.
However, this still doesn't explain why the connection doesn't work properly with your Repeater. Only thing I could think of is the repeater having a separate firewall or a different subnet than your browser. However I don't think this is the case since most home repeaters only act as a WiFi AccessPoint.

schroepfer, 2018-04-11:
I always press the reset-button or unplug power after uploading code to the ESP.
The connection doesn't work properly with my Repeater and router. (Connection sometimes, but timeout after access in browser)
I tested a static IP in WLED setup as well. No success.
No logical behavior. I think, i give up.

wcbuerste, 2018-04-11:
Hi,

did you tested it with another ESP. Sounds a bit like this one might have a (hardware) failure?

Cheers, David

schroepfer, 2018-04-11:
I tested it with a new WEMOS D1 mini Pro and a new NodeMCU.

wcbuerste, 2018-04-11:
Hmm… that´s bad. I´m using the Adafruit Feather HUZZAH ESP8266 without any connection or freezing issues at all: https://learn.adafruit.com/adafruit-feather-huzzah-esp8266?view=all
Have wiped, flashed, updated… it a few times and still working.
I had some issues with a different (Arduino) project last year where it turned out to be my antivirus scanner messing up the stream during upload on USB. After disabling it, everything was fine… Strange, but that´s the only thing coming to my mind right now. 😉

Aircoookie, 2018-04-11:
I was also going to suggest trying a different ESP. However, I still cannot believe how it could not work since the WiFi connectivity is the most basic feature of WLED and it worked reliably for many years, projects and users already. Maybe it's a bug in the WiFi library that doesn't "like" some subnets or something. Again, I'm so sorry that the project doesn't work for you and that you wasted so much time trying to get it to work. :( If you are looking for a similar project, this one might be worth a try: https://github.com/toblum/McLighting . I will keep this issue open since it is unresolved and will notify you once something is changed. Just to be clear, other WiFi projects with your ESP8266 work, correct?

schroepfer, 2018-04-11:
"Just to be clear, other WiFi projects with your ESP8266 work, correct?"
Yes. I flashed ESPeasy on the same ESP's to test the hardware. It worked correct.

MopheusDG, 2018-05-29:
Christian, could this be related to the AP Channel ? I was checking the code and I saw your AP is using channel 1 (not available in all regions).

Just wondering...

Aircoookie, 2018-05-29:
As far as I know, the ESP8266/32 supports the AP on all channels 1-13 regardless of location. It is the responsibility of the user not to use any channels that are not legally allowed in their country. However, channel 1 should work in any region. If not, you could change it in the settings. However, schroepfer's issue was not with the AP, but with the connection to the home network.

You however brought me to an idea! It may be the case that the ESP can't manage a connection to a network operating on the same channel as the WLED-AP. That may be logical because of interference. I'll try to set the AP channel to the channel of my network to test it out!

zwiebelslayer, 2019-03-26:
I have the same problem with a Fritz Box 7490. WLED just doesnt connect to it, but other programs with WIFI work just fine.

sansillusion, 2019-03-26:
I had an issue at my friends place. Turned out he had set his router to a
max of 10 devices connected and the limit was reached with all our
cellphone and computers.

It could be some settings blocking connection in the router or an issue
with your specific router model.

closed 2019-10-25 by Aircoookie

kombi21251, 2020-04-04:
Hello! I have the same problem.
Changing the settings of the router helped me.
I am changed the channel of wireless network from "auto" to "1".
My router is TP-LINK AC750.

GoBooth, 2020-04-11:
I'm having the exact same problem.

I Successfully Flashed it on a Magic Home 5Channel esp 8285 and a NodeMCU 8266
in both cases, it flashes, I connect to it via AP mode, I enter my SSID "Hawaii 24" it reboots... I see the new "WLED" device on my pfsense DHCP table... but I can't ping the IP and I can't reach it via the browser.

What sucks is that REFLASHING it with the same firmware doesn't remove the configuration, I need to flash it was tasmota or something to overwrite it, and then flash it again with WLED.

I have a feeling it's the space in my SSID...
I created a new SSID on my ubiquity.. and I'm about to test it right now.

GoBooth, 2020-04-11:
OK, turns out I might have been flashing the wrong version, 
I was using WLED_0.9.1_ESP8266_1M_full.bin

Tried WLED_0.9.1_ESP8266.bin (which ironically is the one I thought I was flashing with) and it fixed the wifi issue. Wifi connects to it now.

yay!!

Stenvk, 2020-09-19:
Im still having problems with connecting to my wifi. It doesn't even show up in the event log from my router.
I flashed it with 3 different versions (WLED_0.9.1_ESP8266.bin, WLED_0.8.5_ESP8266.bin and WLED_0.10.2_ESP8266.bin) none of them fixed the issue.
I tried using a static ip, with no effect. And auto connecting doesn't work either.

Aircoookie, 2020-09-20:
@Stenvk try disabling the WiFi sleep option, it might help. If you are using a static IP, make sure to enter the gateway IP (your router's) as well. Make sure the ESP is close enough to the router. Lastly, if you are using a combined SSID for 2.4Ghz and 5Ghz, the router might only try connecting on the 5G, which is not supported by ESP8266.
You could also try installing Tasmota or ESPHome and see if that software manages to connect.
Hope you can get it to work :)

Aircoookie, 2020-10-28:
For those using Ubiquity routers and running into connection issues, [this](https://github.com/Aircoookie/WLED/issues/1293#issuecomment-717929824) might help :)

luiguii57, 2020-11-04:
Morning you all.
Was wondering if the no connection issue has been resolved.
Couple of days ago I upgraded my router to the new AT&T BGW320 and now nine of my Nodemcu are working.  The router is not assigning IP to any of them.
I have re flash all of them and still same problem.
Have spoken to AT&T tech and have been unable to find the problem
We did disabled the 5ghz leaving only the 2.4ghz able.
Tried reflashing using the last 4 version and none correct the problem
Please help as I am depending on them to work to control myself house Christmas lights
Thanks

Aircoookie, 2020-11-04:
Don't know why that could happen... sorry. Does the password contain special characters? Does the WLED-AP re-open after a few minutes or does nothing happen at all?

Stenvk, 2020-11-04:
Hi, i also had (still have but I found a workaround) this problem. The method I used to fix it is by setting up an old router as a stand alone wifi network and connecting it with an ethernet cable to my internet. I can control my leds on my normal 2.4 ghz and 5ghz without any issues now.

luiguii57, 2020-11-04:
Yes
The main password for my wifi had special characters $$
Do you think that is cresting the no connection problem?
Thanks

luiguii57, 2020-11-05:
Hi
Id it possible to send details as how to do it
Really appreciate your help
Thanks

I do have an old Netgear model LG6100D

Stenvk, 2020-11-05:
I can't give you step-by-step details but you can probably find some info online with the model you have. Its basically an acces point but with its own wifi network (so you will have your regular wifi and the wifi from the ap). (also I didn't do any research but just plugged it in configured it normally and it just so happened that I can control wled on all my wifi networks)
I hope this will help

luiguii57, 2020-11-05:
Thanks for your help
Plugged the second router and have all my nodemcu working again

Airyzz, 2021-09-17:
Hi, sorry to revive an old thread, but has anyone found an actual solution to this problem that doesnt require two routers? I have also come across this problem, running on an esp8266 (d1 mini) trying to connect to my wifi, which is also a FRITZBox 7490. I have another instance of wled running on a different board which has always worked perfectly, so I dont see why this board doesn't like it.  Flashing Tasmota to the esp as a test worked fine, and was able to connect to the internet no problem, so it must be an issue with WLED and not the board right?

klrock, 2021-12-16:
I was following this discussion as I was having very similar issues and today I found the fix for my problem.  I had noticed that when I tried to access the AP that I could not open it, but I could examine it's properties in Settings.  To my surprise, the assigned IP address was now occupied by my wifi controller.  Then I noticed that  the available wifi list showed that Npcap Loopback Adapter was my current wifi connection rather than my TP-Link router.  I think Npcap Loopback Adapter was from some network sniffer I tried in the past.  Anyway, I deleted it in Device Manager and all is fine now.  Don't know if this will help others but wanted to share.

sbuxreg, 2022-03-02:
I have a nixie tube clock with an ESP8266 on board. I have the same problems with the BGW320 i cannot get it to stay connected consistently. The only solution I was able to come up with was using an old asus router I had laying around as an AP.  The clock stays connected to the AP like a champ.