wled/WLED #15: flicker issue

pete111, 2018-04-22:
Hi @Aircoookie, I created this separate thread. I found out that sometimes my LEDs flicker on my esp8266-01 (from time to time, randomly). I created short video `https://streamable.com/6aygu` by what I mean (all yellow leds are ok, then I sent UDP to change 4 leds to red and it was ok, but when LEDs was changing to default yellow color, you can see flicker). I can't say this happens just while switching color leds to default color, because it happens randomly. For example I wanted to create new video and reproduce flicker for you but already 10 minutes without flicker. :) (I did not try it level shifter as I think esp as  3.3v device should be good without it). Just wanted to let you know this. Maybe you know why it happens. 
Video is here:

Aircoookie, 2018-04-22:
Hey, thanks for letting me know :)
I've also observed this very rarely with one of my ESPs. Around once every 8 hours or so all LEDs turn green for half a second. From your video, it appears like they light up in random colours, which is really weird. Thanks for trying to make a 10 min video for me 😄 I'll look into it but from the code there is nothing which causes this. I guess it's some low level ESP hardware stuff acting up, but I'm not sure...

Osmii, 2018-05-23:
Same problem here! Try to touch with your finger the data line between esp and leds and tell me does this help. In another project i had a lot of flickering. In the end i had to add 33 pF capacitor in series with data line...

Aircoookie, 2018-05-23:
Hi @Osmii , welcome to the project!
Thanks for your advice, trying a small cap sounds like a great idea!
Did the capacitor fix the flicker for you with WLED also, or is the problem still there?

closed 2019-10-25 by Aircoookie