wled/WLED #18: add option to disable blue wifi led

oli2344, 2018-05-14:
Hi,
have you ever considered to add the option to disable the blue status wifi led from the WebInterface?

Aircoookie, 2018-05-14:
Hi,
first of all, thank you for trying out my project!
Yes I have considered this, however there is an issue.

The build-in blue LED and the WS2812 strip are on the same pin(GPIO2/D4 on Wemos D1).
Therefore we can't turn the LED off while addressing the strip.
Also we can't just use any pin for the strip since it uses I2S protocol which only works on this pin.
The "workaround" I used was to desolder the blue LED - i know this is not a good general solution though.

There is a "proper" solution though which you could apply yourself. There is an alternate mode of driving the strip called DMA. It's said to be slightly less reliable than the UART/I2S protocol WLED uses right now, though I never tried it myself. For it to work, you need to connect the strip to pin GPIO3/RX on Wemos D1. Also you need to compile the source yourself in the Arduino IDE after making one small change to the code. Just enter the file "NpbWrapper.h" and in line 17, replace `NeoEsp8266Uart800KbpsMethod` by `NeoEsp8266Dma800KbpsMethod`.

Right now I've got a tight schedule but once I find enough time to try out this DMA method myself, I might look into optionally making it available as an option. Then it would not be a problem to implement disabling or customizing the blue status LED.

MatFl, 2018-12-04:
I think we could turn off the LED while the strip is powered off. At least for me that would be a huge improvement.

Aircoookie, 2018-12-04:
Alright, I'll try and find out whether turning the LED off while the strip is off works!

As a workaround, if you have the option to, you could also use bitbang on another pin and the LED will stay off. For me, bitbang works surprisingly well on ESP8266. You can set the LED pin in `NpbWrapper.h`.

gilshahar7, 2019-08-18:
Hey, i was wondering if you ended up including a way to turn off the blue led when the strip is off.. i am looking for the same thing but cant figure it out.

Aircoookie, 2019-10-25:
This has been requested some more recently. I might open an issue with the NeoPixelBus dev Makuna soon to ask him whether it is possible to have the bus high by default, which would largely solve this LED issue!

SeLLeRoNe, 2020-04-02:
Hi Aircookie,

First of all, thanks for your project and your effort, it is really amazing!
I would like to ask if you managed to go anywhere with this possibility and/or if you heard back from the NeoPixelBus devs.

Thanks
Andrea

Basti149, 2020-07-08:
Danke für das sehr gute Projekt. Nun wollte ich auch mal nachfragen ob es bei dem nächsten Update eine Lösung für die blaue LED gibt. Da sie Nachts sehr stark nervt. Und zustäzlich noch Strom Verbraucht.

Thanks for the very good project. Now I wanted to ask if there was a solution for the blue LED in the next update. Since she is very annoying at night. And additionally consumed electricity.

gilshahar7, 2020-07-08:
> Danke für das sehr gute Projekt. Nun wollte ich auch mal nachfragen ob es bei dem nächsten Update eine Lösung für die blaue LED gibt. Da sie Nachts sehr stark nervt. Und zustäzlich noch Strom [...]

If you have a D1 Mini, the only option is to desolder the LED from the board, works pretty well.

Legsmaniac, 2020-07-08:
That's one option but not the only option.
Another option is to recompile the source code with a different pin set as led output.  It's because GPIO2 is also the LED control pin. 
Or wait for a future upgrade which promises Web UI pin selection.

Basti149, 2020-07-08:
> 
> 
> That's one option but not the only option.
> Another option is to recompile the source code with a different pin set as led output. It's because GPIO2 is also the LED control pin.
> Or wait for a future upgrade which promises Web UI pin selection.

Danke für die schnelle Antwort. Wann kommt kommt das Update raus. Da ich eine Platine gehört und die komponenten fest verlöten möchte.

Thank you for your prompt reply. When does the update come out? Since I have heard a circuit board and want to solder the components firmly.

Basti149, 2020-07-08:
> If you have a D1 Mini, the only option is to desolder the LED from the board, works pretty well.

Danke für die schnelle Antwort. Das wollte ich eigentlich vermeiden und die LED zu zerstören. Es wäre Super wenn die blaue LED als Status verwendet werden könnte. Zum Beispiel um die Netzwerk Verbindung anzeigen zu lassen.

Thank you for your prompt reply. I actually wanted to avoid that and destroy the LED. It would be great if the blue LED could be used as a status. For example, to display the network connection.

mike2nl, 2020-10-02:
@Basti149 
->  It would be great if the blue LED could be used as a status. For example, to display the network connection.

The same is used in Tasmota to do so. WHen the esp trying to connect the wifi ap the led is blinking every time the esp is calling. when the network is connected she stays blue. But a certain command can switch off that led.

I support your idea...

Aircoookie, 2020-10-12:
LED is now off when LEDs are off thanks to #1245  :)

closed 2020-10-12 by Aircoookie

jpborjas, 2021-11-03:
I’m on. Erosion 12.0_1M and the LED stays always on

Aircoookie, 2021-11-04:
@jpborjas what is your board? Some have a permanent (not software controllable) power LED

jpborjas, 2021-11-04:
I have a Node MCU
  
  
  
  
  
  
Thanks,   
  

  
Pablo Borjas

Aircoookie, 2021-11-04:
Which of these two LEDs is not turning off for you? https://lowvoltage.github.io/2017/07/09/Onboard-LEDs-NodeMCU-Got-Two
If it is the one closer to the USB port, a solution might be setting the relay pin to GPIO 16.

jpborjas, 2021-11-04:
Is the GPIO2, the really bright one, farthest from the USB port

Aircoookie, 2021-11-04:
@jpborjas most likely the problem is that the 1M binary has a different value for `LED_BUILTIN`, so that it turns off not GPIO2, but GPIO1. Please try installing a 4M binary (e.g. `WLED_0.13.0-b4_ESP8266.bin`), for example on https://install.wled.me, note that this will reset your settings/presets.

jpborjas, 2021-11-04:
I tried that and I get errors like could not mount filesystem or something
like that. That's why I installed 1M.

jpborjas, 2021-11-08:
I realized that I was setting the size of the file to 1MB during the file
transfer. I used the link you sent me to install using that instead of the
esptool.py, installed version 13.0_b4, and everything looks great.

Thank you so much!

Pablo

shewmonphoto, 2022-08-25:
I've found a stupid simple solution to my issue with the on board LED.... It involves a tiny bit of Black electrical tape.................

glmnet, 2022-09-07:
I have ws2812 connected to gpio3 (rx) so I turned off led by setting a button to gpio 2