wled/WLED #20: Philips HUE issue ?

MopheusDG, 2018-05-29:
Hi Christian, I've been trying to get WLED work on my Wemos but I have a problem. It boots fine, I can see the AP, I can configure my WLAN, reboots, connects to my WLAN fine, but... once it's in my WLAN, as soon as I try to access the WLED via IP, the IP stops responding to pings.
If I boot the device and do not access it, works fine, pings works fine.... but as soon as I try to go the web server, goes down. AND, I notice to be having some HUE Bridge issues, is it possible we have some kind of conflic if you own a HUE Bridge in the same network ?

MopheusDG, 2018-05-29:
NVM... it's something with CHROME in particular.... tried with IE and works fine.

Aircoookie, 2018-05-29:
Don't worry, it's a known issue. I don't think it is Chrome's fault. If you send many requests in a short period of time (even only doing setup and the immediately connecting) can cause the ESP to become unresponsive. Waiting half a minute usually "fixes" it.

What do you mean by "Hue bridge issues"? Have you managed to sync WLED to HUE lights? If yes, that may make the web server unresponsiveness issue worse, because the ESP has to send a request to the HUE bridge every few seconds to check if the bulb color was changed.

MopheusDG, 2018-05-29:
No, to be honest, I stopped trying WLED, I wasn't able to get my Wemos X-Ring to work with the library for some reason... only full white colors. I was able to access via IE but couldn't change any color, and when I went to settings, nothing was showing on that configuration page. The only thing I manage to do is set the device in my network by forcing my AP and PASS in the .ino file.

I'll try again later this week, didn't have much time for testing honestly.

Aircoookie, 2018-05-30:
That's a bummer, but at least you got Espalexa working nicely for you!

Once you'd like to give it another try, here are a few tips:

You should never be required to set the network in the .ino. The Wemos should always open a setup AP "WLED-AP" when fresh-installing WLED. If that doesn't work, something is majorly wrong. Either the code was uploaded incorrectly, or, more likely, the settings memory (EEPROM) is corrupted. You could try opening the `http://<ESP-IP>/cleareeprom` page to reset it. Most likely unfortunately it is nothing of the above and either your ESP is somehow defective or there probably is a major bug in my software that only happens with some modules.

Also, I did a quick search for the X-Ring. Do you have a standard WS2812B ring or the round black one with the Wemos in the middle? If that's the case, please make sure that the small DIP switches in the middle are set so the LEDs are connected to D4 (ESP8266 GPIO2). The LED's can't work on any other port.

Lastly, try opening this in a browser: "http://<ESP-IP>/win&A=255&R=255&G=0&B=0". It is an API call that should set the LEDs to red. The ESP should respond with some XML values. If you see the values but the LEDs are not red, the problem is probably with the LED connection. If the ESP doesn't respond at all, there is some other hardware or software error...

I'm so sorry that WLED doesn't work for you right now!

Aircoookie, 2019-03-27:
Closing old issues which may have been resolved by updates. Please reopen if the issue persists using the latest version 0.8.4 (or open a new issue)!

closed 2019-03-27 by Aircoookie