wled/WLED #22: effect cycling once clicked on the next effect right arrow on effect 57

pete111, 2018-06-10:
Hi @Aircoookie, I found out once you reach 57th effect and click on the right arrow again (it should go to 1st effect), it goes to 53rd effect and then step by step to 57th - so it is not going from start 1st effect once right button is clicked on the effect No 57.

Aircoookie, 2018-06-10:
Hi Pete,
thanks for the feedback! I didn't notice it before, but I know why it happens now and can fix it.
However, I'm not yet sure how exactly I should set it. I have three options:
- Make it go to Effect 0 (no effect, solid color). Normally you'd use the cross icon for that
- Make it go to Effect 1 (blink)
- Make it do nothing (stay at effect 57)

All of these would be logical behavior. I'm considering the 3rd option since the left arrow also does nothing when at effect 0 (solid color).

Which one do you prefer?

pete111, 2018-06-10:
Hi @Aircoookie, I thing 3rd option is pretty ok :)

Aircoookie, 2018-06-10:
Alright, this small change will be added in the next update :)

Aircoookie, 2018-06-23:
Added in 0.7.0!

closed 2018-06-23 by Aircoookie