wled/WLED #23: Philips Hue Sync not working

EdikHa, 2018-06-12:
Hey, 
somehow I'm not able to connect my bridge with WLED. 
IP Adress of the bridge is right and the bridge is working with 6 lamps. 
I've pressed the center button on the bridge and saved the settings page, but can't see any results from it.
The hue status underneath is still saying "inactive" and I can't see any added "lamps" in the hue app.

Am I doing something wrong?
Everything else is working fine

- New version of bridge is used
- WLED version 0.6.4
- Using 5 x esp8266 (NodeMCU's) with RGBW LEDS (Sk6812)

Aircoookie, 2018-06-12:
Hey,
great you are giving WLED a try and most of it  works fine!

Because of limits in the hue API, the WLED light is not individually controllable in the hue app like a regular hue bulb, unfortunately. It can however sync its color to one of your six hue lamps.

You did the setup almost correctly (I really need to add more documentation for this)!

- First, go to the About section in the hue app settings. You'll see a list of the hue lamps. Now you need the number of the light you want to sync to. The number is on the left of each bulbs name, respectively.
- In the WLED Sync settings, find the line "Poll hue light x every y milliseconds". For x, enter the number of the bulb. With y, you can set the update rate (I recommend 1200ms). Then, tick the checkbox.
- Press the center button on the bridge again, then press Save in the settings!

Remember that if you got your ESPs synced with the WLED notifier, you only need to setup one light to get the color from the bridge. Alternatively, you can sync some lights to different hue bulbs by turning of the WLED notification send feature for then.

Hope I was able to help you :)

EdikHa, 2018-06-12:
Now I understood what the "Poll hue light" is doing. 
I've tried it and it worked. 
Thank you.

closed 2018-06-12 by EdikHa

osfersa, 2021-09-10:
Hi,
On the new app for Hue, we can't locate the bulb number, any other way.
Thank you.

Aircoookie, 2021-09-11:
@osfersa hey, please see https://github.com/Aircoookie/WLED/issues/2143#issuecomment-901456911 for a temporary workaround if you have android. Might be something similar out there for iOS too.

osfersa, 2021-09-12:
> @osfersa hey, please see #2143 (comment) for a temporary workaround if you have android. Might be something similar out there for iOS too.

Yes I’m sorry I saw it right after I sent that message.
Got it working using Android app hue configuration

ozett, 2021-10-18:
get get the hue-light number it could also be done in node-red 
[image]

```
[{"id":"46c25efc.09f68","type":"inject","z":"ff2177ca.cecc08","name":"fetch.lights","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"{\"fetch\":\"lights\"}","payloadType":"json","x":840,"y":170,"wires":[["8863c57a.898168"]]},{"id":"8863c57a.898168","type":"hue-bridge-node","z":"ff2177ca.cecc08","name":"hue","bridge":"6753eba7.fd6634","autoupdates":true,"skipglobalevents":false,"x":1040,"y":120,"wires":[["afa9fbe3.d5e9f8"]]},{"id":"afa9fbe3.d5e9f8","type":"debug","z":"ff2177ca.cecc08","name":"dbg.hue.bridge","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1250,"y":120,"wires":[]},{"id":"6753eba7.fd6634","type":"hue-bridge","name":"Philips hue-14-236","bridge":"192.168.14.236","key":"62e4rjUpGtBwXfby0B0I0DsumngmfBO5inR6xM7J","interval":"2000"}]
```