wled/WLED #24: connec to wifi after "global" power cycle

pete111, 2018-06-13:
Hi @Aircoookie ,
I found out, if I power cycle router and Wled at the same time, Wled boots up much faster than router but after that Wled is not trying to connect to wifi once available (once router is fully booted up). It is just minot issue, so I just wanted to let you know. :)

Aircoookie, 2018-06-13:
Hi Pete,

this is normal for now :)

On boot, WLED tries to connect to the router. If that's unsuccessful (because the router has not yet booted up yet), after a set timeout, it will open its "WLED-AP" access point. You can set this timeout in the Wifi settings "Try connecting before opening AP for x s". A high number (like >100 seconds) will give your router enough time to boot up, but also means you'll have to wait that long for the access point to open if you want to use WLED without your home network. However, if WLED manages to connect to your router on boot and your router goes offline after that, the ESP automatically reconnects.

I know this is not how most WiFi devices behave and it's already on my list to improve this behavior ("if WiFi is configured, but not connected, and AP is on, then try to connect every 30 seconds" or something like that)

Thank you :)

Aircoookie, 2019-10-25:
This should now work in latest master! Releasing 0.8.6 today or tomorrow!

closed 2019-10-25 by Aircoookie