wled/WLED #25: led stopped responding to UDP then started again

pete111, 2018-06-19:
Hi @Aircoookie 
currently I found out that for a while (few seconds) WLED (on my esp8266-01) stopped responding to any UDP incoming data (it switched to default color) and after some time (few seconds) it started to respond to UDP again. Any idea why this happens?

Aircoookie, 2018-06-19:
Hey Pete,
please tell me what the UDP source was. A certain Hyperion effect?
Does it happen directly after boot or only after some minutes the ESP is already running?
How many seconds approximately does it stay on the default color and is this time always the same?

The normal behavior is that WLED will go back to the default color 5 seconds after it hasn't received a new UDP packet. So if you can be sure that your UDP application sends UDP more often than 5sec, there must be something wrong. Also, do you have the WLED HTTP UI open while this happens? Having the UI open means there are less resources for UDP processing which might lead to the issue.

Sorry that I'm asking so many questions! I need to make sure I know the exact issue before I can look into it :) Thank you!

pete111, 2018-06-19:
Hi @Aircoookie,
thank you for the prompt reply. It was during hyperion grabbed pictures and sent UDP data to Wled. 

It happens some time after Wled is running (many minutes, e.g. after 45 minutes, time is not constant), it goes to default color for (I think few seconds - but I will be able to tell to exactly how many seconds, because I am preparing video for you - I will update it a bit later).
It happend in the last 10 minutes 2times, so maybe it is really thing of lost packets in network or hyperion did not send data - strange, because it is processing frames continuously (?)

Also, do you have the WLED HTTP UI open while this happens? Having the UI open means there are less resources for UDP processing which might lead to the issue. -- Honestly, don't understand exactly what you are asking me, I tried to open HTTP UI while this happens but it gave me just information "The WLED UI is not available while receiving real-time data."

Aircoookie, 2018-06-19:
Thank you! I believe the ESP becomes unresponsive after many minutes so it may not receive UDP for a few seconds. If you have some moving content on the screengrabber you should notice that WLED no longer syncs and displays a frozen frame for 5 seconds before returning to default color. If that's not the case, it might help for me to increase the timeout a bit (maybe 15sec).
Because it happens so rarely (only after long time and only 1 time every ~5min) makes it hard to debug of course.

About the HTTP UI, sorry for the confusion. It is possible for you to load the UI and then start Hyperion. In that case, the ESP will continue to communicate with the UI as long as the browser tab is open, which might have caused it.

pete111, 2018-06-20:
Hi @Aircoookie , sorry for my delay, I wanted to give you better test results. Just right now after few hours esp restarted itself completely - I left running fast rainbow effect many hours and now esp restarted (first of all it froze for one minute, then reset occurred, clock are running from start, default color shown, then it accepted udp packets again - I made video for it [https://streamable.com/wum92](https://streamable.com/wum92))

I can make extensive test also for previous scenario from yesterday and make video for it, if you want.

Aircoookie, 2018-06-20:
Hi Pete,
no problem, thanks for the time you take to make those extensive tests!
I'll see what I can do to find a solution. Probably the ESP is overloaded when it receives another random packet at the same time of the Hyperion UDP, but I'm not sure... I have to think about a solution. Thanks for your help with the testing and your patience! :)

pete111, 2018-06-20:
Hi @Aircoookie , I am always glad to help. Update for you - I had my wled turned off (with top left on/off button in HTTP GUI and no udp communication was initiatet nor GUI opened in browser) but suddenly my wled restarted itself and went back to default color. So no idea why, but wled restarted itself.

closed 2020-01-07 by Aircoookie