wled/WLED #26: "White" color correction with different types of RGBW strips (WW,NW,CW) and using white channel for luminance

monarchit, 2018-06-23:
I am unable to correctly display the white color properly as my strips are RGB+WW (Warm White), It ends up looking very Yellow/Orange. It would be great to set a default white by mixing white and RGB to create a default white color to be displayed on the effects (even for Ambilight).

Color temperature correction and luminance calculation (if not yet implemented) would make a great difference in color accuracy!

This is an example with FastLed if it helps.

`  // Getting luminance from value to add white leds
  // https://stackoverflow.com/questions/40312216/converting-rgb-to-rgbw
  if (W_ON_XY == true) {
    float tM = MAX(ri, (MAX(gi, bi)));
  
    if(tM == 0) {
      rgbw[0] = 0; rgbw[1] = 0; rgbw[2] = 0; rgbw[3] = 0;
    }
  
    // Figure out what the colour with 100% hue is
    float multiplier = 255.0f / tM;
    float hR = ri * multiplier;
    float hG = gi * multiplier;
    float hB = bi * multiplier;
  
    // Calculate whiteness of color
    float M = MAX(hR, MAX(hG,hB));
    float m = MIN(hR, MIN(hG,hB));
    float Luminance = ((M+m) / 2.0f - 127.5f) * (255.0f/127.5f) / multiplier;
    w = Luminance < 0.0f ? 0.0f : Luminance;
    w = Luminance > 255.0f ? 255.0f : Luminance;
  } else {
    w = 0.0f;
  }

  rgbw[0] = (int) (r * optimal_bri); rgbw[1] = (int) (g * optimal_bri); rgbw[2] = (int) (b * optimal_bri); rgbw[3] = (int) ((w * bri) /3);
}

void convert_ct() {

  int optimal_bri = int( 10 + bri / 1.04);
  int hectemp = 10000 / ct;
  int r, g, b;
  if (hectemp <= 66) {
    r = 255;
    g = 99.4708025861 * log(hectemp) - 161.1195681661;
    b = hectemp <= 19 ? 0 : (138.5177312231 * log(hectemp - 10) - 305.0447927307);
  } else {
    r = 329.698727446 * pow(hectemp - 60, -0.1332047592);
    g = 288.1221695283 * pow(hectemp - 60, -0.0755148492);
    b = 255;
  }
  float w = 0;
  uint8 percent_warm = ((ct - 150) * 100) / 350;

  w = ((optimal_bri * (100 - percent_warm)) / 100) / 3;

  w = w > 255 ? 255 : w;
  r = r > 255 ? 255 : r;
  g = g > 255 ? 255 : g;
  b = b > 255 ? 255 : b;
  rgbw[0] = r * (bri / 255.0f); rgbw[1] = g * (bri / 255.0f); rgbw[2] = b * (bri / 255.0f); rgbw[3] = w;
}`

// Add cold white LEDs according to luminance/ whiteness in xy color selector
`
    float tM = MAX(ri, (MAX(gi, bi)));
  
    if(tM == 0) {
      rgbw[0] = 0; rgbw[1] = 0; rgbw[2] = 0; rgbw[3] = 0;
    }
  
    // Figure out what the colour with 100% hue is
    float multiplier = 255.0f / tM;
    float hR = ri * multiplier;
    float hG = gi * multiplier;
    float hB = bi * multiplier;
  
    // Calculate whiteness of color
    float M = MAX(hR, MAX(hG,hB));
    float m = MIN(hR, MIN(hG,hB));
    float Luminance = ((M+m) / 2.0f - 127.5f) * (255.0f/127.5f) / multiplier;
    w = Luminance < 0.0f ? 0.0f : Luminance;
    w = Luminance > 255.0f ? 255.0f : Luminance;
  } else {
    w = 0.0f;
  }

  rgbw[0] = (int) (r * optimal_bri); rgbw[1] = (int) (g * optimal_bri); rgbw[2] = (int) (b * optimal_bri); rgbw[3] = (int) ((w * bri) /3);
}`

`
void convert_xy()
{
  int optimal_bri = bri;
  if (optimal_bri < 5) {
    optimal_bri = 5;
  }
  float Y = y;
  float X = x;
  float Z = 1.0f - x - y;

  // sRGB D65 conversion
  float r =  X * 3.2406f - Y * 1.5372f - Z * 0.4986f;
  float g = -X * 0.9689f + Y * 1.8758f + Z * 0.0415f;
  float b =  X * 0.0557f - Y * 0.2040f + Z * 1.0570f;

//  // Apply gamma correction v.2
//  // Altering exponents at end can create different gamma curves
//  r = r <= 0.04045f ? r / 12.92f : pow((r + 0.055f) / (1.0f + 0.055f), 2.4f);
//  g = g <= 0.04045f ? g / 12.92f : pow((g + 0.055f) / (1.0f + 0.055f), 2.4f);
//  b = b <= 0.04045f ? b / 12.92f : pow((b + 0.055f) / (1.0f + 0.055f), 2.4f);

  // Apply gamma correction v.1 (better color accuracy), try this first!
  r = r <= 0.0031308f ? 12.92f * r : (1.0f + 0.055f) * pow(r, (1.0f / 2.4f)) - 0.055f;
  g = g <= 0.0031308f ? 12.92f * g : (1.0f + 0.055f) * pow(g, (1.0f / 2.4f)) - 0.055f;
  b = b <= 0.0031308f ? 12.92f * b : (1.0f + 0.055f) * pow(b, (1.0f / 2.4f)) - 0.055f;

  if (r > b && r > g) {
    // red is biggest
    if (r > 1.0f) {
      g = g / r;
      b = b / r;
      r = 1.0f;
    }
  }
  else if (g > b && g > r) {
    // green is biggest
    if (g > 1.0f) {
      r = r / g;
      b = b / g;
      g = 1.0f;
    }
  }
  else if (b > r && b > g) {
    // blue is biggest
    if (b > 1.0f) {
      r = r / b;
      g = g / b;
      b = 1.0f;
    }
  }

  r = r < 0 ? 0 : r;
  g = g < 0 ? 0 : g;
  b = b < 0 ? 0 : b;

  float ri = r;
  float gi = g;
  float bi = b;
  ri = ri > 1.0f ? 1.0f : ri;
  gi = gi > 1.0f ? 1.0f : gi;
  bi = bi > 1.0f ? 1.0f : bi;
  ri = ri < 0.0f ? 0.0f : ri;
  gi = gi < 0.0f ? 0.0f : gi;
  bi = bi < 0.0f ? 0.0f : bi;
  float w = 0.0f;

  // Getting luminance from value to add white leds
  // https://stackoverflow.com/questions/40312216/converting-rgb-to-rgbw
  if (W_ON_XY == true) {
    float tM = MAX(ri, (MAX(gi, bi)));
  
    if(tM == 0) {
      rgbw[0] = 0; rgbw[1] = 0; rgbw[2] = 0; rgbw[3] = 0;
    }
  
    // Figure out what the colour with 100% hue is
    float multiplier = 255.0f / tM;
    float hR = ri * multiplier;
    float hG = gi * multiplier;
    float hB = bi * multiplier;
  
    // Calculate whiteness of color
    float M = MAX(hR, MAX(hG,hB));
    float m = MIN(hR, MIN(hG,hB));
    float Luminance = ((M+m) / 2.0f - 127.5f) * (255.0f/127.5f) / multiplier;
    w = Luminance < 0.0f ? 0.0f : Luminance;
    w = Luminance > 255.0f ? 255.0f : Luminance;
  } else {
    w = 0.0f;
  }

  rgbw[0] = (int) (r * optimal_bri); rgbw[1] = (int) (g * optimal_bri); rgbw[2] = (int) (b * optimal_bri); rgbw[3] = (int) ((w * bri) /3);
}`

`void convert_ct() {

  int optimal_bri = int( 10 + bri / 1.04);
  int hectemp = 10000 / ct;
  int r, g, b;
  if (hectemp <= 66) {
    r = 255;
    g = 99.4708025861 * log(hectemp) - 161.1195681661;
    b = hectemp <= 19 ? 0 : (138.5177312231 * log(hectemp - 10) - 305.0447927307);
  } else {
    r = 329.698727446 * pow(hectemp - 60, -0.1332047592);
    g = 288.1221695283 * pow(hectemp - 60, -0.0755148492);
    b = 255;
  }
  float w = 0;
  uint8 percent_warm = ((ct - 150) * 100) / 350;

  w = ((optimal_bri * (100 - percent_warm)) / 100) / 3;

  w = w > 255 ? 255 : w;
  r = r > 255 ? 255 : r;
  g = g > 255 ? 255 : g;
  b = b > 255 ? 255 : b;
  rgbw[0] = r * (bri / 255.0f); rgbw[1] = g * (bri / 255.0f); rgbw[2] = b * (bri / 255.0f); rgbw[3] = w;
}`

Aircoookie, 2018-06-23:
Hi,

thanks for trying out the project and the example code!
I've added the feature you suggested in the latest version 0.7.0 that was just released! Go to LED settings and check the "Auto-calculate white from RGB instead" option. Please check if this solution satisfies your use case. I'd be happy to hear some feedback!

I don't know if it is a good idea to add color correction for the different white shades right now, but I'll consider it! (I see the white source as a source of bright ambient light and you make the choice of it's color temperature when buying WW, NW or CW). For ambilight, for now only the RGB component is supported, but that will be changed soon. I'll definitely try getting the example code to run and try out its performance! Thank you!

monarchit, 2018-06-24:
Thanks so much for the update.

I have been testing this and there seems to be an issue when I enable an effect through the desktop UI while using the new mobile UI I can't seem to disable this effect anymore through the mobile UI until I
reload the page. Also when I set an effect on the mobile UI I can't seem to stop it and go back to normal color pick mode. I am unsure on how to test for the new white rendering option.

I was unable to use Ambilight at all with the default settings. I am using a Wemos Mini ESP8266 and the light were incredibly slow to respond to anything. To fix this and flickering issues I had to use the FastLED Library's #define FASTLED_ALLOW_INTERRUPTS 0 and increase the Serial.begin baud rate to 230400 which gave me butter smooth performance using psieg's Prismatik version. This version supports some kind of color temperature adjustment but it messes with some of the colors in my case when I use RGBW and Adalight on my own version of Ambilight. I was able to obtain a decent white level even if low brightness but my yellows looked greenish.

This solved for me, but I guess it can be implemented differently if not yet implemented.
https://github.com/FastLED/FastLED/wiki/Interrupt-problems

Regarding the white point definition, it's true that light temperatures are a choice of ambient light but as it is it is now it has a limited range of functionalities. For example, there could be a feature like LightFX and PhillipsHue that allows the user to pick a different color temperature and set a mood, they seem to do this beautifully by mixing a little bit of blue to make white lights and vice-versa with red/yellow. White light is good for concentration and Warm white is good for lounge areas and so on. Calibrating the white could allow very bright and correct ambient light with a kelvin picker added on top of the color picker.

Also once RGBW is integrated on Adalight it could allow for better calibration of screen colors to match the strips.

I have managed to successfully add a multistrip RGBW Adalight using a modified version of Fasted that supports RGBW. using the following:

https://gist.github.com/baaldfg/557a708b322a9157c33cd3695c8b1154
+
https://github.com/coryking/FastLED  (This supports some kind of subtraction white calculation) 

subtract_min:
M=min(R,G,B)
W'=M, R'=R-M, G'=G-M, B'=B-M

This is not perfect but it works ok for the most part.

This is another example:
http://codewelt.com/rgbw

you could add some parameters like
white-transform = ignore, ww, nw, cw

or just allow to pick an RGBW value for pure white and define it like that.

Example of applications:

[image]

[image]

Aircoookie, 2018-06-24:
Hi,
thank you for your very in-depth reply!

About the issue of not being able to reset the effect in the mobile UI, I just tried it (with desktop UI still open) and it works for me. There seems to be an issue right now that causes the ESP to become unresponsive after some web requests... I'm still after that one!

You can test the auto white rendering option when you have RGBW enabled and activate the "Auto-calculate white" option. Then you should notice that your white channel slider does nothing (this is a bug, it should disappear completely, like in RGB-only mode). Then set a white-ish color with the color picker or sliders. You should notice the light being much brighter now because the Auto RGBW mixes in some white channel. My function is based on the example from http://codewelt.com/rgbw and I found that it works really well.

I've tested the Ambilight (serial Adalight) via the AmbiBox program for Windows. For me, it offers really accurate colors, using only the RGB channels, not the white. Also, it reacts very quickly to on-screen changes (keep in mind that I use an ESP32 for it, but did a quick test with my Wemos D1 and also found no issues)

However, I'd be surprised if  #define FASTLED_ALLOW_INTERRUPTS 0 did anything to improve the situation, since WLED is based on the NeoPixelBus driver, not FastLED.

I agree with you that a color temperature picker would be a great idea! I'll look into implementing one that also respects the WW/NW/CW strip type some time soon.

What do you mean by "once RGBW is integrated on Adalight"? The Adalight protocol only supports RGB, so do you refer to a modified protocol that allows RGBW, or an auto-white-channel feature on the ESP side like we were talking about earlier? In each case, there is one factor of concern with RGBW, and that is that white areas would appear significantly brighter than, for example, yellow areas of the same brightness. To mitigate this, we could gradually turn off the RGB channels when increasing the W brightness, but this would IMO not yield any benefits (brightness, accuracy) over pure RGB, but feel free to prove me wrong :)

Thank you for all the suggestions and help!

monarchit, 2018-06-25:
My pleasure, I've been trying to integrate a lamp-style sketch and Ambilight too so it's great that this version has so many features, I am very happy to help out. I'll be waiting for MQTT support to enable homebridge too.

Yes, I was referring to auto-white-channel you implemented to be enabled for Adalight as well.

Regarding a possible issue with LEDs being too bright, the white channel could be limited by the user, you could use some input boxes on the settings page to define what pure white would look like and it would use it to mix it with other colors to create different shades.

I'm using Prismatik and allows for temperature and gamma correction.  I don't like that as pure white is not as bright for me the algorithm seems to use white led alone to produce white and it looks too warm for me because it does not use the added RGB to produce it, having a white backlight on a monitor that matches the screen is good for eye care and Prismatik allows for moonlight mode and minimum brightness for example. If I correct the temperature it works great, it mixes in some RGB but it seems to limits the overall brightness a lot somehow as it is done inside the Prismatik software.

Whenever I use Adalight with RGB alone, all the colors are not as bright, explosion scenes look way better with a white component hehe. 
I have a 60pixel/m SK6812 and Adalight with RGB alone it's just not bright enough as the LEDs seem to be smaller as they have the W component on the same chip. but for example on WLED sliders if I set RGB to max and play around with the white value (about 40-50%) the white color in my case (WW) is rendered perfectly and if i try the same (50% white) with max yellow it becomes incredibly accurate because of the added Warm White, without it my yellow looks muddy and greenish. I think it's an issue with most RGB strips as the green channel tends to go towards blue. Other than that the max RED, GREEN and BLUE alone should be left without as it does now, while just max yellow needs a kick of nw/ww. There seems to be some SK6812 strips that are RGBY for this reason: https://www.youtube.com/watch?v=2Z4ZzfL5ahY. But I think that's an entirely different thing by itself.

In fewer words, I have tested the feature you implemented and it's working great for me. My only issue is that what suppose to be pure white looks way too warm in my case but it's generating it using RGBW altogether which is super great instead of just White LED alone. Would look amazing with just 40-50% warm white on it and yellows would look great too with up to 80% ww added based on how intense it is, it truly fixes the green muddy effects.

Regarding the interrupts option, I have mistaken this with the serial baudrate, it's what it fixed for me I guess the other option was just me trying things, sorry about that. 

I did some more tests and if I use Serial.begin(230100); Prismatik works almost in realtime but with  Serial.begin(115200); it does not, the background light seems to slow down a lot and drag behind at this speed for me. The slight flicker was caused by something else on my pc it seems.

As for Adalight with FastLED RGBW, this is the version I am using for reference on the effect it has. I just replaced the original FastLED on the Arduino IDE folder with the one from: https://github.com/coryking/FastLED. I have included this on the zip here too. If you try this with Prismatik, you will notice that the white is generated with the white LED only on default temperature settings, this might be good with NW but it's not great with WW as it needs some RGB mixed in. This sketch is using baud rate 230100.

[Adalight-FastLedRGBW.zip](https://github.com/Aircoookie/WLED/files/2132171/Adalight-FastLedRGBW.zip)

Aircoookie, 2018-06-27:
Great, thanks for the testing and your detailed reply!
I'll look into adding a sort-of dynamic color management system soon.
Based on what we talked about, it should be able to:
- Add a configurable amount of RGB to pure white to make it more natural
- Add some white to a certain color hue (yellow for example)

A development challenge with the WS2812/SK2812 in general is that they "only" support 8-bit or 256 brightnesses per channel. That means, if we want to only add a little bit of color correction it is hard because the adjustment steps are quite coarse at low brightness (this problem is amplified by the master brightness multiplier and the "standard" gamma correction, but I'll see what I can do.

Also, I agree with you that I don't like the normal yellow (FFFF00), it looks way to greenish and adding WW definitely makes it better!

About the baud rates, I'll experiment with them a bit. If I too notice a responsiveness improvement, I will raise the default baudrate to 230100, 512000 or even 921500. By the way, how many LEDs are you using for your Ambilight? I only deployed it on my PC monitor with 64 LEDs yet, but I could easily see a higher amount of LEDs (for a TV ambilight) require much more bandwidth.

Thank you again for your suggestions on how to improve WLED! I really appreciate your help!

monarchit, 2018-06-27:
I am using 101 LEDs, if I use anything higher than 230100 it doesn't work for me.

Everything sounds great, I didn't know about this 8bit limitation. I'd be happy to do some testings once it's out. 

Thank you too for the great dev!

monarchit, 2019-04-01:
> Great, thanks for the testing and your detailed reply!
> I'll look into adding a sort-of dynamic color management system soon.
> Based on what we talked about, it should be able to:
> 
> * Add a configurable amount of RGB to pure white to make it more natural
> * Add some white to a certain color hue (yellow for example)
> 
> A development challenge with the WS2812/SK2812 in general is that they "only" support 8-bit or 256 brightnesses per channel. That means, if we want to only add a little bit of color correction it is hard because the adjustment steps are quite coarse at low brightness (this problem is amplified by the master brightness multiplier and the "standard" gamma correction, but I'll see what I can do.
> Moreover, I'd like to avoid doing too much floating point maths for the calculation, since, in case of ambilight for example, it needs to be done for every single LED, which could significantly slow down things, we need to test it.
> 
> Also, I agree with you that I don't like the normal yellow (FFFF00), it looks way to greenish and adding WW definitely makes it better!
> 
> About the baud rates, I'll experiment with them a bit. If I too notice a responsiveness improvement, I will raise the default baudrate to 230100, 512000 or even 921500. By the way, how many LEDs are you using for your Ambilight? I only deployed it on my PC monitor with 64 LEDs yet, but I could easily see a higher amount of LEDs (for a TV ambilight) require much more bandwidth.
> 
> Thank you again for your suggestions on how to improve WLED! I really appreciate your help!

Hello again,

I have noticed many improvements on WLED. I wanted to know if any progress has been made on color enhancements using WW or W.

I remember RGBW was not supported on Ambilight mode. I wanted to know if this has changed in newer versions.

valdisd, 2021-02-03:
Adding +1 for the ability to choose how white is the white 4th channel. Kelvins or templates like WW, CW or NW.
As in Discord channel I learned that the feature to choose how white is your white is not yet implemented. 

From Discord chat:
"I, unfortunately, ordered RGB  + Warm White (not Neutral White).  At the time not knowing the WLED and not understanding, that you could do the auto calculation. And now if I use suggested modes: Brighter or Accurate, my colors are actually not accurate when blending in the white channel. They are more yellowish/orange because of the warm white, not neutral white?"

blazoncek, 2022-10-10:
CCT and color correction (not calibration) implemented since 624993fc89266f4c206faeec535b542f52654519

closed 2022-10-10 by blazoncek