wled/WLED #29: MQTT support implementation details question

Aircoookie, 2018-06-29:
Hello all,

some of you already asked for MQTT support, so I want to implement it for WLED some time soon.
This is the first feature of WLED that will be added without me using it, so I'd like to get some feedback from you on how you want the implementation to work! I'm planning to use the popular PubSubClient library.

For those planning to use MQTT, I'd be very happy if you could tell me something about your requirements and if they are met by the things I though about already:

- What topics does WLED need to subscribe to? Is it enough if each ESP subscribes to the 2 configurable topics `<yourcustomtopic>/<devicename>/<property>` and `<yourcustomtopic>/all/<property>.` The custom topic could be anything you want, for example `/myhome/bedroom/wled`. With the "devicename" you could target one or with the "all" every device in that "group".

- What "properties" would you like to see supported? Brightness and color for sure, also one to do any API call that would be possible via HTTP is what I thought of. In what format should brightness and color be supported? Comma separated ASCII integers, raw bits or hex strings?

- Do you need WLED to publish any topics? Like current brightness, color, FX, or anything else? If yes, I think this would be easiest if WLED publishes to the topic `<yourcustomtopic>/<devicename>/out/<property>` every time the color or other property updates, right?

- Do you have any other suggestions?

Thanks for your help! Since I don't use an MQTT broker or a central home automation framework, you will know your specific requirements better than I do :)

raymiec, 2018-09-23:
Sounds like pretty much what we need, 

here's an example of a light i use MQTT in Homeassistant 
    state_topic: "homie/livingroom/light/on"
    command_topic: "homie/livingroom/light/on/set"
    rgb_state_topic: "homie/livingroom/light/color"
    rgb_command_topic: "homie/livingroom/light/color/set"

With these topics i could turn it on and off and set the colours.
(homie is the esp firmware i used on the bulb)

Being able to control start and stop the effects would be pretty cool, especially the user defined one.  (movie starts, i can automatically make it call a theatre chase effect on wled, thats sort of thing.)

I'm happy to do any testing etc.

Aircoookie, 2018-09-23:
Thanks for the feedback, Raymie! I'll get to work on it soon :)

FX changing will be supported because I'll add support for all HTTP API calls to MQTT.
So you could publish "FX=13" to the topic "wled/livingroom/light/api" and it would set it to the chase effect, for example :)

Aircoookie, 2018-09-23:
Woah, thats a lot of Python :) I'd imagine it wouldn't be too easy to port the gist of it to C++...
That said, syncing Hyperion to WLED lights wourld be really cool!

I'm not sure if you know it yet, but it already works the other way around! You can add a UDP protocol 0 device on port 19446 in Hyperion and then use your WLED strip as an ambilight or similar!

kendersec, 2018-09-30:
Looking forward to the MQTT protocol support! I've tried setting up Hyperion but haven't found a good way to setup the service alongside HomeAssistant in the server. Seems to have QT dependencies and image processing pieces that aren't needed - I just want a proxy that translates orders from the JSON Hyperion protocol to the UDP protocol (haven't found any other tool that uses the UDP protocol).

Fashed the latest version of WLED with E1.31 support and had some success there but I have to find a good controller (LedFx looks promising but still very green).

raymiec, 2018-09-30:
ah sorry i wasn't clear, i posted the code as an example.

That just shows how Hyperion is controlled through HomeAssistant.

Forgetting Hyperion, something similar to that code to control WLED from HomeAssistant would be sweet.

Aircoookie, 2018-10-01:
Sounds great! Quick update about MQTT implementation: Pubsubclient library is working nicely, can reliably control the lights via a test broker! Now all thats left to do is add the settings where you can select your custom topics, and to add some documentation in the wiki. You'll be able to find MQTT in the development branch in max. 2 days time. Master branch will not be updated for now because there are instabilities in AP mode i still need to work out.

Yes, I agree. It would be awesome to have a module for HA that translates from their internal protocol to the WLED http API!

aidbish, 2019-02-10:
Does Wled publish all of its state changes, on, off, fx, etc. i can only see brightness and colour?

Aircoookie, 2019-02-10:
Hi @aidbish ! No, the current MQTT implementation only supports brightness and color, unfortunately. The brightness value contains on/off, if it's 0 the light is off, else on.
With the upcoming JSON API, more states will be changable/published hopefully. I don't really want the implementation to publish to even more topics as that could negatively affect performance.

Currently, you can activate the HTTP API response to be published to /wled/devicetopic/v by uncommenting from line 61 in wled17_mqtt.ino, but your client will need to parse the normal XML response to obtain the fx and other values and you have to increase the MQTT buffer in Pubsubclient.h, so it's not an optimal solution.

aidbish, 2019-02-10:
Thanks for the prompt reply. Hopefully with the json api more states can be published. I am using Home Assistant as my automation software and would like to implement WLED into it including all its states

Mariu86, 2019-02-10:
#Aidbish can you tell me how hard it's to implement wled to home assistant ? I am a newbie in software but I will like to implement home assistant in my house. Thanks

aidbish, 2019-02-10:
@Mariu86 Hi mate, i haven't figured out how i will implement yet

Mariu86, 2019-02-10:
If You figure out please tell us here. Thanks

Obicom, 2019-03-02:
Hi, I also use Homeassistant and use currently McLightning ... this works perfect together with HA and could be a good example. for mqtt implementation. I see some problems with the current weld behavior. For example the effects must be setup by a list for the topics and the same values will be shown in the light entity. With wled one list item would be e.g. "FX=13" and this is then the item in the list of effects. Nobody knows which effect it is.  Also we would need a feedback back to ha if a change is done by the webUI. Next prblem HA expect for color set a value like 255,255,255 and not a hex value. For my a nogo is currently the "no security" without username and password for my local broker. This are all only my 2cents ... but really awesome tool ...

Aircoookie, 2019-03-02:
Hi @Obicom, thanks for the heads up! The ha mqtt components api looks good, I'll try supporting that!

About effects, I agree this is a problem. Normally you are expected to HTTP GET `[IP]/json/effects` to obtain a list of the effect names, but of course this is not possible using MQTT alone. While I could certainly have a topic publish the name of the currently selected effect, this wouldn't help much, since setting the FX would still be possible by ID number only.

Feedback back to HA is provided. Any change to the light values will result in WLED publishing the state to `/devicetopic/g` (brightness) and `/devicetopic/col` (color). We could extend this to publish a JSON response to another topic.

Color is also not a problem, I can extend it to use that syntax.

The "no security" is intentional for now, because MQTT "authentication" sends passwords in plain and can therefore not be considered secure in any way. I plan on adding TLS support at some point, but until then, I might add the MQTT auth with a big warning that it's not actually secure since you are not the first one to ask for this feature.

Hope we can find a nice solution to integrate WLED with HA. Ideally I'd really appreciate it if we made a separate HA component for WLED, because the generic MQTT type required quite a bit of config by the user (most of the values like name, max-brightness, etc. could be provided automatically, in addition to that a native component could solve the FX/palette name issue), but I realize this might be a lot of work, so using the MQTT component would be an option as well. Let me know what you think.

Obicom, 2019-03-02:
Maybe a flag in Sync Interface for HA support could solve this point. If the flag is set the topic would follow the HA mqtt light component rules. To get an impression how it looks like in HA here some pictures:

[image]

[image]

[image]

Aircoookie, 2019-03-06:
That UI looks good! Really want to support HA now, it will definitely happen soon!
Yes, I'll have an option to enable "Home Assistant mode" in the MQTT settings.

About the effects, I don't really like the way HA implemented it with requiring a string.
It leads to this sort of madness code (of course having the only benefit that the FX name will stay the same even if the FX ID change):

```cpp
void setEffect(String effect) {
  if (effect == "static")
    ws2812fx.setMode(FX_MODE_STATIC);
  if (effect == "blink")
    ws2812fx.setMode(FX_MODE_BLINK);
  if (effect == "breath")
    ws2812fx.setMode(FX_MODE_BREATH);
  if (effect == "color wipe")
    ws2812fx.setMode(FX_MODE_COLOR_WIPE);
  if (effect == "color wipe inverted")
    ws2812fx.setMode(FX_MODE_COLOR_WIPE_INV);
  if (effect == "color wipe reverse")
    ws2812fx.setMode(FX_MODE_COLOR_WIPE_REV);
  if (effect == "color wipe reverse inverted")

  //goes on for 150 lines and wastes lots of global memory and cpu
}
```

Luckily i thought of a workaround, which is to name the FX something like "1: Blink" in the configuration yaml. Then i can just let the esp parse the int at the beginning but the user still has useful FX names in the list!

sujitrp, 2019-03-18:
Based on toblum/McLighting project using openhab & MQTT : 

toblum/McLighting  mqtt : McLighting listens for commands on the topic "<CONFIGURED_HOSTNAME>/in" and responds on "<CONFIGURED_HOSTNAME>/out
any change using mqtt / http / web soket / HTML UI publishes the changes to ..../out topic 
due to this fetaure i am able to consume keep status updated in all the clients . 
any client can give input command once the change completed the ..../out topic gives an update .

Note : <CONFIGURED_HOSTNAME> brings limitation . In terms of MQTT topic WLED has good strategy of user define  topic . 
My suggestion will be user define topic / out to publish the any change in WLED  by any source of input.  

https://github.com/toblum/McLighting

In this project  toblum & debsahu are doing fantastic job . i can see both of then part of WLED project too. 

Following is Openhab Dashboard :  Alexa ,Siri , Google home integration done . 
My project video 
https://www.youtube.com/playlist?list=PLY5D5GD5dZDAtn9f0cT8c4ydVDBNiu0uo

I shifted to WLED because it is more matured with functionality and effects . 
 

[image]

[image]

[image]

[image]

stockklauser, 2019-07-22:
> Maybe a flag in Sync Interface for HA support could solve this point. If the flag is set the topic would follow the HA mqtt light component rules. To get an impression how it looks like in HA here some pictures:
> 
> [image]
> 
> [image]
> 
> [image]

Hay, can you share your Configuration for Home Assistant?
Did you use JSON or MQTT?

raoulteeuwen, 2019-09-01:
@stockklauser could you please share what you have working re WLED and HA, and with what code? I think i have all the components running, but haven't figured out how to connect the parts to work together. Tx in advance!

TimmyUll, 2019-09-28:
@Obicom can you share your config? I have not got it to work with seperate whitechannel.

closed 2019-10-25 by Aircoookie