wled/WLED #31: Help needed Integration with Google assistant

pradhyuman18, 2018-07-02:
How can I Control it with Google assistant

Aircoookie, 2018-07-02:
Hi,

sorry that is not possible because the Google Assistant doesn't have a local API. I'd need to have a complicated cloud infrastructure to support Google home which I can't do as a small open source project.

You can do it with the free IFTTT website though, which is compatible with google Assistant. It requires you to do a router port forwarding so the ESP8266 is accessible from the internet. Then you need two IFTTT Webhooks applets that trigger `<YourPublicIP>:<YourESPNetworkPort>/win&T=1` for on and `<YourPublicIP>:<YourESPNetworkPort>/win&T=0` for off, respectively.

Aircoookie, 2018-08-04:
This is not a bug, and there is no way for me to add native Assistant support in the near future sadly.
Therefore I close this issue now.
Feel free to reopen it if you have any further ideas :)

nightflasher, 2019-01-22:
Könnte man nicht eine von Google unterstütze api integrieren? Die api von tuya smart (smart life) ist offen und soll leicht verständlich sein. Viele sub-hersteller (smart life oder auch hama) nutzen die.
Vielleicht hilft ja der link? 
https://docs.tuya.com/en/cloudapi/cloud_access.html#access-mode

Aircoookie, 2019-01-23:
Könnte man überlegen. Würde sicherlich funktionieren, aber im Moment möchte ich keine cloudabhängige API integrieren. Es ist aber möglich, das Ganze mithilfe von IFTTT und Blynk zu bauen: https://www.instructables.com/id/GOOGLE-ASSISTANT-CONTROLLED-SWITCH-USING-NODEMCU/ dazu könnte ich mal ein an WLED angepasstes Tutorial erstellen, weil Blynk ja schon implementiert ist.

nightflasher, 2019-01-23:
Hey, ja mit ifttt habe ich schon einiges umgesetzt; aber die weiblichen Mitglieder der Familie bombardieren die Google home Geräte mit Phrasen die ich vorher nicht angelegt habe und sind dann enttäuscht, wenn auf "hey Google, lightbox Farbe Lavendel" die Farbe dann nicht umgesetzt wird.
In cloud APIs geht's dann aber.

Vielleicht könnte man die Farben ja sprachbasiert mit rgb Werten hinterlegen? /win&color=red/rot

aenertia, 2019-12-25:
This is now possible as there is a local API exposed in the New SDK i think this should be reopened. https://developers.google.com/assistant/smarthome/concepts/local

bitterdog, 2019-12-28:
> This is now possible as there is a local API exposed in the New SDK i think this should be reopened. https://developers.google.com/assistant/smarthome/concepts/local

right on!

Aircoookie, 2019-12-28:
Not yet looked into it in detail, but I agree that this is a promising change!

Algebruhm8, 2020-01-09:
Any update as to having google automation? I've tried using WebCoRE with Smartthings, no luck. Couldn't do Blynk as I'm new to programming. I already set up my ESP8266 with my WS2812b lights but I want at least Google home to turn them off or on. Any other ideas? I tried adafruit but i'll have to see as I followed a tutorial and I got a bit lost

tomXGames, 2020-01-10:
Are there any updates for this feature? I'm really looking forward to be able to control my lights with my google home!

axlan, 2020-01-25:
I spent a little time getting the Blynk webhook working for one of my projects. It takes a bit of leg work, but here's a write up of what I went through to use Google Assistant via IFTTT and Blynk: http://www.robopenguins.com/web-based-door-control-update/

null-route, 2020-04-07:
Relevant to this issue: Google announced that their Local Home SDK has emerged from developer preview and is now GA: 

https://developers.googleblog.com/2020/04/local-home-sdk-ready-for-actions.html

t6a, 2020-04-09:
This would make WLED to controller even more user friendly and ready for the mass

geekafy, 2020-04-23:
Just reading the updates, this is exciting! Can't wait to see this feature land! :-) Keep up the great work!

akshatkumar369, 2020-05-06:
Is there a tutorial how can i integrate wled with google assistant?

jgrafwallner, 2020-05-11:
Watching this open issue eagerly!

spiff72, 2020-05-30:
Following this one as well.  My kids will appreciate the ability to control their desk LEDs with the Google home speakers in their rooms!

Legsmaniac, 2020-05-30:
+1 here for Google Integration in the near future too.
I'm a Google fan more than I am an Alexa fan.
Although, besides IFTTT, I *could* control it with Home Assistant but I would rather see direct native support.  🙃

Legsmaniac, 2020-05-31:
> Is there a tutorial how can i integrate wled with google assistant?

Found [this](https://docs.google.com/presentation/d/1cMY1bHTxkYlPpF4K-1XfVbxrV5TV8U05ez9tcZb3Cg8/htmlpresent).  Any help?

justusiv, 2020-07-14:
I was able to set this up. First using this guide for having a static FQDN with a cert https://www.youtube.com/watch?v=hqRT2VKhxJ0 i had not router loopback problems. Then i mainly used the https://www.home-assistant.io/integrations/google_assistant/ guide to do the rest. I think i also watched https://www.youtube.com/watch?v=MMtCzX6US9M and https://www.youtube.com/watch?v=mDqllkmxiOI the only gotcha was google no longer accepts the apikey and will only accept a json fil. so i had upload my json file i got from google to home assist. 
google_assistant:
  project_id: myproj-845100
  service_account: !include myproj-43756435645.json

The part i dont have working and would be happy to do a write up would be how to change things like fx via google.

justusiv, 2020-07-17:
after briefly talking to @Aircoookie and "e_town" on discord i was able to setup saying things like "OK google turn on Bedroom popcorn effect" 
Since there are currently 103 effects and 52 palettes i decided to wipe up a quick powershell the generates the data for the configuration files.
No real time was spent on this but should work for the basics. 
https://gist.github.com/justusiv/177c0750a1936486b735a2a2d503e424

Fade97, 2020-10-13:
Hi, just wanted to ask if there are any news on the integration?

anjeshshrestha, 2020-10-26:
maybe it can have more future support once Connected Home over IP becomes more developed and supported

https://github.com/project-chip/connectedhomeip#connected-home-over-ip

Aircoookie, 2020-10-26:
CHIP seems awesome! Perhaps this is finally the standard we've been waiting for that will ideally work with all major voice assistants without needing a "manufacturer server". I just hope that the overhead for implementing security won't be too much for an ESP8266, they tend to be quite unusable in most things crypto.

flipside101, 2020-10-30:
> 
> 
> maybe it can have more future support once Connected Home over IP becomes more developed and supported
> 
> https://github.com/project-chip/connectedhomeip#connected-home-over-ip
> https://www.connectedhomeip.com/

isnt this what zigbee was suppoed to be until it got butcherd by the very people involved in CHIP ?

ShiHotKeyMushroom, 2020-11-27:
> The part i dont have working and would be happy to do a write up if i figured out how would be how to change things like fx via google.

I haven't had the chance to go through those links yet, but is it controlling it via http request on the local network? If so I think I can help you with controlling the fx.

tronicsmasta, 2020-11-28:
Ok got it working using Google assistant and IFTTT. Took all day of experimentation. Will make an instructable soon

ShiHotKeyMushroom, 2020-11-28:
I am looking forward to it! Does it allow for the effect to be changed?

tronicsmasta, 2020-11-28:
> I am looking forward to it! Does it allow for the effect to be changed?
> […](#)

Yes!!

tronicsmasta, 2020-11-28:
Done! Thanks Aircookie for making such a flexible platform!
https://www.instructables.com/WLED-on-ESP8266-IFTTT-Google-Assistant/

ShiHotKeyMushroom, 2020-11-28:
Would it be possible to use ingredients in the "if" side of ifttt to
control certain aspects of the lights?

tronicsmasta, 2020-11-28:
> Would it be possible to use ingredients in the "if" side of ifttt to control certain aspects of the lights?
> […](#)

I haven't tried that yet. I was excited I got it to work with http requests. Will try with dynamic parameters today. That is what the little + symbols are in between if this + then that.

ShiHotKeyMushroom, 2020-11-28:
I tried setting this up for myself...but my port forwarding doesn't have
specific internal ip as one of the parameters...so...gotta figure out
what's going on there...

tronicsmasta, 2020-11-28:
> I tried setting this up for myself...but my port forwarding doesn't have specific internal ip as one of the parameters...so...gotta figure out what's going on there...
> […](#)
Send me an email, my username @ gmail dot com

f1vefour, 2020-12-06:
> Done! Thanks Aircookie for making such a flexible platform!
> https://www.instructables.com/WLED-on-ESP8266-IFTTT-Google-Assistant/

Thank you for the instructions.

tronicsmasta, 2020-12-06:
Welcome!

LukePrior, 2020-12-17:
Looking forward to an official integration!

kurtjcu, 2021-04-10:
I HATE the idea of opening ports and using IFTTT. Very bad idea imho! Is there anyone working on a direct GA integration? I am keen to work on one just checking to make sure I am not re-inventing the wheel. :)

luismasg, 2021-04-26:
google launches something called  google actions local SDk. it says all the right things.

it allows  thr google.home mini to  make requests  to  other devices.

https://developers.googleblog.com/2019/07/developer-preview-of-local-home-sdk.html?m=1

i haven't read it  completely yet

axl12ap, 2021-04-30:
> google launches something called google actions local SDk. it says all the right things.
> 
> it allows thr google.home mini to make requests to other devices.
> 
> https://developers.googleblog.com/2019/07/developer-preview-of-local-home-sdk.html?m=1
> 
> i haven't read it completely yet

Good starting point, have you any idea on how to do it?
@Aircoookie did you read the article?

Aircoookie, 2021-04-30:
@axl12ap yes, they even had a preview of it two years ago. Unfortunately, it does not seem like the new SDK would be useful for WLED's use case as the local option is merely there to reduce latency, device discovery is to my understanding still not possible without a "manufacturer cloud"...

luismasg, 2021-05-08:
ok screw it. it can't be done stand alone.

I  have no problem writing an express  app on a raspberry pie and send post commands to it.

or _something_ 

I'm intrigued  with getting a hubitat.
I've seen  i believe, the hookup guy mention it.

on 'google, light up my bed',  send GET 
to wled

i've seen no comments or projects or tutorials on something like this.

any ideas?

balucanb, 2021-05-08:
Just read through the whole issue, I noticed only ESP8266 was mentioned, will  a ESP32 work with this also?

Aircoookie, 2021-05-08:
@balucanb sorry, it does not work with ESP32 either. Almost all WLED features are the same on both platforms.

spiff72, 2021-05-08:
Not sure this is 100% relevant to this thread issue thread, but if you have Samsung Smartthings, WLED can be integrated with that, and Google assistant has some ability to control them.  I used this to do it:
https://community.smartthings.com/t/wled-with-smarthings/175628/8

Ryanb58, 2021-05-20:
I might be able to build a service that does this.

IMO, the hard part would be building a way for IT to talk to our WLED instances without port forwarding.

One method I thought of would be to have a setting in WLED where it would poll an HTTP server every few seconds until it got a response that told it to change its current settings. 

It's not the push like notifications we've come to love, but it's an idea. I'll let everyone know if I actually make an attempt at setting something up like this.

luismasg, 2021-05-20:
@Ryanb58 
i just realized my comment was imcomplete.
yes, the port forwarding thing.
i was talking about installing the Google smarthings sdk on  a rpi and make the rpi masquerade as a lamp or whatever. 

after it recieves the command from the google cloud. then it forwards it to your local network to wled. it it posible that I'm simply describing what node-red does(doh!) or home assistant.
i still have some learning to do.

surely all the google enabled things don't require port forwarding.

Aircoookie, 2021-05-22:
@Ryanb58 that would be absolutely amazing!

> One method I thought of would be to have a setting in WLED where it would poll an HTTP server every few seconds until it got a response that told it to change its current settings.
>
> It's not the [...]

How about MQTT? The publish/subscribe type messages mean no polling and thus almost instant change. I've thought about the architecture `[Google] <> [Integration] <> [MQTT-Broker] <> [WLED-Device]` before, but I honestly have no idea how to build a service that integrates with Assistant, authenticates the user, and so on.

Ryanb58, 2021-05-23:
**Little bit of progress:** 
Today, I was able to successfully send commands from my Google Assistant to an AWS Ec2 instance I set up(after the oauth 2.0 authentication was established). No progress with sending commands to the actual WLED devices yet. Still need to figure out a way to get it to connect to my public server that is exposed to the internet(required by Google Assistant).

@Aircoookie The ways to send change requests(JSON, MQTT, etc…) are only available on the local network. To pursue the MQTT approach, I'd need a Samsung SmartThings device or something similar. I'd like to go at this with an approach that doesn't require me to have to buy any more hardware. Depending on how I can establish the connection

**Side Note/Discovery:**
With Google's requirement to have a public OAuth 2.0 server running for authentication... it is going to(already is) costing me $$ to operate. IDK how I am going to deal with that, but one thing at a time.

Ryanb58, 2021-05-23:
@Aircoookie I may not understand MQTT. I'm going to do more research into it. My last comments about it may not apply.

Aircoookie, 2021-05-23:
@Ryanb58 wow, that is awesome! If you manage to implement something that works, I'd definitely be happy to cover the hosting $ if it's not outrageous.

About MQTT, indeed, it works outside the local network, which I believe is a key requirement for this sort of thing to work.
Both the WLED device and your AWS service connect to the public MQTT server (also called broker) via TCP. We can use a general purpose free public MQTT server like `broker.hivemq.com` for the time being, if this gets big, we can consider hosting our own MQTT broker later.

How it works is also quite simple, the WLED subscribes to the `/wled/abcxyz` topic (where `abcxyz` is some unique identifier, currently using part of the MAC address) on the MQTT server and when the AWS service sends a control command to the MQTT broker (called a publish), the MQTT server forwards that to the subscribed client (the WLED device). The most tricky part would be  giving the service the correct identifier for the WLED device the user would like to control, so that it can publish to the correct MQTT topic. Maybe for a start, the user would need to fill in the identifier and device name manually in the same place that they authenticate (I'm very sorry, OAuth2 goes a little over my head, got no idea how it works)

Ryanb58, 2021-05-24:
@Aircoookie I am looking into using AWS IoT to manage MQTT messages. It looks like, by default, it uses x.509 certificates for security. Are there any plans on supporting this security method(or any) during the MQTT rewrite?

Aircoookie, 2021-05-26:
@Ryanb58 not currently, on ESP8266 we definitely do not have enought memory for crypto. On 32, it is less of an issue, but it is still quite hard. The only way that worked when I tried it was hard-coding a certificate, which of course breaks all communication if it expires, and only allows interfacing with one specific pre-set server.
That said, the connection between MQTT broker and AWS could be configured as TLS/x.509 with relative ease, while still using unsecured TCP between broker and WLED device.
It boils down to whether we'd need to send confidential data (e.g. OAuth tokens) over MQTT, I would be very uncomfortable without TLS, but for simple device name/announce/identifier/brightness or color change command types of traffic. It also only applies if we had a dedicated MQTT broker, with a public one, theoretically anyone could mass-subscribe to topics and intercept traffic, even if all transport paths were secured.

Ryanb58, 2021-06-01:
@Aircoookie I played around with MQTT a bit more over this last weekend, and learned a lot. I went through the steps of setting up a local Mosquito MQTT server and controlling my WLED lights from it; from a python script. That gave me a way better understanding of how it operates.

I didn't have to send any OAuth Tokens over the wire to MQTT broker. 

If I get some more time this next weekend, I will attempt to push it all onto an EC2 instance and test out my Oauth App's code alongside a dedicated MQTT broker instance. It will be over an insecure transport, but should give me a good feel for what this will look like running together. I'll keep ya updated.

Ryanb58, 2021-06-05:
Just testing out the integration a bit tonight. Figure I'd share the progress in a video.

https://www.youtube.com/watch?v=s2G-iIrtfhc

The Code isn't ready for prime time… Lots of embedded secrets and passwords… But I'll keep working on it to get it up to a server for other people to try in the near future.

In its current state... it is insecure.

f1vefour, 2021-06-05:
Very nice, this will make wLED infinitely more useful for me and I'm sure a
ton of others.

Looking forward to seeing this implemented.

spiro-c, 2021-06-07:
I was looking at this and try to find some Google assistant to mqtt integration and i came across this three projects that can be use to control wled over Google assistant for two of them need to have full SSL, setting up google actions  port forwarding so you can run them 
https://github.com/i8beef/HomeAutio.Mqtt.GoogleHome
https://github.com/mikejac/node-red-contrib-google-smarthome

but one of them is much easier it just need node-red no need for port opening and SSL 
https://github.com/andrei-tatar/node-red-contrib-smartnora

luismasg, 2021-07-06:
yesterday I  tried a theory.
here's another breadcrumb .

I used a shelly 1 and added an action to have it turn on Wled on switch.

it only works for  turning on and off but it is very close to what I wanted.
this feels like something that could be done in hubitat or home assistant.

[image]

Ryanb58, 2021-07-12:
After a few weeks of messing with this stuff, there isn't really anything we can add to this repo to make this work. It involves a cloud provider setup with a connection to your IoT devices. Thus, maybe this PR should be closed until one of these two items are completed.

- More mature local fulfillment by Google.
- MQTT connections over TLS on the module.

My original setup involved an insecure MQTT broker on the web. Which was abused. Thus, wouldn't provide a good user experience and is a security risk.

catarrer, 2021-11-19:
Is there any update on getting wled working natively with google home? I'm looking forward to this as i have two nest hub 2 and a sonos one with google Assistant. 
The lights can only be cobtrolled by the one echo spot i own.

[12 more replies]