wled/WLED #33: Forced to use GPIO2?

jollydodo, 2018-07-30:
First of all, thanks for creating such an awesome project!
However, I was wondering why it is necessary to use PIN 2 (GPIO2)? I'm using a custom led level shifter pcb with a sn74ahct125n chip. This makes it easy to create the 5v logic and saves me a lot of wiring issues/looks better.

I have a couple of these pcb's lying around, but they are designed to work with either D2 or D6. Is there a way to get this to work or do i have to use the messy route?

Aircoookie, 2018-07-31:
Hi, thanks for your support!

From my experience you can 99% get away with directly connecting the NeoPixel strip to the ESP. It's not best practise, but it generally works.
Like you've said, on ESP8266 the UART protocol restricts the pin choice to GPIO2, unfortunately. You can get to use GPIO3 by choosing another protocol called DMA. On ESP32 we don't have that pin restriction and can connect the pixels to any pin.

I hope you can enjoy WLED despite this issue!

Aircoookie, 2018-08-04:
Since this is not a bug, I'll close the issue for now.
Feel free to reopen it if you have something further to add to the topic :)

closed 2018-08-04 by Aircoookie