wled/WLED #34: ESP8266 Exception 28 (Adafruit Feather Huzzah)

synssins, 2018-08-04:
What a great project! I'm coming over from the McLighting V2 project, trying to find something I can use standalone without relying on a WiFi signal (for our tent lights when camping), and discovered WLED00...

I'm running into an issue though. I'm using RGBW WS2812b LED strips. When flashing the OTA Updater sketch first, then uploading the pre-built 0.5 BIN, the units work beautifully, except for the coloring on the strips. The 0.5 image does not seem to support RGBW, or at least I can't find a way to specify the LED strip in the documentation or the sketch...

When I build the sketch myself, I leave everything as default except the SSID/PW for my home wireless (for my test units here), and enabling the serial debug.

I receive a constant Exception 28 reboot within half a second of the unit connecting to my wireless here at home.

`v3de0c112
~ld
Load EEPROM
CSSID: Den of SynUSING AP7
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC
C_NC

Connected! IP address: 192.168.1.146

Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0104a8c0 depc=0x00000000

ctx: cont 
sp: 3fff3850 end: 3fff3ad0 offset: 01a0

>>>stack>>>
3fff39f0:  3fff3a60 00000002 3fff7be8 40222785  
3fff3a00:  0104a8c0 3fff1b34 3fff3a50 40223090  
3fff3a10:  00000000 00000000 3fff2334 4021d3cb  
3fff3a20:  3fff3a60 00000004 00000000 4020b33f  
3fff3a30:  00000001 00ffffff 00000000 40224398  
3fff3a40:  3ffe9980 3fff1b34 00000000 402185cb  
3fff3a50:  3fff43b4 00000000 4020b314 402177e0  
3fff3a60:  3ffe95a8 0104a8c0 feefeffe feefeffe  
3fff3a70:  feef0035 feefeffe feefeffe feefeffe  
3fff3a80:  00000001 feefeffe feefeffe feefeffe  
3fff3a90:  feefeffe feefeffe feefeffe 3fff2aa4  
3fff3aa0:  3fffdad0 00000000 3fff2a9d 40218744  
3fff3ab0:  feefeffe feefeffe feefeffe 40223278  
3fff3ac0:  feefeffe feefeffe 3fff2ab0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
Load EEPROM
CSSID: Den of SynUSING AP7
`

synssins, 2018-08-04:
I completed removed the entire WLED00 folder from my Arduino directory and re-extracted from the latest download. Placed everything back into place to confirm a fresh start, and the unit still crashes when trying to launch AP now with the same exact issue.

This is after performing an esptool erase_flash command to ensure the entire unit was purged of all settings.

Aircoookie, 2018-08-04:
Hi,

awesome that you like my project!

Would you mind trying the latest 0.7.0 version? The 0.5 in the bin-folder is no longer up to date, I use the releases page now: https://github.com/Aircoookie/WLED/releases . That said, even the old builds shouldn't insta-crash. If you experience the exception with the recent build also, I can help find the error. This might be linked to https://github.com/Aircoookie/WLED/issues/11 , maybe. Once you get it up and running, there is an option to switch to RGBW mode in the LED settings, so don't worry, this issue is not related to your strip type.

Thanks for your support!

synssins, 2018-08-04:
Thanks for the very prompt response!

I have also tried building 0.7.0 myself, and installing the prebuilt binary. No luck.

I WAS able to get 0.6.3 up and running without issue, and currently have it cycling in the home office right now.

Aircoookie, 2018-08-04:
That is indeed very weird. I was not expecting 0.6.3 to work when 0.7.0 doesn't. I've just checked again and 0.7.0 works fine when compiling with DEBUG enabled for Adafruit Huzzah (compile settings: 80MHz, 4M (3M SPIFFS), baud 115200). Changing the WiFi credentials in the code should technically work, but how it's supposed to be is that you flash the sketch as-is. Then it will open the access point WLED-AP with password wled1234. You connect to that (IP 192.168.4.1) and then in the web interface, you put in the data for your home wifi!

If you can send me the exception stacktrace for 0.7.0, i can work out where the issue is for you. For now, I'm happy for you that 0.6.3 seems to work :)

synssins, 2018-08-04:
I need to figure out how to pull the exception stacktrace for you. lol Any instructions?

I'm a hardware guy, this code stuff is over my head.

All WLED00 sketch settings are direct from github, fully reset at this time to defaults, including the SSID information within the sketch.

Aircoookie, 2018-08-04:
No worries, it's just this exception stuff the ESP prints out to serial when crashing, like you pasted into your first comment.

synssins, 2018-08-05:
I'm home now and have a brand new Feather Huzzah sitting in front of me. I have it flashed to 0.6.3, and I have the 0.7.0 BIN on the system. 
The Huzzah is on the WiFi (when powered on), so I can control/etc. 

I can update the unit to 0.7.0 using the BIN, or I can flash the 0.7.0 latest build directly from Arduino.

What would you like me to do?

Aircoookie, 2018-08-05:
Both should do the same thing. Try the BIN first. I'm personally using a modified version of Neopixelbus (https://github.com/svenihoney/NeoPixelBus) to compile the bins, but i also always test if it works with the current official versions.

Do you know which version of ESP8266 arduino core you use? No worries if you don't know it. I just want you to be aware that i still develop for 2.3.0 since the sketch sizes with that version are smaller and it allows me to continue support the low-end ESP-01 that only has 512K storage.

synssins, 2018-08-05:
I'm also using 2.3.0, per the note you had put in the readme.

I just updated the brand new test unit with 0.7.0 using the BIN. After the "Success" and reboot, it goes into the boot loop with Exception 28.

I performed a flash erase with esptool, reflashed with the updater, then uploaded 0.6.3 and it came right back up and works as normal.

Connected to my PC, performed a stock (no changes whatsoever) flash to 0.7.0 from the Arduino IDE and ran into the same issue as the BIN upgrade. It crashes and boot loops.

synssins, 2018-08-05:
It is I who should be thanking you... Rather than say "Go buy this exact hardware module cause I know it works", you are instead helping me out.

Here's the results from the Serial Monitor.

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld

Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0104a8c0 depc=0x00000000

ctx: cont 
sp: 3fff3230 end: 3fff34a0 offset: 01a0

>>>stack>>>
3fff33d0:  3fff3440 00000002 3fff6a88 4021ca89  
3fff33e0:  0104a8c0 3fff1504 3fff3430 4021d360  
3fff33f0:  00000000 00000000 3fff1d04 40217a6f  
3fff3400:  3fff3440 00000004 00000000 4020686f  
3fff3410:  00000001 00000582 00000000 3fff23b8  
3fff3420:  3fff2109 3fff1504 00000000 40212e32  
3fff3430:  3fff4b74 00000000 40206848 402121e8  
3fff3440:  3ffe9438 0104a8c0 feef0035 feefeffe  
3fff3450:  00000000 feefeffe feefeffe feefeffe  
3fff3460:  feefeffe feefeffe feefeffe 3fff2474  
3fff3470:  3fffdad0 00000000 3fff246d 40212f94  
3fff3480:  feefeffe feefeffe feefeffe 4021d548  
3fff3490:  feefeffe feefeffe 3fff2480 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16

Aircoookie, 2018-08-05:
Thank you for the kind words :)
Alright, I have successfully decoded the exception.
I don't yet know what the exact issue is, but it may be related to mDNS, a feature that was broken during 0.6 versions and fixed in 0.7.0, which would explain why 0.6.3 works for you.

I've changed some other things that might cause the exception in some recent development. You can try if the development version fixes it for you.
Get it here (you need to compile the source in Arduino, there are no binaries): https://github.com/Aircoookie/WLED/tree/development

If the dev version doesn't work, you can try the following:
- Flash 0.6.3 again
- Go to WiFi settings and clear the "led" from the "mDNS address" text field, then save
- Do NOT erase flash completely with esp-tool
- Update to 0.7.0

If 0.7.0 then works, we have confirmed that the issue is with mDNS.

However I'm happy that 0.6.3 works for you :)

Aircoookie, 2018-08-05:
Ok, don't bother with the dev version! I was just able to reproduce the issue! I really need to extend my testing upon releasing versions, 0.7.0 works great when you are connected to a home wifi in 0.6 and updating to 0.7.0, but not when you do a fresh install of 0.7 or not connected to wifi. Dev version doesn't work either. Working on a fix :)

synssins, 2018-08-05:
So the Dev version does not work for me. Same issue as before. 

Something interesting I just discovered... I have several ESP units (All Adafruit Feather Huzzah) around the house with Pixels attached. I've updated several to 0.6.3 and they're working fine, except that using the web interface from one, controls the others. Hitting the power button to turn one off from it's own interface results in the others also turning off... That should be a bug report in and of itself.

Attempting to update the unit to 0.7 after setting up the WiFi connection results in a page not found error and then the unit becomes non-responsive until many minutes after a reboot. Once it comes back up, I try the update again via the web interface (manual OTA update), and it goes non-responsive again. 

This may be related to the whole "all devices controlled by one" thing.

synssins, 2018-08-05:
Ah HAH! I don't think I knew what the arrows meant in relation to multiple units. 

I'll test the dev branch out once you update it and confirm it is resolved on this end before you make the 0.7.1 push. :)

synssins, 2018-08-05:
Dat Mobile UI...
ლ(▀̿̿益▀̿̿ヽ)ლ

The unit boots without the boot loop or crash, and I am now able to successfully install and configure it. 
Very excited to see where this project goes. I have SmartThings in the house, and would love to integrate these into the theater accent lighting... Having the lights change sequence/color as the movie experience in the theater progresses through trivia, trailers, etc will add a touch of "HOLY SH**" when people come over. 

Thank you for such a prompt response and fix!

closed 2019-03-27 by Aircoookie