wled/WLED #37: Sync multiple units over I2C/I2S? Multiple input pins to control functions?

synssins, 2018-08-07:
This is a two-fer question for my specific use case... WLED00 will work better for my automotive project than anything else I've found, and with my distinct lack of programming ability, it will work almost perfectly as-is, if I want to control things via wifi.

What I'd like to do to make it 100% is have certain patterns/colors/functions happen upon any one (or more) inputs being triggered. Basically, a pin goes high, and thing happens. Specific use case: Turn Signal triggers GPIO pin. Amber chase occurs. Pin goes low again, revert back to previous state. Secondary pin controlled by park lamps equals dim red. Turn signal while park lamps active equals amber chase over dim red.

How difficult is it to add additional button capabilities to support something like this? 

Syncing multiple units via I2C would allow some intercommunications between modules for things such as parade mode, where activating the input on a master unit would cause all units to go into pattern cycle mode, or some such.

Thoughts?

Aircoookie, 2018-08-07:
Triggering specific effects/colors when a certain input goes high should be easily doable. We can use the user modification file to listen for pin states and trigger actions. Do you want the effect to stay while the pin is high (like a switch), or listen for a pulse and revert to previous state once you get another pulse (like a pushbutton)? I can send you some example code to accomplish either case.

I2C could certainly be implemented, but that'd be a bit more involved. I doubt many people want wires to connect the ESPs since then you might aswell go with an Arduino instead of a WiFi chip like ESP8266.

How many ESPs do you need? If you have up to 4, you can set one as wifi access point and connect the other 3 to it and use the standard UDP wireless sync (in the upcoming 0.7.1 version that will be supported, right now it only works if all ESPs are connected to a "home" network). If you need more, you can use some sort of small portable router in your vehicle.

With the 0.8.0 update coming in September (hopefully), you'll be able to break up a strip into multiple "segments" and display different effects/colors on each. Maybe that would suit your application better than having multiple ESPs in your setup.

Hope I'll be able to help you get your setup awesome! Thanks for the support!

synssins, 2018-08-07:
> Do you want the effect to stay while the pin is high (like a switch), or listen for a pulse and revert to previous state once you get another pulse (like a pushbutton)? I can send you some example code to accomplish either case.

It would need to stay while the pin is high, and then revert to previous. This would allow input from multiple sources on multiple pins. Park with brake overlay and turn signal functionality on top of that when all three pins are active... 

In the end, the goal is to have normal function for driving, then at a car show or in a parade, flip a switch that toggles cycling through various patterns for "wow" mode.

Right now, one ESP per side for the tail lights/rear turn/park/brake, one total for the front should work. The wireless UDP between units might be a problem when driving down the road due to possible engine interference, signal drop out, etc. That's why I was curious about wire control.

Aircoookie, 2018-08-07:
> Park with brake overlay and turn signal functionality on top of that when all three pins are active...
Overlaying effects is not possible with the current version, but we could use the manual overlay function to get it to work!

> goal is to have normal function for driving
Please do NOT use WLED for a security critical feature like car lights (except if the original lights of the vehicle are still used additionally). The software is not guaranteed to be stable enough for that. So only use it on private property or to show off at cons/etc with concept cars. Otherwise it would probably not be legal (depends on where you live)

What I would suggest is that you put one standard AVR arduino uno/nano/whatever in each light, reading its GPIOs and setting the driving effects (indicator/brake/park/standard lights). They are FAR more reliable than a ESP with complex software running on it (but I still have to warn you about legal consequences regarding the use of DIY solutions in public traffic, keep that in mind). Then, when you want to show off, you flip a switch that disconnects the arduino from the strip and instead connects an ESP running WLED.

synssins, 2018-08-07:
> Please do NOT use WLED for a security critical feature like car lights (except if the original lights of the vehicle are still used additionally). The software is not guaranteed to be stable enough [...]

This is a very valid point and I am glad you brought it up. I've been building/running custom lighting solutions in my vehicles (and was a vendor to a niche car community for aftermarket DOT approved retrofit lighting for that car community). My original tail light assembly consisted of over 1500 LEDs in a grid with individually controllable segments for the purposes of sequential lighting. I understand and accept all responsibility for the work I have performed on my vehicles. 

My original LED tail lights, set up as the "Millenium Falcon", as dubbed by the community: [Fiero LEDs](http://www.denofsyn.com/personal/images2/LEDs/tails/IMAG0048.jpg)

I have some arduinos already set up as you are discussing and they work for the barebones stuff, it's the ws2812fx stuff that needs to be integrated so that it can only operate patterns in a very specific scenario, requiring the key be off, the toggle turned on, and a remote relay triggering another pin. Once those requirements are met, the lights do their thing. :)

I appreciate (and enjoy) the discussion on this.

blazoncek, 2021-10-03:
Some of the features are now implemented with multiple buttons and/or UDP syncing and upcoming network bus.
Closing.

closed 2021-10-03 by blazoncek

NonaSuomy, 2022-11-15:
@blazoncek What does "upcoming network bus" mean can you share more information?
Thank you.

blazoncek, 2022-11-15:
Yes.