wled/WLED #39: Increase FX speed?

StormPie, 2018-08-12:
Hey!

Version 0.7.1 😃😃😃! I love the colour warmth settings 🌞

I have recently acquired a 1M 144 LED strip, for more LED madness. WLED looks absolutely awesome on it! However, now that i've got such a high density strip, it seems the effects could be that little bit faster to make high speed effects look super cool on HD strips, and even faster on less dense strips.

Is this possible?

Thanks heaps

Aircoookie, 2018-08-16:
Hey,

of course, I should add such an option! On a 60 LED/m strip many animations will be 2x slower and on 144 LED/m even 5x slower! In version 0.8.0 I will add these super-cool FastLED noise effects. Then i can also add a setting where you select your strip density, so it adjusts the timings so that the effect will run the same speed on a 30 and 144 strip of the same length :)

StormPie, 2018-08-16:
Heck yeah! Super keen for the update 😄

closed 2019-10-25 by Aircoookie

Luke-Beck, 2022-03-11:
I'd really like for this to be considered again please

czweb, 2022-12-08:
I hope that the maximum speed can be greater than 255, or there is a speed multiplier that can be adjusted

blazoncek, 2022-12-08:
> I hope that the maximum speed can be greater than 255, or there is a speed multiplier that can be adjusted

It will not.
Name effects that need speeding up and we'll see what we can do.