wled/WLED #43: priorities of services

pete111, 2018-08-16:
Hi @Aircoookie , maybe I have not read wiki the right way, but what happens when I enable e.g. hyperion realtime, alexa, hue? I mean from prioritization point of view - will these services fight together? e.g. when hyperion changes realtime colors, then from time to time hue will reload remote color and will shows it for a while (flash) and hyperion continues in changing colors. Then alexa will send command to change color and flash again :) Will these services fight together or is there any prioritization implemented in Wled? (e.g. higher/lower priority for specific services)

Aircoookie, 2018-08-19:
Hi, that is a great question!
There are basically two operation modes in WLED, normal and realtime mode. Realtime mode always has priority and after a timeout you can set in the settings, WLED will return to the last set normal mode.

Normal mode consists of color and effects basically and can be changed by:

- Web UI
- HTTP API
- Alexa
- Hue
- UDP sync of multiple WLED ESPs
- Button press

WLED will always display the latest color/effect it received from any of these sources.

Realtime modes will override the normal color for a few seconds until it restores the previous normal color if no new realtime packet is received in the timeout period. It includes:

- UDP realtime (WARLS, DRGB, DRGBW)
- Hyperion UDP
- E1.31
- Serial Adalight protocol

WLED is not designed to receive multiple realtime sources somultaneously, so that is undefined behavior. It would probably crash/reset the ESP because too many packets are arriving at the same time.

If any normal mode updates are received while realtime is active, they will update the internal state, but NOT update the strip, so there's no flicker. However, the realtime animation may lag while the ESP is busy processing the normal mode request.

Always happy to help you out :)

pete111, 2018-08-19:
oh, that is awesome clarification Aircoookie! Thank you many times. Again I learned something about Wled behaviour :)

closed 2019-03-27 by Aircoookie