wled/WLED #45: Blynk effect selection and params fix.

kayrozen, 2018-08-31:
Hi ! 

I somewhat fixed wled_blynk.ino and it is working. Not sure if it's the best way tho but works anyway. I'm not setuped yet to fork and pull req. so here's the code snippet i've changed. Around lines 58-77. 

```
BLYNK_WRITE(V4)
{
 effectCurrent = param.asInt()-1;//fx
 strip.setMode(effectCurrent);
 colorUpdated(9);
}

BLYNK_WRITE(V5)
{
 effectSpeed = param.asInt();//sx
 strip.setSpeed(effectSpeed);
 colorUpdated(6);
}

BLYNK_WRITE(V6)
{
 effectIntensity = param.asInt();//ix
 strip.setIntensity(effectIntensity);
 colorUpdated(6);
}
```

Aircoookie, 2018-08-31:
Thank you!

Somehow I totally overlooked that this was broken...
Forgot to update the strip effect itself because I wrongly assumed colorUpdated() would do that...
Thanks heaps for finding this bug and even fixing it! :)
I will incorporate your changes in the upcoming update (on Sunday I'll release support for a few more effects)

kayrozen, 2018-08-31:
Good ! 

I'm also working on integrating the WUnderground API library to get weather info and play with them. 

i'm able to pull out a great deal of infos and apply them to color. Code is messy right now but i'll setup myself with git and all this w-e and proper fork to clean up code and be able to share in an efficient way :)

closed 2019-10-25 by Aircoookie