wled/WLED #48: how to flash? :-)

Mazdaracer121, 2018-09-14:
Hallo. gibt es hier jemanden der mir das flashen auf den ESP8266 erklären könnte?

Aircoookie, 2018-09-15:
Gerne, das ist eigentlich ganz einfach.

Hier ist eine Step-by-step-Anleitung :)
- [Arduino IDE](https://www.arduino.cc/en/Main/Software) (neueste Version) runterladen
- Im Menü "File -> Preferences" unter "Additional Board Manager URLs" folgendes einfügen: 
  `http://arduino.esp8266.com/stable/package_esp8266com_index.json`
Dies installiert die Software und Arduino-Libraries für den ESP8266.
- Dann WLED runterladen (in Github grünen Button mit "Clone or download" -> download ZIP)
- In der extrahierten ZIP "wled00.ino" mit Arduino öffnen
- Im Menü "Tools -> Boards" den "NodeMCU 1.0 (ESP-12E Module)" auswählen. Dieser funktioniert für die meisten Entwicklerboards. Wenn nicht, einfach dein Board in der Liste auswählen.
- Links oben den Pfeil (Upload) zum Flashen klicken

Wenn du auf irgendwelche Probleme beim Flashen oder dem Gebrauch der Software stößt, melde dich gerne nochmal :)

Viel Spaß!

Mazdaracer121, 2018-09-15:
Wunderbar, hat 1a geklappt. Super Arbeit die du da geleistet hast 👌 Vielen Dank dafür

closed 2018-09-15 by Mazdaracer121