wled/WLED #50: Cannot connect to ESP module

justdienow, 2018-09-28:
Has anyone had any issues trying to connect to the ESP module for what I assume the initial WiFi config? When I use my PC even with my WNIC configured with a static IP of 192.168.2.5, the web UI is not accessible. I have also tried to use my phone or laptop and they both say the password is incorrect.

Aircoookie, 2018-10-01:
Hi, 192.168.2.5 is the wrong subnet, I think. Try connecting to 192.168.4.1, preferably with your PC set to DHCP. If you want to try static IP, attempt 192.168.4.2. Password is incorrect? That's weird. In the initial config, it is always "wled1234". Is the WiFi AP name "WLED-AP"? if it is not, you might have flashed the software incorrectly or the flash contains invalid data. You could try flashing this sketch to reset all configuration data. After that, flash WLED and it will restore the defaults:

```
#include <EEPROM.h>

void setup()
{
  EEPROM.begin(3072);
  for (int i = 0; i < 3072; i++)
  {
    EEPROM.write(i, 0);
  }
  EEPROM.commit();
} 

void loop()
{
  delay(500);
}
```

Hope this can help you :)

justdienow, 2018-10-03:
Hi Aircookie,

Thank you for your response. The IP was a typo, what I meant was 192.168.4.5 (I was half going to sleep at the time sorry). 

I was also trying to debug the code trying assigning different strings to the char array **apSSID** to see if the standard "WLED" SSID would change, which it didn't only assuming that's what it's for?

This also lead me down the path of thinking the ESP8266WiFi and other associated libraries were broken. I even tried using both 2.3.0 and 2.4.2 ESP cores with no luck.

Have you had much success connecting to an ESP module using an Apple device? I have tried using your OTA updater and works on both Mac and Windows OS connection wise, however, as soon as I flash an ESP module with WLED, I am not able to connect to the ESP module.

I will try what you had suggested and will report back to you.

Thank you and kind regards,
Antonio