wled/WLED #51: Alexa Color change???

SeriousOri, 2018-10-02:
Hello is it right that Alexa only switch on and off and adjust the brightness? But Alexa can not change the colors ?

Aircoookie, 2018-10-03:
That's right, unfortunately. Amazon did not implement the part of the protocol required for color change in the local hue protocol we use. So there is nothing I can do about this, sadly.

SeriousOri, 2018-10-04:
Hello, thanks for the answer. Is it possible to combine the sketch of Sid (Hachster io)? Since we have 6 solid colors. Unfortunately, no dimming function is available.

https://www.hackster.io/shiva-siddharth/control-rgb-led-strip-using-amazon-echo-alexa-and-nodemcu-b06e92

Aircoookie, 2018-10-07:
Yes, it would be possible to control the color by emulating 4 devices at once (Brightness, Red, Green, Blue). I might implement this once i have time as an example sketch for my project [Espalexa](https://github.com/Aircoookie/Espalexa) which supports emulating multiple devices simultaneously.

In the future, I might change WLED to use Espalexa and add an option to enable this sort of color support. Downside is still that you can't just say "Alexa, set light to Orange", but have to do "Alexa, set Red to 100%. Alexa, set Green to 50%. Alexa, turn Blue off.". This isn't practical at all. I might look into the "Routines" feature of Alexa, which might make this easier for a subset of colors you want to support. That said, every user would have to program those routines themself.

Aircoookie, 2018-12-16:
While testing today, I have noticed that there might have been a change in the Alexa software that would allow color changes via the local API. I will investigate this further and if it is true, we should have Alexa color setting support before Christmas :)

nelzon101, 2019-01-13:
Hello. Very nice project!! All respect! It is now possible, to change colors with alexa? Greetz

Aircoookie, 2019-01-13:
Yes, the functionality has been added

sjude68, 2019-02-27:
I am able to turn On light with Alexa, but not off though Alexa replied OK.

Even color changes doesn't work.

Aircoookie, 2019-02-27:
Hi @sjude68 , you are experiencing the https://github.com/Aircoookie/Espalexa/issues/6 issue.
You have three options:
- use 0.8.3 [binary](http://bin.wled.me), flashable directly in WLED ("Manual OTA Update" in security settings)
- compile 0.8.3 using Arduino core 2.3.0
- use latest git master version (0.8.4-dev) with core 2.5.0

All of these will let you turn off or dim the device, but color might still not be supported depending on your Echo model. I'm actively working on a fix!

sjude68, 2019-02-27:
Thanks. I uploaded the binary using OTA Update and now Alexa can turn ON & OFFand change brightness of the lights.

I am using Echo Dot, 1st generation.

Waiting for Color implementation.

Aircoookie, 2019-03-27:
@sjude68 Using the new 0.8.4 release, you should now be able to change the color using your Dot! You might need to delete and re-discover the device in the Alexa app once.

sjude68, 2019-04-01:
Aircoookie. It works well with my Echo Dot. I connected 3 esp D1 Mini. All are on sync and giving command to 1 changes color in all.

closed 2019-04-01 by Aircoookie