wled/WLED #52: Not much to complain .... THANKS :-)

jolll-dk, 2018-10-03:
What an amazing piece of code you have written !!
So far I have it controlling 15 meters of ws2811  on two esp's.

Only thing I miss so far, is in the Mobile theme ... 
The on/off button is the same, whether the strips are actually on/off.

Kind Regards Bryan

Aircoookie, 2018-10-03:
Hi Bryan,
awesome that you love it! Makes me very happy :)
Thanks for the suggestion of the On/Off indicator in the mobile UI! I'll consider adding it in the future!

Aircoookie, 2018-10-04:
That's peculiar. Did a quick test and for me the timed light is working with both UIs.
Maybe the lights are taking too long to turn off for you to notice or the set target brightness doesn't differ enough from the initial brightness.
Please try to go to LED settings, set the duration to 1 minute and the target brightness to 0. Now the LEDs should turn off pretty quickly.
Also the timed light does not work while realtime modes (E1.31/Hyperion...) are active.
If that's not the issue, we need to troubleshoot further :)

Mariu86, 2018-10-06:
Thank you very much Aircoookie! It's a very useful code. I have the same problem like Bryan in the Mobile theme the on/off button and the timed light botton do not change color when you push it and you don't know if the light it's on/off if you are not in the room, in the web theme everything it's ok the button it's yellow when you push it!

Aircoookie, 2018-10-06:
Thank you @Mariu86 ! :)

Yes, I understand this is inconvenient. Because you both requested this feature, I've added the function that the power button in the mobile UI will indicate whether the LEDs are on or off for the upcoming 0.8.0 version. Also the nightlight button will have a similar indicator!

Keep in mind though that, unlike the classic UI, the mobile UI doesn't continuously update the light state from the ESP for performance reasons. So, if you change the light values via other means than the UI (API, UDP, other client), the changes will only display in the UI after a page refresh.

jolll-dk, 2018-10-10:
Hi again Mr Cookie :)

I got the timed light working by changing the firmware from esp32 to esp8266.
Thanks again for this wonderful piece of code - it's really better than any commercial
chinese junk I've tried so far. And that's by a factor 1000 :D

Aircoookie, 2018-10-16:
Wow, thank you! Makes me really happy to hear :)

I'm confused. Did you use the esp32 firmware on esp8266 (you mentioned earlier that you use esp-12f)? If so, i'd be quite amazed if esp8266 could work so well with a esp32 binary to the point that only small features like timed light won't work. I highly doubt it'd boot at all :D

jolll-dk, 2018-10-16:
esp-12f booted well with the esp32 firmware. Only thing I could find that did not work was timed light.
However now - with the right firmware - everything works 👍 

New version 0.80 - WOW really nice new effects and pallettes !! Amazing.
On/off button in mobile UI - thumbs up :)

One wish - with the amount of effects now, it would be nice if some could be disabled in the web interface.

Do you accept donations for your work?

Regards Bryan

Aircoookie, 2018-10-16:
Glad you love the new update!

Yeah, I agree. With all the new palettes and effects the list has become very long. Right now it is not possible to hide some effects, because the entire list is hardcoded. In the future I want to support custom user-added color palettes, which will require dynamic lists. Once I got them up and running, I can add a kind of favorite system, where you can select your favorite effects and have a shorter list containing only those you care about :)

If you are sure you want to support the project, there is a paypal link in the bottom of the readme ;)

kayrozen, 2018-10-17:
Thanks for the 0.8.0 update ! timed macros and MQTT are features i will use extensively. I'll chase bugs in the next few days. Thanks again :)

jolll-dk, 2018-10-17:
>  timed macros and MQTT are features i will use extensively.

Any tips on how to get started with timed macro's?
I did'nt find anything in the WiKi.

jolll-dk, 2018-10-17:
> If you are sure you want to support the project, there is a paypal link in the bottom of the readme ;)

Thanks :) One might one day learn to read all the way to the bottom :D

Aircoookie, 2018-10-17:
Timed macros are relatively basic right now :)
Firstly you need to know how to use macros themselves.
In "Time & Macro" settings you can set up to 16 different macros. Each one is an API request (there is a wiki page for the API, an example would be "win&T=1" for turning the light on)
Below that, you can find a list where you can set the hour, minute and the macro number to execute.
8 timer "slots" are supported, so you could have 4 intervals during the day where the light is on, for example.
At the moment, the macro will be executed at the set time every day, but weekday support will be added eventually. Keep in mind that you need to select your timezone and turn NTP on at the top of the page.
The ESP will take care of daylight saving time and stuff automatically.

Haha that's true :D I also almost never read the entire readme :)

kayrozen, 2018-10-17:
Just something (feature?) learned the hard way. Is this something wanted ?

In the Macro field, you cant save if your string starts with anything BUT a uppercase letter EX : T=0 not win&T=0 or just &T=0.

closed 2019-03-27 by Aircoookie