wled/WLED #53: Using a Sound Sensor?

bendobos, 2018-10-04:
Since this project uses W2812FX, which has support for sound sensors, I was thinking of using one. How would I go about doing that? 

I don’t know if this is at all possible but the ability to turn it on and off would also be amazing.

bendobos, 2018-10-05:
> Hi,
> even in WS2812FX, the "sound sensor capabilities" are very limited. All it can do basically is watch for a certain amplitude, and then progress the animation by a frame using the WS2812FX.trigger() function. Also this only makes sense or even works with very few effects like random color or multi dynamic. I agree a "clap on/off" would be way more useful. If you can tell me the sensor you have I could write you a piece of code for WLED that implements that function for you.
> If you are looking for a full blown music visualization, LedFx is a great option. It is a software you have to run on another PC or Raspberry, but interfaces nicely with WLED and makes for awesome real time effects!

“Clap on/off” is a really cool idea! The sensor I ordered is called KY-037.

Thanks for the tip for the sound visualization but it is a bit overkill for me.

Lazerjules, 2019-01-14:
Hi Aircoookie

As many have said, awesome work! This really is a great project. 
I wanted to +1 on bendobos suggestion of some audio interfacing, I would really like to be able to use a microphone connected to one of the analogue inputs and to be able to select some patterns which would produce a VU meter effect, I agree that many of the effects currently wouldn't fit with amplitude based progression. 
I have used [this](https://www.hackster.io/wannaduino/vu-meter-on-steroids-arduino-nano-and-neopixel-ws2812b-7f78e1) project on hackster which is great, perhaps if you feel it's within the scope of your project some sound triggered VU effects would be a nice inclusion? 
To be able to run your project as it stands but with these added would I think be very useful.

closed 2019-10-25 by Aircoookie