wled/WLED #54: Wrong Power Supply???

Mazdaracer121, 2018-10-10:
Im using this awesome software currently on a Wemos D1 Mini and connected my LED Strip on the 12V+ Power line of my PC Power Supply. Everything works Fine.

When i connect a other 12v 10 Amp Power supply, The Strip beginns to flicker and ignores the Commands from the Wemos D1.

Any ideas why?

Aircoookie, 2018-10-10:
Uh oh thats bad. Happened to me too, once. The WS2812B are 5V only. The Wemos has a voltage regulator that convert any voltage from 4 to 15ish volts to the 3.3V it needs to work. You probably fried the strip by now. You could perhaps rescue parts of it by cutting the beginning off. Sorry that this happened to you. I don't know why it should have worked with the PC 12V, you are probably referring to the (red) 5V line, which is the correct voltage. I hope you haven't killed too much of the strip mate :(

Mazdaracer121, 2018-10-15:
Hey. the Wemos i connect via USB Port, Dont worry :-)

But my LED Strip itself uses 12V. I DOnt Know why this is so. On The PCB of the Strip is Printed 12V, D0, and GND

wcbuerste, 2018-10-15:
Hi, so you are using a level shifter I guess? I had issues with some of them because they were to slow.

Cheers, David

Aircoookie, 2018-10-15:
Ok, this is weird. Does your strip have 1 IC for 3 LEDs? Then it might be a WS2811 strip which can accept 12V. Also, have you connected the Ground of the 12V supply and the ESP? If you haven't, the LEDs can't read the signal from the input pin. That may be why it worked for you with the PC because it has common ground with USB, unlike your 10A supply :) @wcbuerste a levelshifter shouldn't be necessary practically, even the 12v ones use 5V for input logic (and mostly work with 3.3V) :)

wiesendaniel, 2018-11-15:
A  level shifter might be the solution.

5V ws2812b have a  minimum input voltage at around 3.5V when powerd with 5V.
This can become a problem on its own when using a 3.3V uC like the esp8266.
It can work and in most cases it does but if you choose to make your wires a little too long and it's getting unreliable. This is due to the fact that the 3.5V tolerance is a theoretical/"guaranteed" number.
It's best to use a 3.3V <-> 5V level shifter to be stable.

In case of a 12V stripe which in fact exist you have a even bigger gab. I don't know there specs but I think it must be around a minimal input of 8-9V.

The behaviour you see might be some bits and bytes can be understood but most of them wont.

There is a little hack you can do at least on the 5v stripes. if you power the first led with lets say 4.8V (by leveraging the voltage drop across a diode or something.) The actual minimal input voltage is lowered as well to around 3.3V.  The output signal from the first led will be at 4.8V as this is the voltage the ic on the led is driven. 

The first led will be a bit dimmer, but it's getting reliable. Im not sure how this would work for 12V tho.

closed 2019-02-11 by Aircoookie