wled/WLED #55: e1.31

codehackr, 2018-10-11:
When i click on the checkbox to enable e.131 it does not save, has this feature been enabled?

Aircoookie, 2018-10-11:
Thanks for your bug report!
E1.31 is always enabled. The checkbox is to switch between Unicast (default for most occasions) and Multicast mode. Apparently there is a bug currently which causes that you can't enable multicast mode. I will fix it ASAP.

Unicast E1.31 should work as expected regardless of this.

codehackr, 2018-10-11:
Ah yes, in my case i really need multicast, thank you for fixing! Also sent you an email :)

codehackr, 2018-10-12:
I am unable to get unicast to work.. It shows in wireshark as the packet being rejected on port?

Aircoookie, 2018-10-12:
Sent you a mail :) The issue's cause is most likely that the ESP is running in AP mode.

Aircoookie, 2018-10-16:
I think you got E1.31 working now :) Please reopen if you experience any further issues with the topic

closed 2018-10-16 by Aircoookie