wled/WLED #58: Is possible to use 5050 RGB Led Strip instead of WS2812b ?

MopheusDG, 2018-10-15:
I have a long strip available for free and I'd like to know if it's possible to easily adapt this project to use that full strip of 5050 as at least 1 big RGB Led in this project ?

MopheusDG, 2018-10-16:
I do have a circuit already runnning with an small 5050 led strip (20 cm) and it was quite easy to build for that size, what I'm wondering if that's "adaptable" to the code easily somehow... what do you think ?

Thanks @Aircoookie !

Aircoookie, 2018-10-17:
Ok, it seems like the ESPs PWM might even be stable enough so we don't need another Arduino.
To see if your circuit is enough, look at the datasheets of the three transistors and find out their max. current (consistent drain). You'll probably need MOSFETs for this. Then you can ideally measure the current draw of your long 5050 strip on full white (if you can't, it should be about 1.5Amps/Meter) and see if it divided by 3 is lower than the max. current in the transistor datasheet.

Here is a small (untested!) example code on how to control the 5050 strip from wled. Connect R-G-B transistor source pins to GPIO 12-14 (D6, D7, D5 on Wemos D1 mini).
You just need to replace the NpbWrapper.h file with this version: https://gist.github.com/Aircoookie/fc0670a077eff1e7cfd4aa707b581214
I hope it works well. There could be flicker with the PWM, in which case we'd need to find another solution :)

MopheusDG, 2018-10-17:
It seems to be working perfectly fine ! The video is not very good with colors, but it's quite acurate. I need to replace the LM7805 that I'm using for power as it's getting hot, but the MOFET's, and the code seems to be working fine !

https://youtu.be/xQs4N_8qOuU

Thanks @Aircoookie, will make some tests later on a safer power supply (and my 5 mts strip).

MopheusDG, 2018-10-18:
For some reason, Alexa is not able to turn off the RGB 5050, she finds the device, it says "ok" but the leds stay ON (I think they go down in intesity and effects in progress get interrupted, but the lights stay on).

Any ideas ? By the way, what could be a good Windows software to make the leds change colors with the music or movies ? Tried Hyperion with Kodi but was too confusing...

By the way, great work with the WLED project man... it's very cool.

daksh1305, 2019-06-25:
The Wled project is really cool i have integrated the code on rgb5050 leds but alexa is not working.

Can anyone please help me with this.....????

Ukhauk, 2019-11-16:
Hi thanks for sharing this this awesome project Aircoookie. 
For info I have used the modified NpbWrapper as linked above with this shield Velleman KA01 RGB Shield Kit for Arduino 
https://www.vellemanformakers.com/product/rgb-shield-for-arduino-ka01/
I plugged the shield into a Wemos D1R2 https://www.instructables.com/id/Programming-the-WeMos-Using-Arduino-SoftwareIDE/
pins need changing in the NpbWrapper to suit the shield as follows :-
	  //init PWM pins
	  pinMode(D6, OUTPUT);//12
	  pinMode(D5, OUTPUT);//13
	  pinMode(D3, OUTPUT);//14

DenielTorres, 2019-11-29:
Hi guys! Awesome project, thanks a bunch Aircookie! I was just wondering about something: I have a Blitzwolf lt11 led strip which uses 5050 led chips. Currently, I'm running Tasmota on it and wanted to see if I can just change the firmware on the controller and it would work? 
Thanks very much in advance! 
Daniel

raresserban, 2019-12-14:
Hello, I am running a Sonoff L1 which uses an ESP chip, was wondering if I can flash it with WLED and have the 5050 strip still work. I saw that Tasmota is compatible but not sure about this.

Def3nder, 2020-01-01:
With release 0.9 the “solid RGB stripes” aka 5050 stripes are supported.

connectedoperations, 2020-01-01:
5050 refers to the led without an individually addressed chip?

Mariu86, 2020-01-01:
Yes, the entire strip has the same efect and color, this is an analog strip. The 5050 it's in fact the size of the led 5mmx5mm.

tonyn0, 2020-01-04:
I just tried that wrapper in the latest version and it's not compiling. Any ideas?

exit status 1
'NeoEsp32RmtWS2813_V3Method' was not declared in this scope

Aircoookie, 2020-01-04:
@tonyn0 you can try and replace the method name with `NeoEsp32Rmt0Ws2812xMethod` and it should compile. I will update the gist.

That said, I do not recommend using it anymore as the latest WLED source offers native support.
Just uncomment `#define WLED_USE_ANALOG_LEDS` in line 8 of NpbWrapper.h :)

tonyn0, 2020-01-04:
Thanks. I thought I had the latest but did not!

tonyn0, 2020-01-04:
Umm. Can you check the source ZIP? Line 8 is not what you say it should be.

Aircoookie, 2020-01-04:
Did you use the "Clone or download" button in the top right of the main repo page?
The line definitely is correct:
https://github.com/Aircoookie/WLED/blob/master/wled00/NpbWrapper.h#L8

tonyn0, 2020-01-04:
User error! :D Thanks. All good now.

Aircoookie, 2020-01-06:
@tonyn0 awesome! Would you mind sharing? I was going to add analog LED support for ESP32 using ledc, but didn't have the time yet :)

tonyn0, 2020-01-06:
@Aircoookie I'm brand new to Github, so I need to figure out how. ;)

Def3nder, 2020-01-06:
> @tonyn0 awesome! Would you mind sharing? I was going to add analog LED support for ESP32 using ledc, but didn't have the time yet :)

Hi Aircoookie,

I added support for ESP32 for both IR-Receiver and solid 5050 RGB(W) stripes last weekend.
I will create a PR in the next days. 

Would you like to have this separated in two PR's ?
...one for the solid RGB stripes using ledc 
...one for IR-Remote ?

(the IR-remote replaces the "on/off" switch in the config by an integer number and enabled different IR-Remote types (0=off, 1=1st IR-remote, 2=2nd IR-Remote, etc.) so we can switch between different types without recompiling.

Aircoookie, 2020-01-06:
@Def3nder that sounds good!
More granular pull requests are welcome, that makes it easier for me to review and approve the changes, which means less work and a quicker merge :)

Smokrow, 2020-01-14:
> With release 0.9 the “solid RGB stripes” aka 5050 stripes are supported.

Do you know how I can activate this?

Def3nder, 2020-01-15:
Hi @Smokrow ,
look in the platformio.ini file - there are some build-flags:
- WLED_USE_ANALOG_LEDS
- WLED_USE_H801 
- WLED_ENABLE_5CH_LEDS

... if you have problems, just tell me your PINs for R,G,B and W (if you have an RGBW strip) and your Button-PIN (if used) then I compile a release for you.

Smokrow, 2020-01-16:
Hi @Def3nder.

This worked like a charm :) Thank you

tonyn0, 2020-01-31:
I plan on adding support for four, RGBW channels based on the _rough_ plan below, unless someone knows in advance that it _won't_ work. ;) I'm using ESP32s so just starting with that.

_wled00.ino_
#ifdef WLED_USE_ANALOG_LEDS
	//define led pins, RGBW order, channels 0-3
	byte ledpin[16] = {15,2,5,4, 5,18,19,21, 13,12,14,27, 26,25,33,32}
#endif

_npbwrapper.h_
void Begin(NeoPixelType type, uint16_t countPixels)
#ifdef WLED_USE_ANALOG_LEDS 
  #ifdef ARDUINO_ARCH_ESP32
      for( byte i = 0; i < 16; i++) {
	ledcSetup(i, 5000, 8);
	ledcAttachPin(ledpin[i], i);
	}

void SetPixelColor(uint16_t indexPixel, RgbwColor color)
#ifdef WLED_USE_ANALOG_LEDS
	if (indexPixel >3) return; //set analog LEDs from first four pixels

void SetRgbwPwm(uint8_t r, uint8_t g, uint8_t b, uint8_t w, uint8_t w2=0, uint8_t i)
#ifdef ARDUINO_ARCH_ESP32
	ledcWrite(i*3, r);        //RPIN
        ledcWrite(i*3+1, g);  //GPIN
        ledcWrite(i*3+2, b);  //BPIN
        ledcWrite(i*3+3, w);  //WPIN  ** handle greyscale conversion
#endif

Def3nder, 2020-01-31:
Hi @tonyn0, this will break 5-channel support 😢 
...and you would get the first 4 pixels mapped to three strips.

My suggestion would be to **use the first pixel of each segment as "analog color"**.
So if we have 4 analog strips, you would define the 16 PINs for RGBW (or 20 PINS for RGBWCT) and create 4 segments. This way could can run different effects on each segment.

With your idea you could define 4 segments, too, **but** the effects will not work on segments with a `SEGLEN of 1`.

To get good results you with analog LED strips the segment needs to be at least 30 pixel or bigger.

tonyn0, 2020-02-01:
Eh, I can't even see myself using RGBW. A good app negates the need. It may be more of a pain, but I didn't mind too much with another system that I used. This may be why I bumped an issue mentioning color correction. ;) With prices dropping, RGBW may have already come down to the level of RGB, though.

How am I grabbing the first four pixels instead of three? Isn't indexPixel each LED?

Def3nder, 2020-02-01:
sorry, I meant "you get the first 4 pixels mapped to four strips" 😄 
Indeed, an analog RGBW strip with `4-in-1-LEDs` does cost about 5 USD for 5m (if you have some time - delivery takes about 4 to 5 weeks): look [here](https://aliexpress.com/item/32622948477.html).

And: the "white" color is so much warmer (if you grab the `RGBWW version`) then normal RGB strips.
(and: the `12V strip` does not has problems with white's getting "yellowish" when strip is longer than 3m)

tonyn0, 2020-02-01:
Stop thinking RGB white needs to be equal values. This is what color correction is for. ;)

Def3nder, 2020-02-02:
yes, you are right. <but<. this needs to be implemented and right now it isn't.

In an other thread with Aircookie we are just discussing the use of a new color model.
This would be R, G, B, W, Color Temperature, Light-capabilities.

Then you could enter the capabilities of your light like "RGB-only", with white at 2700k or with dual white at 2700k and 6500k and so on.

The UI would get an additional slider beside white: color temperature.

But this is future music 😄

tonyn0, 2020-02-02:
Yes, I can work on that as well if no one else does.

That definitely needs color correction. ;)

tonyn0, 2020-02-03:
I've got RGB working, but just the first output. Some debug printing shows I'm only seeing an indexPixel of 0 in SetPixelColor (I expect to see 0-3). =/

pniewiadowski, 2020-02-03:
Can some one show me the connection diagram? How I can hookup these LEDs to wemos?

raoulteeuwen, 2020-02-03:
Hi @pniewiadowski . No idea what you already know or what you already tried to research yourself. Nor what your configuration is. You can find diagrams here: https://www.google.com/search?q=wemos+d1+led+strip+diagram&source=lnms&tbm=isch&sa=X&biw=1543&bih=846 . Based on your question it might be needed to start with some basic knowledge: i've listed sources (video's) that helped me get introduced into the subject that might help you as well: https://raoulteeuwen.blogspot.com/2019/08/starting-with-led-strips.html .

colder1989, 2020-05-15:
> With release 0.9 the “solid RGB stripes” aka 5050 stripes are supported.

First of all, thank you for your fantastic work ... I had a question for connecting the 5050 LEDs (B + G + R + 12V) .. Do I need transistors?  what are the default pins?  should i edit the platformio.ini file?  is it not possible to connect an IR transmitter to the Pin and control the LEDs?  Thanks again

sabag, 2020-06-18:
So. do we have RGB "analog leds" support in the 0.10.0 version ? I am asking because I am trying to get the changed file into the source but it fails to compile (probably many changes since it was written).

Basti149, 2020-07-09:
> 
> 
> So. do we have RGB "analog leds" support in the 0.10.0 version ? I am asking because I am trying to get the changed file into the source but it fails to compile (probably many changes since it was written).

Laut der Firmware kann man jetzt 4 pin LEDs also RGB auswählen. Habe ich gerade gesehen. Gucke mal unter den LED Settings.

closed 2020-09-27 by Aircoookie

DennisXK, 2020-12-15:
hello I would like to connect an analog RGB band to the WLED. Which hardware do I need exactly. N Chanel or or P Chanel Fets. could someone provide a small circuit board. thanks

zigi-dz, 2021-02-22:
any boddy can help me i'm new and i wanna know hot to modify the folder of wled to adjut it for led 5050

n1md4, 2021-03-21:
> hello I would like to connect an analog RGB band to the WLED. Which hardware do I need exactly. N Chanel or or P Chanel Fets. could someone provide a small circuit board. thanks

I used this https://learn.adafruit.com/rgb-led-strips/usage with an external 12v supply all LEDs illuminate.

n1md4, 2021-03-21:
> any boddy can help me i'm new and i wanna know hot to modify the folder of wled to adjut it for led 5050

I would love to know this too.  I have read quite a lot about users with 5050 strips, even to the point of noticing they're officially supported now.  

I flashed WLED_0.11.1_ESP8266.bin, connected the the UI so the ESP part is working.  The LEDs are being powered too, as mentioned above, but I am unable to change any LED over the UI. 

Please can some give an example of exactly what I should do to get this working.  I have 2 sets of these lights sitting around doing nothing, it would be a shame for them to end up in landfill :(

blazoncek, 2022-12-27:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
The sample schematics are available on [KB](https://kno.wled.ge) as well.

blazoncek, 2022-12-27:
Oh, come on!

It's on the Getting started page.
[Getting started](https://kno.wled.ge/basics/getting-started/)

gasperzupancic, 2023-01-02:
@blazoncek well it is for all the ws* strips, but not for the basic 5050 strip. Somehow i thought better and got 5050 led strip myself and then found this thread beacuse i wasnt sure how to connect it:)

blazoncek, 2023-01-02:
@gasperzupancic you just need to add MOSFETs. Basic schematics is on KB as well.

Dogu105, 2024-03-21:
I am using a non-addressable 5050 RGB LED strip, I set up the MOSFET circuit and tried using PWM RGB and the pins, but I couldn't get any results. I don't know what I'm doing wrong, I've looked at old topics but couldn't find anything useful that would work with the current version of WLED. Can you help me?

blazoncek, 2024-03-21:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.