wled/WLED #59: WLED-master in keine Header-Dateien (.h)

Cable1980, 2018-10-16:
Hallo alle zusammen ich habe beim kompelieren das Problem das mir die Arduino IDE die Fehlermeldung bringt das im Ordner keine Header-Dateien vorhanden sind !
Ich möchte das Projekt auf einem ESP 8266 umsetzten  . Ich bin in dem Bereich noch neu dabei was mache ich fasch ?

Aircoookie, 2018-10-16:
Hi, danke für dein Interesse an WLED!
Vielleicht sind nicht alle benötigten Dateien zum Kompilieren vorhanden. Hast Du die zip runtergeladen, extrahiert und dann wled00.ino geöffnet? Wenn nicht, sende mir bitte die genaue Fehlermeldung, damit ich weiß, welche Datei fehlt. Ich nehme an, dass entweder der ESP8266 Arduino core, NeoPixelBus oder FastLED fehlen. Diese drei Bibliotheken müssen für die neueste Version vorhanden sein. Ich hoffe, dann wird es laufen!

[English Summary]
This issue is about some header files missing in the IDE. The cause is most likely a library that is not yet installed (Arduino Core, NeopixelBus or FastLED).

Cable1980, 2018-10-16:
Es Läuft Supper Sache !!! Daumen hoch und vielen dank es waren alle Bibliotheken vorhanden habe nur Arduino Core neu installiert  !  :+1:

closed 2018-10-16 by Cable1980