wled/WLED #60: Reverse mode won't work with analog clock in 0.8.0

kayrozen, 2018-10-17:
Hi ! 

In 0.8.0, if i apply the 180deg reverse mode AND activate the analog clock function, Pixels aren't changing at all. In normal mode, everything is ok, clock is working as intended. 

In my investigation, it probably is something in WS2812FX.ccp in the setPixel on line 100 changes that modified the behavior. I'm not able to find the exact cause right now.

Aircoookie, 2018-10-18:
Hi! Thank you! I could reproduce the issue instantly. Just now I pushed the fix to the dev branch :)

You were right that setPixel was the culprit. The clock pixels are "locked" so that effect updates don't "draw" over them. The problem was that the line that flips the pixel index for reverse mode was above the line that checks if a pixel is "locked", so the software assumed incorrect pixels to be locked and drew over the ones that were meant to be locked,

Thank you once again! These bug reports are very valuable to me, since I can't really test every possible combination of settings/features working together for each version completely.

Aircoookie, 2018-10-18:
Whoops, accidentally hit close issue button :)
I've found  and fixed another related issue. If you toggle back from clock overlay mode to no overlay, the clock pixels will stay locked until a reboot. Also fixed in dev branch just now.

Mariu86, 2018-10-19:
Another problem with the clock when I turn off the strip and turned back on my strip turns another pixel on +1 and if I repeat on/off process will ad another pixel +2 , I have a 300 pixel strip and the clock its set form 0 to 120 pixel if I turn on/off the strip will be 122 pixels, the 2 pixels will stay on all the time even if I turn on/off the strip.

Mariu86, 2018-10-19:
It's my bad , when I updated to 0.8 I did not check the nr of LEDs in led setup there was 110 LEDs and in the time & macros was 120, but now I found a problem that I have now and in 0.71 version, if I move the first led in the time settings the 5 min marks are moving but the hours/min/seconds marks starts like nothing changed.

kayrozen, 2018-10-20:
OH ! I don't know if it's related but i also have another bug in the OL analog clock. 

In one of my clocks that has 222 leds , the clock layer is between led 30 and 192. Clock is ok but if i activate 5mins marks, it starts them from 0 not 30.

Aircoookie, 2018-10-20:
Great find! I can't currently release a fix since I'm in the middle of additions to the effects library.
Still, I believe I have found the issue. You can try to fix it by replacing line 175 in wled11_ol.ino:
Replace `if (pix > overlayMax) pix = pix - overlayMax;` with:
`if (pix > overlayMax) pix -= overlaySize;`
I hope that'll fix the issue for you. The 5min marks somehow still assumed overlayMin == 0.

PS: Stunning clock you got there! It's epic what you all create using my software :)

kayrozen, 2018-10-21:
it fixed it. but now i'm running into another problem haha. 

Since it's 162/12, it gives 13.5 leds. it rounds up wierdly. So marks don't really fit the layout. and it puts 13 of them. 

I'm fiddling with this line (174) but no success for now : 
`pix = overlayMin + analogClock12pixel + (overlaySize/12)*i;`