wled/WLED #61: narrowing conversion error on Compile

DaveDischord, 2018-10-18:
I've gotten an error on compile, running Arduino 1.8.7, attempting to load on an ESP32Thing.  Any idea what I could be missing/not doing?  Its a fresh install on a new PC, so it could be anything.
Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "SparkFun ESP32 Thing, 80MHz, Default, 921600, None"

sketch\WS2812FX.cpp:44:22: warning: extra tokens at end of #include directive

 #include "palettes.h";

                      ^

In file included from sketch\WS2812FX.cpp:42:0:

WS2812FX.h:472:5: error: narrowing conversion of '1000u' from 'uint16_t {aka short unsigned int}' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

     };

     ^

WS2812FX.h:472:5: error: narrowing conversion of '16711680u' from 'uint32_t {aka unsigned int}' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

In file included from sketch\WS2812FX.cpp:42:0:

sketch\WS2812FX.h: In constructor 'WS2812FX::WS2812FX()':

sketch\WS2812FX.h:271:26: warning: large integer implicitly truncated to unsigned type [-Woverflow]

       _segments[0].speed = DEFAULT_SPEED;

                          ^

In file included from sketch\WS2812FX.cpp:43:0:

C:\Users\dturney\Documents\Arduino\libraries\FastLED/FastLED.h: At global scope:

C:\Users\dturney\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.001

 #    pragma message "FastLED version 3.002.001"

                     ^

In file included from C:\Users\dturney\Documents\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from sketch\WS2812FX.cpp:43:

C:\Users\dturney\Documents\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

sketch\WS2812FX.cpp: In member function 'void WS2812FX::resetSegments()':

sketch\WS2812FX.cpp:315:110: warning: large integer implicitly truncated to unsigned type [-Woverflow]

   setSegment(0, 0, 7, FX_MODE_STATIC, (const uint32_t[]){DEFAULT_COLOR, 0, 0}, DEFAULT_SPEED, 128, NO_OPTIONS);

                                                                                                              ^

sketch\WS2812FX.cpp: In member function 'uint16_t WS2812FX::mode_oscillate()':

sketch\WS2812FX.cpp:1851:20: warning: narrowing conversion of '(((((int)((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::stop) - ((int)((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::start)) + 1) / 4)' from 'int' to 'int16_t {aka short int}' inside { } [-Wnarrowing]

     {SEGMENT_LENGTH/4,   SEGMENT_LENGTH/8,  1, 1},

                    ^

sketch\WS2812FX.cpp:1851:40: warning: narrowing conversion of '(((((int)((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::stop) - ((int)((WS2812FX*)this)->WS2812FX::_segments[((int)((WS2812FX*)this)->WS2812FX::_segment_index)].WS2812FX::Segment::start)) + 1) / 8)' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]

     {SEGMENT_LENGTH/4,   SEGMENT_LENGTH/8,  1, 1},

                                        ^
[... 1079 lines cut ...]
D:\Development Program Files\WLED\wled00\wled12_alexa.ino:154:42: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

             "<friendlyName>Philips hue (");

                                          ^

D:\Development Program Files\WLED\wled00\wled12_alexa.ino:163:29: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

             "<serialNumber>");

                             ^

D:\Development Program Files\WLED\wled00\wled12_alexa.ino:166:49: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

             "<UDN>uuid:2f402f80-da50-11e1-9b23-");

                                                 ^

D:\Development Program Files\WLED\wled00\wled12_alexa.ino:187:20: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

           "</root>");

                    ^

narrowing conversion of '1000u' from 'uint16_t {aka short unsigned int}' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

Aircoookie, 2018-10-18:
Seems like this new compiler version is picky and doesn't like:
- implicitely truncating uint16_t to uint8_t (narrowing conversion)
- casting from a string constant to a char*

Tomorrow I'll update my IDE and ESP32 core and see if i can reproduce the issue. Most likely these errors will go away if I explicitly cast them to the correct data type, but that would be really annoying :)

debsahu, 2018-11-13:
I was going to create an issue, but found out the issues:

1. speed is _uint16_t_ (not _uint8_t_) in **WS2812FX.h**:
```
  public:
    typedef struct Segment { // 21 bytes
      uint16_t start;
      uint16_t stop;
      uint16_t speed;
      uint8_t intensity;
      uint8_t palette;
      uint8_t  mode;
      uint8_t  options;
      uint32_t colors[NUM_COLORS];
    } segment;
```

2. WS2812FX is heavily modified, meaning the default parameters must be passed correctly in **WS2812FX.h**:
```
    segment _segments[MAX_NUM_SEGMENTS] = { // SRAM footprint: 20 bytes per element
      // start, stop, speed, intensity, palette, mode, options, color[]
      {0, 7, DEFAULT_SPEED, 128, 0, FX_MODE_STATIC, NO_OPTIONS, {DEFAULT_COLOR}}
    };
```
@Aircoookie Let me know if you want me to create a PR.

Aircoookie, 2018-11-13:
Hi @debsahu thanks for your comment!

About 1., speed is actually _uint8_t_ in WLED. It is a relative value where 0 is slowest and 255 fastest. This was how it was originally in WS2812FX. Later the creator decided to switch to _uint16_t_, where it is the time in ms to complete one animation cycle. On the surface, that makes a lot of sense, however I decided to keep using uint8_t for several reasons: 
1. my protocols would have needed to be changed in a way that's not backwards compatible 
2. it is counterintuitive that a higher value is a slower speed. 
3. Take 15000ms. This would be really fast for some effects and awfully slow for others. Relative values give the individual effect control over the actual ms delay for each step.

2. Great find! My next commit will include the patch. This is what probably causes the ESP32 compile error in some (PIO) environments.

debsahu, 2018-11-13:
@Aircoookie I feel the pain of WS2812FX! We then need to change `#define DEFAULT_SPEED      (uint16_t)1000` to `#define DEFAULT_SPEED      (uint8_t)256` (Just put 256 here, you can use an ideal value) I am coming from https://github.com/toblum/McLighting to here. Let me know you want me to port anything from there to here. Ask for any help you need, if I have time I will work on it. Very interesting project here.

Aircoookie, 2018-11-14:
Thank you! Great that you like my project and that you offer me your help! I can manage the Arduino version right now, but i can see you already put a lot of work into getting WLED to build in PIO! Native PIO support would be amazing, but I still want it to run on Arduino, because that is more accessible for beginners and simple. The obvious solution would be to have another branch/repo which is PIO-optimized, but then any changes made in the Arduino version would need to be manually added to the PIO version, which is too cumbersome. Therefore, it'd be ideal if the master repo also supported PIO. I think we could do this by adding the `platformio.ini` and migrating most .ino files to .cpp/.h . Still, the separation of /lib and /src would not be possible because Arduino IDE apparently can't include local libraries which are not in the /src folder. Do you have any ideas about the feasibility of one branch for both IDEs?

About McLighting, wow, you have contributed to that project a lot! Can you personally think of some McLightning features you'd like to also see in WLED? Porting might be an issue since McLighting is (L)GPL and WLED is MIT, so it can't use GPL code without also being GPL or having an exclusive license. I don't want to switch to GPL, but instead maintain the more permissive MIT license for various reasons.

debsahu, 2018-11-14:
1. You already have it on your todo list: color temperature for white light: cool to warm
2. Home Assistant(HA) Integration
3. Maybe related to 2, switch to async MQTT library to be able to easily send light configuration to HA

> About McLighting, wow, you have contributed to that project a lot! Can you personally think of some McLightning features you'd like to also see in WLED? Porting might be an issue since McLighting is [...]

@toblum What are your thoughts on licensing portions of code to WLED project?

toblum, 2018-11-17:
Hi @Aircoookie and @debsahu,

honestly I didn't think to much about the licence when starting McLighting two years ago. Personally it's completely OK for me here that this code is used by others as it's a community project. So for me a switch to MIT would be OK as long it's also OK for you @debsahu. 

BTW: I didn't have had a look at WLED for over a year but it looks like a great project. I like the numeroud protocols you implemented and the good documentation. I will also give it a try. It is a pity that the project is so little visible on the "usual places".

Best regards
Tobias

Aircoookie, 2018-11-17:
Hi @toblum !

awesome to see you here! I'm glad to hear that you'd consider allowing us to use some of the McLighting code in WLED. Of course I'll also credit you and @debsahu as contributors.

Thank you! Yes, WLED has evolved a lot recently. The FastLED palette effects really made a difference! Don't worry, it is surely gaining traction. A year ago I had an estimated 10 users, now already around 300-500. Looking forward to hearing your thoughts about the recent version!

Best regards
Christian

toblum, 2018-11-20:
@debsahu If it's OK for you, I can also change the licence in the repo to MIT.

debsahu, 2018-11-20:
@toblum fine with me!

toblum, 2018-11-20:
OK, done. McLighting is now MIT licensed.