wled/WLED #63: UDP Realtime Control with Python

greenhulk, 2018-10-21:
Is there a small Python sample script to manipulate single pixels?

I've built this easy example here, but i don't know what the message to the UDP Realtime Control must look like:

```
import socket

UDP_IP_ADDRESS = "192.168.8.166"
UDP_PORT_NO = 21324
Message = **"Hello, Server"**

clientSock = socket.socket (socket.AF_INET, socket.SOCK_DGRAM)
clientSock.sendto (Message, (UDP_IP_ADDRESS, UDP_PORT_NO))
```

Greetz
Andy

Aircoookie, 2018-10-22:
Hi Andy,

I'm sorry that I can't give you an example that I can guarantee will work, since I'm not proficient in Python.
However, your code is 90% correct. Basically, you'd just need to send a byte array instead of a string.
You can find the full documentation here :https://github.com/Aircoookie/WLED/wiki/UDP-Realtime-Control

The gist of it is though that the 1st byte sets the protocol, the 2nd the time in seconds to show the realtime lights before exiting to normal mode. After that, the pixel colors are sent. This is the easiest
in protocol 2 (DRGB) where you just send the RGB values chained. An easy example in HEX would be `02 02 FF 00 00 00 FF 00 00 00 FF` which will set the first pixel to red, the second to green and the third to blue.

In Python it could look something like this (I'm not sure if it'd work though)
```
import socket

UDP_IP_ADDRESS = "192.168.8.166"
UDP_PORT_NO = 21324
v = [2, 2, 255, 0, 0, 0, 255, 0, 0, 0, 255]
Message = bytes(v)

clientSock = socket.socket (socket.AF_INET, socket.SOCK_DGRAM)
clientSock.sendto (Message, (UDP_IP_ADDRESS, UDP_PORT_NO))
```
I hope you can get this to work!

greenhulk, 2018-10-22:
Thanks, that was exactly the right hint!
I just had to adjust the code a bit, now it works!

```
import socket

UDP_IP_ADDRESS = "192.168.8.166"
UDP_PORT_NO = 21324

v = [2, 2, 255, 0, 0, 0, 255, 0, 0, 0, 255]
Message = bytearray(v)

clientSock = socket.socket (socket.AF_INET, socket.SOCK_DGRAM)
clientSock.sendto (Message, (UDP_IP_ADDRESS, UDP_PORT_NO))
```

closed 2018-10-25 by Aircoookie