wled/WLED #64: Random white flashes - ESP8266

kayrozen, 2018-10-22:
Since 0.8, when i select a FastLED pattern and palette, leds sometime flashes 1 frame white. 

I can reproduce it on a Wemos D1 and a NodeMCU.

Aircoookie, 2018-10-22:
Hi, does it only happen while you are selecting a new effect/palette or occasionally at random while the effect is active? Can you tell me a combination of effect and palette where the issue happens?

kayrozen, 2018-10-22:
occasionally at random while the effect is active. It's Noise 16-1.

Aircoookie, 2018-10-24:
I've finally seen it today. However it's very rare (only happened twice with 2 hours in between) and I couldn't figure out the reason yet. There are two main potential causes I can think of: First one is an overflow of a helper variable that causes undefined behavior for a frame (during wrap-around).
Second is that the ESP tries to update the LEDs before the previous update cycle was completed. This is likely mostly on larger strips since the fixed 50fps refresh rate on FastLED effects is quite fast. However, if this was true, the flashes should happen more frequently on larger strips - which I was unable to reproduce.

Aircoookie, 2018-10-25:
If you want to help investigating, try if decreasing the FastLED fx refresh rate helps. When you scroll all the way down in `WS2812FX.cpp` you can see the `return 20;` at the end of each FastLED effect function. This means the LEDs are updated every 20ms - 50 times per second. You could try and increase the return value of one of the effects (like Noise 16 1), and see if the issue goes away when you change it to 33 (30fps) or 50 (20fps). This will make the animation slower, but for testing that doesn't matter. I remember I once had a LOT of white flashes when i implemented the fade transistions - it was because I accidentally updated the LEDs twice in a row. How often do they happen in your case? Every few seconds or only a few times per hour?

Thank you for offering to help me, since my longest testing strip only has 84 LEDs!

Aircoookie, 2019-10-25:
It seems like this is #230 :)

closed 2019-10-25 by Aircoookie