wled/WLED #65: MQTT should report when it connects to the broker

kendersec, 2018-10-30:
I'd like to turn other things on when the ESP with WLED is turned on and connects to the broker. Espurna seems to ocassionally report the status of the system and also when first connected. Would you consider doing something similar?

This is what I get from Espurna:

> courtyard/floodlight/app ESPURNA
courtyard/floodlight/version 1.13.2
courtyard/floodlight/board ITEAD_SONOFF_BASIC
courtyard/floodlight/host COURTYARD-FLOODLIGHT
courtyard/floodlight/ip 192.168.0.202
courtyard/floodlight/mac B4:E6:2D:3A:BF:E6
courtyard/floodlight/rssi -89
courtyard/floodlight/uptime 596446
courtyard/floodlight/datetime 2018-10-30 21:55:18
courtyard/floodlight/freeheap 23488
courtyard/floodlight/relay/0 1
courtyard/floodlight/vcc 3187
courtyard/floodlight/status 1
courtyard/floodlight/loadavg 1

Aircoookie, 2018-10-30:
Hi, by "report" do you mean that the ESP should publish to a certain topic that it is connected now?
That would actually make a lot of sense. If it is enough for you if WLED publishes the current state to the `/device/g` and `/device/c` topics and your other things subscribe to those topics, you can just add `publishMQTT()` below line 83 (`mqtt->subscribe(subuf);`) in wled17_mqtt.ino and it will not only update when it changes color/brightness, but also when it initially (re)connects to the broker.

kendersec, 2018-10-30:
Yes, I think publishing current state to /device/g and /device/c on boot would work/make sense. Thanks!

Aircoookie, 2018-11-18:
This feature should work as intended now. If there are any further issues, feel free to reopen!

closed 2018-11-18 by Aircoookie