wled/WLED #66: request fastled effect

Toon-nooT, 2018-11-02:
Hi,
i've been trying out several web ESP8266 interfaces for my WS2812B strings. So far i like WLED best. Cheers!
The only thing i miss is this very nice [twinkling 'holiday' lights that fade in and out](https://gist.github.com/kriegsman/5408ecd397744ba0393e) from Mark Kriegsman . In my opinion, it's much better than the twinkling from WS2812FX.
It's using fastled and i managed to implement it and get it working in eg [McLighting](https://github.com/toblum/McLighting) but not in WLED.
So my question is: is it possible to include this effect?

Aircoookie, 2018-11-02:
Hey,
awesome that you like WLED!
Yeah, I agree sadly it's not very intuitive to add new effects to WLED, even less FastLed ones. I'll try and add Kriegsman's effect to WLED for the next version! Even then, I'll keep this issue open so you or others can request more cool effect modes you might find that are not yet in WLED.

Toon-nooT, 2018-11-05:
i just tried the dev branch and it works great! Thanks for including it. Happy that i'm not alone enjoying this mode. I will look at the modifications you had to make and try to include some more modes.

RaiderXXL, 2019-01-09:
Hi All,

love your project! 

I have a Hyperion LED that has different mood-blob effects.

Is it possible to import them in your project also?

Greetings from Germany

Pascal

Aircoookie, 2019-01-09:
Hi Pascal,

thank you for your gift :) Awesome that you enjoy WLED!
I also really like the Hyperion mood blob effects and i think it shouldn't be too hard to add a port of them to WLED 0.9.0!
In the meantime, you can use the "Lake" effect with slow-medium speed and different palettes, it should create a similar atmosphere.

RaiderXXL, 2019-01-09:
Hello,

;-) will try this thanks a lot...

other question: what does the 2 curved arrows??? ;-)

Grüßle

Aircoookie, 2019-01-09:
The two circular arrows are for the sync feature. If you have more than 1 ESP8266 running WLED, they'll sync up and display the same color. If the top arrow is lit, the current ESP will receive updates from other ESPs, if the bottom arrow is lit it will send them itself if the color is changed :)

RaiderXXL, 2019-01-09:
WOW that's possible?!? great!
thank you !

RaiderXXL, 2019-01-11:
New Suggestion, dont know how  to tell in english... it would be greate if i could use the FastLed Palette without a animation.... did you understand? 😜 multible Color on the stripe without animation... 

Greetings 

Pascal

Aircoookie, 2019-01-11:
Not a problem :) It's already supported, just select the Palette effect and put the speed slider to 0!

RaiderXXL, 2019-01-12:
So simple... thanks! 
Nice Weekend!

Pascal

RaiderXXL, 2019-01-25:
Hi, have tested ... but there is still a movement with speed 0 :-( 
Is it possible to use the Palette with the effect Solid? 

Greetings
Pascal

(Ich liebe dein Projekt immer mehr!)

Toon-nooT, 2019-02-07:
Suggestion to include this Fastled effect with palette: https://github.com/atuline/FastLED-Demos/blob/master/confetti_pal/confetti_pal.ino

SeriousOri, 2019-05-04:
Hello AirCoookie I use your WLed every day.
I have it in the IKEA Fado with 24 LED's.
I noticed that the fire effect is not well done.
I saw a very nice on YouTube and tasted on an Arduino nano.
Unfortunately, I do not know how to add it to WLED.
Is there a way to add that to your closest update?

[https://github.com/RoboUlbricht/arduinoslovakia/blob/master/neopixel/neopixel_fire01/neopixel_fire01.ino](url)

Aircoookie, 2019-05-04:
@SeriousOri yes I know there are better Fire FX out there then the one included in WLED right now. You can vary the speed and intensity, but I agree it doesn't really look quite right. I'll take a look at the effect you posted. If I like it, it will be included in the next update :)

SeriousOri, 2019-05-05:
Hi.  I'm sure you'll like it.

Aircoookie, 2019-11-26:
Closing. You may open individual issues for effect suggestions :)

closed 2019-11-26 by Aircoookie