wled/WLED #67: Why only gpio2 on ESP8266?

gent99, 2018-11-04:
Is there a reason, why we have to use gpio2 on ESP8266? On my WMOS D1 MINI I'd like to use D1 (wich is gpio5). Could i change wled05_init near line 244 to use LEDPIN also on ESP8266?

Kind regards

Aircoookie, 2018-11-04:
Hi,
yes, the reason is i am using the UART serial hardware interface to push the data to the LEDs. This is said to improve reliability and reduce flicker. UART is only supported on GPIO2 though. There is another hardware method called DMA, which is only available on GPIO3. You can also use software bitbanging, which is availabe on any pin. Be aware though that it isn't really tested and may cause flicker.

To enable this, set LEDPIN in NpbWrapper.5 to 5 and change the PIXELMETHOD in line 22 of NpbWrapper.h to "NeoEsp8266BitBang800KbpsMethod"

Ga1j1n, 2018-11-09:
So D0 on the D1 mini should read #define LEDPIN 3 ?

Aircoookie, 2018-11-09:
I don't know if D0 would work at all, but if it did, it'd be GPIO 16. define LEDPIN always uses the GPIO number, not the Wemos/NodeMCU "D" numbers, which differ. GPIO 2 is D4, GPIO 3 is RX, GPIO 4 is D2, for example. You can see the whole pinout [here](https://community.blynk.cc/t/virtual-pin-referencing-physical-didnt-work-can-help/22354). You can also do `#define LEDPIN D0`, in that case the arduino IDE will select the correct gpio for you IF you selected the right board (Wemos D1 mini)

About "#pragma message "No hardware SPI pins defined. All SPI access will default to bitbanged output" this stuff is from the FastLED library and it always prints. I'll try and define some (unused) pins, so FastLED doesn't throw this (in our case) useless message anymore.

Ga1j1n, 2018-11-09:
My Bad,

I thought D0 was GIO3 and GOI2 was unavailable on a D1 mini, changed pin to D4 GPIO2 now.

I going to use your excellent project to make a 3 sided 64 WS2812 RGB cube assuming I can figure out a way to supply almost 4A 5V per panel from 2 Li-Ion cells, do you use level shifters  for your ESP32 based products or will they drive directly?

Aircoookie, 2018-11-09:
Well, i don't offer any "ESP32 based products" to buy, but for my personal projects, both ESP8266 and ESP32, I have not used a levelshifter in any of them without any problems (and I built like 25 of them). If you want to connect 3 panels in parallel (all to the same pin) you might run into issues. In that case, take a single WS2812 LED and connect it to the ESP. Then, connect the inputs of the three panels to the output of the pixel and check "Skip first LED" in WLED LED settings. If you want to chain the panels one after the other, that shouldn't be a problem.

About the batteries, you need voltage regulators and protection circuits to get a stable 5V. Even then, I'm not sure if you could achieve 4A with two cells. I'm no expert in this regard.

debsahu, 2018-11-14:
For RMT use in ESP32, it might be better to follow https://github.com/Makuna/NeoPixelBus/tree/Esp32RMTExperimental than https://github.com/svenihoney/NeoPixelBus

Aircoookie, 2018-11-18:
@debsahu I will try out that branch and see if it performs differently than the svenihoney branch. If I don't notice any ill effects I agree we should switch.

@gent99 I'm closing this issue since I explained why gpio2 is recommended, but most other pins would also work.

closed 2018-11-18 by Aircoookie