wled/WLED #69: Not persisting settings on ESP8266 ESP-01 512K

wiesendaniel, 2018-11-08:
## Expected behavior
Persisting all settings over restarts and power losses

## Actual behavior
Settings are back to default after restart or power loss

## Steps to reproduce
* Flash binary release on a ESP8266 ESP-01 512K via a simple progammer (CH340)
  * I started with v0.8.0. After installing every thing seemed fine at first but as I was trying to setup wifi everything was back to default after the restart.
  * I went back to the release page to see v0.8.1, 19 minutes ago with a note that said: "Restored 512K ESP8266-01 support". Scrolled down a bit to read "0.8.0 will not work on ESP-01 with 512K flash" and was releaved for a moment. But testing showed the same result. 
  * To make sure it's not one of thoes bigger changes in 0.8. I tried 0.7.1 as well. Same result. 
* Switched the ESP chip to one from a different batch and repeated the steps above. All with the same result.  (actually 5 modules from 3 different batches. I'm reasonably sure that it's not caused by the hardware by now) 

## Further steps
I tried to build from source (0.8.0/0.8.1) but didn't got it working.  (Linker errors are my personal nightmare) But don't worry about that, I've tried to port it to PlatformIO as I'm not used to work with Arduino IDE and multiple source files. And it has a solid dependency manager which really helps a lot with reproducibility.
I'm only mentioning it because:
A. The error referrers to a size issue in the linking proccess and might be relevant. But it's most likely caused by the toolchain PIO useses. Which to my knowledge is more strict than the aruidno toolchain. Could still be a hint at least.
```
../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/esp01/firmware.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp01/firmware.elf] Error 1
``` 
B. I'd like to know what you think about PIO and if you have interest in porting WLED to PIO. If so, let me know, I'd love to help.

### Some more notes
Thank you really much for this project. I started a project a few month back which was pretty similar. I want to connect some RGB leds to my Home Assistant setup via MQTT (due to it's auto discovery feature). As I was picking it up again I found this awesome project while researching and was pleasantly surprised to see the astounding feature set you put together here. So I really have no interest in rolling my own anymore. Instead I want to help to make WLED compatible with Home Assistent over MQTT. My C/C++ foo is a bit rusty, but I'm pretty solid in python and already did a little messy but working POC with auto discovery. If you are interested just let me know.

PS: Sorry for going a bit OT at the end. I hope you don't mind. If you prefer a different way of communication for things that aren't really an "issue", please let me kown as well.

Aircoookie, 2018-11-08:
Hi!

Thanks for your very detailed bug report! The issue basically boils down to the fact that the flash is so limited (512K). In its standard form, the WLED binary itself is already almost 500kB big. This is an issue, because we need at least 64kB for the SPIFFS (filesystem where the settings are stored). So, because there is not enough space, you experience this issue that settings are not retained.

To make it work (assuming v0.8.1) requires several steps, compiling the source with some custom options. You can find the details in the very top of wled00.ino. To make this easier for you, here is a precompiled binary you can flash via esptool or similar: https://drive.google.com/file/d/116dvL0SMoMFTa3BxMXs4UAcgVXHwgHC0/view?usp=sharing

The error you get has got nothing to do with PIO, it just basically says the standard program is too large to fit the 512K. The workaround is to disable some features (OTA updates and the mobile web UI). So, you would have the best experience using a 4M board like NodeMCU or Wemos D1, but I'm actively focused on maintaining ESP01 support, as well.

I have never used PIO, but heard good things about it. For me personally the Arduino ide does just fine, but I will give PlatformIO a shot tomorrow. If I like it, porting WLED will of course be done. Wow, I would be very grateful if you helped me port it! (since as i said I don't yet have any experience with PIO) Shouldn't be too much work though, since I'm 95% using native C++ code. Only the native Arduino functions would need to be swapped for respective PIO ones, I suppose.

Awesome that you believe WLED covers your needs! MQTT is already supported in WLED, so it should work. Let me know if we need any extensions though to make it work perfectly with Home Assistant!
I hope you can get WLED and HA setup flawlessly with the new binary! (I should add it to the releases, it is quite involved to build the ESP01 version from source) Have fun!

You could also contact me via e-mail at "dev.aircoookie@gmail.com" or join my Discord server (link in the bottom of the readme) if you'd prefer. That said, I have no problem whatsoever with using the issue sytem as a discussion board, I just respond on whatever channel you and others prefer :)

wiesendaniel, 2018-11-08:
Thank you for the quick and detailed answer.

The custom build is very appreciated. Sadly I only get around to test this on Sunday. I will report back as soon as I have results. I really love the fact that you plan to support ESP-01 further on.
It's nice to have a really cheap chip to build nice little things as a gift e.g.

Actually the change should not be that big as PIO supports the Arduino framework, but also others that might be smaller and free up some space. If the libraries are not using any Arduino code this might be worth a consideration. But I never got around to try anything but the Arduino Framework.

It expects the source to be in a folder called `src` and your local/private lib code in a folder called (you guessed it) -> `lib`. It prefers its file extension to be .c/.cpp tho. It works with .ino as well and I changed the src dir in the config. 

I pushed my first try if you'd like to take a look at it. https://github.com/wiesendaniel/WLED/tree/feature/pio-port

It was generated by the init command the pio cli. But it's not working quiet well. I'm under the impression that the managed libs aren't actually used. Because it's not working anymore if I delete the dependencies folder. 

Basically everything is configured in the `platformio.ini` file. You can set multiple build targets in it as well as all your dependencies and build flags.
Whats neat about the dependency section is that you can suffix them with @1.2.3 and get the exact version you need. They are installed in a local folder for each project which is ignored by git (see .gitignore). I was not quiet sure which versions you used so I left it blank.
Every thing else in the config is copied from my other projects so the options that are commended out where experiments over there and might not be relevant to this project.

With PIO it's possible to have travis ci build every target on push to master e.g. or even do the hole release for you. Really wicked stuff. And testing... I still need to look into unit testing in C++. :-D

I recommend the VSCode version of PIO as it is much more capable out of the box in terms of code completion and stuff like that. (Disclaimer: I'm not associated with either of those projects, but lernt to love them ;-) ) Just install VSCode and download the PlatformIO Extension. I'm working on Mac an Linux for Windows you might need to take some extra steps I'm not sure about. 

I found the discord invite link right after posting this. Joined! I'm on GMT+1 BTW if that matters.

wiesendaniel, 2018-11-10:
Ok, I got some time to dig in to the code now. Should have done way that earlier.

I have some time to work on the PIO port later. I let you know if I get somewhere with it.
Im going to open another issue, as this hole PIO thing doesn't have any relevance to this issue. 

I might have the chance to test you custom build in a few hours. I have a very good feeling we can close this issue than. ;-)

Aircoookie, 2018-11-10:
I've replied to you in the new PIO issue!

Wonderful, I'm looking forward to your testing :)

wiesendaniel, 2018-11-10:
Works like a works like a treat. Thanks again :)

closed 2018-11-10 by wiesendaniel