wled/WLED #76: Hue -> Echo Plus

DaveDischord, 2018-11-19:
I currently have an Echo Plus with built-in Hue bridge.  Do you or anyone know of a way to get the information needed for syncing via this route?

DaveDischord, 2018-11-20:
> To be honest, I don't think this works, sorry. You can try downloading the
> hue app and see if it discovers the Echo plus as a valid hue bridge. If
> that's the case, it'll work just like with a normal bridge, otherwise it
> can't. I vagualy remember to have heard somewhere that Echo Plus doesn't
> implement the HTTP API part of the hue protocol, only the zigbee part. You
> can of course still use the "normal" Alexa function (on/off and dimming,
> just no color control) and group it together with other lamps as you'd like.

OK, that's what I figured.  A side-question, I have gone through setup, and
Alexa has found my devices (I have 2 new ones) and with each, when I try to
adjust the brightness through the Alexa app or even turn on/off, it won't
change the LEDs, and on the app itself, it'll hold the new value for maybe
2 seconds, then automatically slide back to the previous state.  The Web
GUI works fine though.

Mariu86, 2018-11-20:
I have the same problem, in my setup Alexa it's turning on the LEDs but if I want to turn off it's not working, the same problem like DaveDischord. Thanks

Aircoookie, 2018-11-20:
Ok, I'll look into it. Somehow Alexa is what breaks every other version for some weird reason. You are both using v0.8.1, right?

DaveDischord, 2018-11-20:
I'm using 0.80, but I'll update asap

Mariu86, 2018-11-21:
I'm using 0.80 to.