wled/WLED #79: limit led power consumption (using estimation)

viknet365, 2018-11-28:
Powering a full 5m led strip with 96 RGBW/m  from an ESP8266 seems a bit crazy, but if the effect only drive a few leds its perfectly doable and can be very nice (hello meteor effect).
the only problem is what happens when you crank up the volume :=)

Spoiler: 
```
 Above one amp the schottky diode before the voltage regulator will remind it for you by emitting a strong odor of burned PCB.
```
Since there is already and estimation of the power consumption, it would be nice to avoid lighting more led when this limit is reached.

Assuming the test could be done in real time there is a few things we could do:
1 - Reboot and load probably failsafe settings.
2 - Reduce the overall brightness.
3 - Only light the leds at the start of the strip and don't light the other once the max consumption is reached.
4 - Something else.

viknet365, 2018-11-28:
:+1:  for the tip, in fact I already shorted the schottky wrapping wire on the top of it , but it's here for a reason: after a few unplug from usb something else was burned and the esp no longer boot :-) and I am too lazy reviving one of my many ESP.

Mariu86, 2018-11-28:
It's a good addition this thing ! I have 2 strips 1 with 300 LEDs and 1 with 200 and I must be careful not to set them to max brightness because the power supply it's only 8amp for the strip with 300 leds rgbw and a 4 amp for the strip with 200 LEDs rbg. will be useful to have some safety features.

debsahu, 2018-11-28:
FastLed already has some code that can take care of current limitations: https://github.com/FastLED/FastLED/wiki/Power-notes