wled/WLED #85: WLED-AP doesn't appear

cipiasentini, 2018-12-09:
I didn't want to open an issue but I don't know what the problem might be. Im using a lolin nodemcu v3 (clone). I've uploaded the wled00 sketch and with the tool uploaded the Sketch data via Arduino IDE. The firmware flashes correctly, but I get no "WLED-AP" when scanning the APs. Also the blue inboard led stays always on (I don't know if it is supposed to do so). What could be wrong? I append the uploading messages:
```
Archiving built core (caching) in: /var/folders/pp/mn2cvqg92m5ft3qcf0cf48bc0000gn/T/arduino_cache_5211/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_7028f5f64bb29b9b2663cc4c619c9102.a
El Sketch usa 503040 bytes (48%) del espacio de almacenamiento de programa. El máximo es 1044464 bytes.
Las variables Globales usan 51320 bytes (62%) de la memoria dinámica, dejando 30600 bytes para las variables locales. El máximo es 81920 bytes.
/Users/rltops/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13/esptool -vv -cd nodemcu -cb 115200 -cp /dev/cu.wchusbserial14120 -ca 0x00000 -cf /var/folders/pp/mn2cvqg92m5ft3qcf0cf48bc0000gn/T/arduino_build_597140/wled00.ino.bin 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.wchusbserial14120
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.wchusbserial14120 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 507184 bytes from /var/folders/pp/mn2cvqg92m5ft3qcf0cf48bc0000gn/T/arduino_build_597140/wled00.ino.bin to flash at 0x00000000
	erasing flash
	size: 07bd30 address: 000000
	first_sector_index: 0
	total_sector_count: 124
	head_sector_count: 16
	adjusted_sector_count: 108
	erase_size: 06c000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 1000
	espcomm_send_command: receiving 2 bytes of data
	writing flash
................................................................................ [ 16% ]
................................................................................ [ 32% ]
................................................................................ [ 48% ]
................................................................................ [ 64% ]
................................................................................ [ 80% ]
................................................................................ [ 96% ]
................                                                                 [ 100% ]
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
```

Mariu86, 2018-12-09:
You must configure the wled-ap in the sketch or if you don't set the ssid and password in the sketch it will start automatically after a few seconds. The blue led it's all the time on !

Mariu86, 2018-12-09:
After the sketch it's uploading you cand press the reset button to see if everything works.

cipiasentini, 2018-12-10:
> You must configure the wled-ap in the sketch or if you don't set the ssid and password in the sketch it will start automatically after a few seconds. The blue led it's all the time on !

as said on wiki, the AP could be configured via web. I didn't change anything, I wanted the default WLED-AP access point to appear.

Sceptorrh, 2018-12-10:
Did you try to delete all flash content? Not only the sketch. Wifi settings are stored there.

cipiasentini, 2018-12-10:
What a genius. I had to do that. Thanks man

Aircoookie, 2018-12-10:
Thanks @Sceptorrh ! I would not have thought so quickly that some wrong flash data might be the issue! Normally WLED sets a check byte to 233 to see if the settings flash is valid. Still, there is a 1 in 256 chance that the data in that byte is already 233, so WLED could assume that it is a valid save even if it's not.

closed 2018-12-10 by cipiasentini