wled/WLED #86: I can not for the life of me get this to work [848 LEDS]

neilyboy, 2018-12-09:
Hey there. Your WLED software is exactly what I am looking for! I am currently running Bruh Automations ESP-MQTT-JSON-LED software [https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs]. I am having flickering issues. So I was hoping to run your software.

I have 848 WS2812B Leds all wired up to a 5V 60A power supply. I have power injected every 5M. I get a clean 5.07v reading at the NodeMCU (far end of my power chain). I have simply uploaded the wled00.ino to both the original NodeMCU and a fresh NodeMCU 1.0 (while troubleshooting). The only modification to the code I have made was the requested password changes. When I first plug it in I get the first 10 LEDs to light for a second (then the go out). If I press the reset button again I get the first 10 LED's lighting (last one being green in color). 

I then thought ok maybe it is my NodeMCU boards so I tried just a simple WemosD1 Mini Hooking up the 5v, GND and Data to D4. See Pic Below

[image]

I then get the same thing. If I press the reset button the first 10 LED's light up but then go out and nothing happens while changing things in the web-gui.

[image]

I have configured LED Count to 848 in the LED Preferences tried changing max current from default 750 to both 500 and 900. Nothing seems to make a difference. 
I have  a feeling I am missing something simple but would love any advice you can give me. Your software is EXACTLY what I am looking for so well polished and perfect! I can not thank you enough! I just hope to get it running! 

Sorry if this is not the right place for this but I searched the interwebs and can not find this project on a forum anywhere (unless I am am searching something wrong). Thank you so much in advance!

Neil

neilyboy, 2018-12-09:
I forgot to mention that while running the nodeMCU I also tried a logic level shifter (which is how I had it wired up with the bruh automations package). That did not change anything. Still have nothing after the initial 10 led's light up after reset. I really appreciate anyone who can help me with this. 
Neil

Mariu86, 2018-12-09:
You tried the 0.8.1 version? I have now the 0.8.1 on a strip with 300 leds and it's working ok

neilyboy, 2018-12-10:
Mark this guy as resolved.  How the heck I missed that.... I don't know but thank you! 
Neil

Mariu86, 2018-12-10:
You must make a video to show us the magic of wled :D ...

neilyboy, 2018-12-10:
Im disappointed I pointed the leds at the ceiling.. other than that it looks fantastic. Im putting the kids to bed now but I'll shoot a youtube vid tomorrow for sure! Thanks again buddy! 
Neil

Mariu86, 2018-12-10:
No problem, I am happy for you. Next week I will make my setup for Christmas with a couple of ledstrips and wled to drive them. 😀

Aircoookie, 2018-12-10:
@neilyboy Great that you love it! Whoa, that looks awesome! As Mariu pointed out, you need to set the ABL mA number to the amp rating of your LED power supply. This is a safety feature because some users were connecting to many LEDs without an external PSU. For close to 900 LEDs, it will just turn them all off, because each LED uses about 1mA when black and that alone exceeds the amp limit for just the MCU. Sorry for the temporary inconvenience, but awesome that it works now!

@Mariu86 thanks for your help! :)

closed 2018-12-10 by Aircoookie

JaymerJaymer, 2019-03-02:
did you ever post of video of this in action?

neilyboy, 2019-03-03:
I can grab a video today.  I'll post back later
Neil

neilyboy, 2019-03-04:
https://youtu.be/XnlZ7tuUhGk

Sorry for the poor quality. My phone struggles in low light.  It's much better in person.  I did however place the leds facing upwards when I should have pointed them out.  I plan on doing my house soffits this spring and I will point them outward. 
Neil