wled/WLED #87: Led flickering

Mazdaracer121, 2018-12-12:
Hey Aircookie. Great Project by the way 👌
I got a Problem that the last 30 leds in a 224 led line flickers... In solid color and in effects too.
Is there any workaroud?
Led stripe Gnd and Gnd From Wemos d1 mini are connected 
Im upload a video to show the Problem.
https://youtu.be/MOGMHYbo3Dc

Aircoookie, 2018-12-12:
Hi, thanks for the report! That much flicker is really strange (there is another issue https://github.com/Aircoookie/WLED/issues/64 with occasional white flicker when in Fastled effect modes)

I'm assuming you are using 0.8.2 and you set the ABL milliamp limit  to that of your power supply (for 224 LEDs you don't need more than 15A in any case)
I think this is an issue with the LED driving. You can try if using a different pin (and thus a different driver method) works (you can change it in `NpbWrapper.h`). Otherwise it might be that your setup actually requires a levelshifter 3.3V->5V like some people recommend. You might also have success with a shorter or different data cable.

Mazdaracer121, 2018-12-30:
Level Shifter dont Help.
I Changed the Led pin from 2 to 3, but than, the Stripe Dont response to my Wemos D1 mini...
My data cable is a 3cm

Mariu86, 2018-12-30:
You try to change the wemos d1? Maybe it's a hardware problem try to switch to another wemos or a nodemcu, I think I have some problems like you when I put a 1500 uf condensator to my setup. Happy holidays to all!

Aircoookie, 2019-10-25:
It seems like this is #230 :)

closed 2019-10-25 by Aircoookie