wled/WLED #88: MQTT connection issues

alextrical, 2018-12-12:
I'm currently experiencing issues not being able to connect to any MQTT broker, including tcp://test.mosquitto.org:1883 could someone please point me in the right direction to setting this up?

Aircoookie, 2018-12-12:
Hi!
Did you try leaving the tcp:// and :1883 out? Connecting to just test.mosquitto.org should work, tcp protocol and port are automatically assumed.

alextrical, 2018-12-12:
I have tried the plain domain name too, (though i will set that now you have confirmed that is the correct format) along with doing the same for a local hosted server on a raspberry pi, though i cant seem to see any messages when i'm subscribed. 

I will check now and let you know if that fixes things :) 
oh, and btw thank you for making something awesome and getting back so fast

closed 2019-03-27 by Aircoookie