wled/WLED #92: Out of Scope Definition of getUnixTime

aenertia, 2018-12-20:
in `wled10_ntp`

the set countdown block Line - 
`void setCountdown()
{
  countdownTime = timezones[currentTimezone]->toUTC(getUnixTime(countdownHour, countdownMin, countdownSec, countdownDay, countdownMonth, countdownYear));
  if (countdownTime - now() > 0) countdownOverTriggered = false;
}`

produces a compilation error relating to an out of scope function 'getUnixTime'

I believe this needs to be declared within the function as a private variable prior to it being called - but my C foo is lacking. 

Interestingly this code compiles fine under platformio (esp32 target) but not with the arduino IDE.

Aircoookie, 2018-12-20:
`getUnixTime(...)` is implemented in `src/dependencies/time/Time.cpp` and should be included automatically when compiling. I will investigate this further. Can you tell me which ESP32 Arduino SDK version you use? 1.0.0 or an older version? I'm still running on some April 2017 dev build, time to upgrade anyways!

Aircoookie, 2018-12-20:
Alright, I have a hunch, but I can't verify it myself right now. Try deleting the file `Time.h` in `src/dependencies/time/` and also the `#include "src/dependencies/time/Time.h"` in line 60 of wled00.ino.
Maybe it'll help.

aenertia, 2018-12-21:
Yeah - I think it stems from having multiple versions hanging around. As mentioned PIO doesn't suffer from the issue, and TBH the arduino ide is an absolute trash fire.

marc-gist, 2019-10-14:
This is still happening :(

marc-gist, 2019-11-16:
Anyone able to fix this and compile under PlatformIO and VSCode yet?

marc-gist, 2019-11-19:
Now worse...  Could not find a version that satisfies the requirement '1.1.0' for your system 'windows_x86':

Aircoookie, 2019-11-19:
@marc-gist Sorry.. I thought PIO versioning was the same as Arduino library version. Turns out it isn't. Does changing the version in platformio.ini to `ESPAsyncUDP@697c75a02` work?

Also will update my PIO today and see if can reproduce the getUnixTime issue.

Aircoookie, 2019-11-19:
Just pushed an update to master! This does fix the issue with the ESPAsyncUDP dependency (therefore also fixing CI), but it may also solve the issue with getUnixTime. Unfortunately, i couldn't reproduce that one, neither with the Arduino IDE nor with PIO.

marc-gist, 2019-12-08:
still getting this error in VS Code with PlatformIO :(

nordeep, 2019-12-14:
One of the reason of the compiling error - you have installed original Time library in PlatformIO Global Storage. You have to remove Time library from Global Storage(PIO Home -> Libraries -> Installed).
Perhaps the best solution would be move getUnixTime() function to another place. Or make another TimeModified library to avoid collision with original Time library - https://github.com/PaulStoffregen/Time

closed 2020-04-19 by stale[bot]