wled/WLED #93: Weekday support for macros

MatFl, 2018-12-21:
I was just about to start implementing weekday support when I noticed, that it is already kind of supported.  There just seems to be no way to change the setting.

Do you already have something planned for this? Otherwise I would give it a try. Should I use the dev or master branch for things like this?

And another question: Do I just modify the html stuff in html_settings.h or what is the best way to change the UI?

Thanks!

Aircoookie, 2018-12-21:
Hi!
Yes, I initially planned to support weekdays when I implemented the timed macro feature, that's why it is internally supported, however at the time I ran into the roadblock that there are 7x 8 checkboxes required and updating them all manually would cost way too much memory and bandwidth for the ESP8266.

Since then, I found a possible solution, and that is sending the raw bytes to the settings UI page only (I mean this stuff:
```
byte timerWeekday[] = {255,255,255,255,255,255,255,255}; //weekdays to activate on
//bit pattern of arr elem: 0b11111111: sat,fri,thu,wed,tue,mon,sun,validity
```
and have some javascript actually build a table with a checkbox for each day of the week.

To fully support weekday configuration, we need both that javascript as well as the communication between the WLED server and UI (you can see how it works in wled02_xml (sending internal values to UI) and wled03_set (saving values from UI)). My current plan is to add weekday support relatively soon, in v0.9.0. Sadly it will take about a month from now until I have enough time to seriously start developing new features again (I'm occupied with university stuff right now, but some of it will massively benefit WLED, I promise!). So if you need the feature sooner, feel free to give it a shot!

Yes, the html_settings.h is the file which contents actually matter, but for a better overview I recommend using the "data/settings_time.htm" file while making changes.

Have a great Christmas week!

MatFl, 2018-12-22:
Okay, if I find enough time in the next few weeks I will give it a shot :)

closed 2019-10-02 by Aircoookie